openwrt/packages #25439: https-dns-proxy: can't set log verbosity

tobelogin, 2024-11-24:
Maintainer: @stangri 
Environment: ARMv8 Processor rev 4, FriendlyElec NanoPi R2S, OpenWrt 23.05.3

Description:
I'm trying to let https dns proxy log to a file and trying to set the logs' verbosity to warning, but it seems failed when i use ps command to check its command line parameter.
[image]

tobelogin, 2024-11-25:
Just checked the source code and knew it, sorry for disturbing...

closed 2024-11-25 by tobelogin

stangri, 2024-11-25:
There are people in the core team who are actively trying to suppress the OpenWrt-specific information for some packages, but there's a README which lists all the available uci config options: https://docs.openwrt.melmac.net/https-dns-proxy/

tobelogin, 2024-11-25:
Well, sorry to hear that, but actually it's just because i misunderstood the usage of '-v' option. In the upstream code, this option receives no parameter, each time this option appears in command that starts this app, the log will be more verbose. however, i think the followed parameter will tell the app how verbose the log should be...