openwrt/packages #25441: FreeRADIUS - still seriously maintained in openWRT?

wolfgangr, 2024-11-24:
Maintainer: rosenp@gmail.com  (sorry, you were the last name I found comitting)

Environment: 
OPENWRT_ARCH="x86_64"
OPENWRT_BOARD="x86/64"
VERSION_ID="23.05.5"

real Hardware is a Fujitsu S920, 16 GB Flash, 4 core AMD GX-415GA, 4 GB Memory
( not yet in "production", average load << 0.01 )

Description:  
followed this instruction:  
https://openwrt.org/docs/guide-user/network/wifi/freeradius  
https://openwrt.org/docs/guide-user/network/wifi/freeradius#set_up_freeradius_for_initial_testing

Contrary to the documentation, it required three awful hacks to arrive at least at the "hello world" type   

```Ready to process requests. ```

Error #1
```(TLS) Failed loading legacy provider```

With some luck and patience, the internet revealed an issue 1 1/2 years old:
https://github.com/openwrt/packages/issues/21428  
installing `libopenssl-legacy` solved this one  
 ... just to hit the next wall - Error #2: 

```
including configuration file /etc/freeradius3/policy.d/eap
including files in directory /etc/freeradius3/sites-enabled/
including configuration file /etc/freeradius3/sites-enabled/default
/etc/freeradius3/sites-enabled/default[1054]: Parse error in condition
/etc/freeradius3/sites-enabled/default[1054]: (EAP-Key-Name && &reply:EAP-Session-Id) {
/etc/freeradius3/sites-enabled/default[1054]:  ^ Expected a module return code
Errors reading or parsing /etc/freeradius3/radiusd.conf
```

still in RADIUS infancy, I just commented out the offending stanza - without any clue what it's purpose might have been.

just to hit the next wall - error #3:

```
reading pairlist file /etc/freeradius3/mods-config/attr_filter/coa
Couldn't open /etc/freeradius3/mods-config/attr_filter/coa for reading: No such file or directory
Errors reading /etc/freeradius3/mods-config/attr_filter/coa
/etc/freeradius3/mods-enabled/attr_filter[58]: Instantiation failed for module "attr_filter.coa"
```

again, I just uncommented the offending section .... 
and finally arrived at the the desired

```Ready to process requests```

However, in the ligtht of this experience, I seriosly doubt whether it's a good idea to run FreeRADIUS at an OpenWrt box at all - evn if it is an x86 with ample RAM and Flash, compared to embedded devices.   
Looks like I'm the first one to try RADIUS for more than a year???

And I even doubt whether RADIUS is required for 802.11X at all.

Some background on my use case:
I'm setting up acces for some farm premises
- lots of area, in my case 20.000 sqare meters
- at least 6, maybe >> 10 WiFI AP
- small numer of users << 10
- growing number of IoT gadgets (DHCP leases hitting 100 soon)
- cctv planned ( ~ 20 cams required for decent coverage) - have been lifted 3 times the last 5 years...
- some incooperative stuff on OSI 2 level ( photovoltaic inverters, TV media)
- desire to separate sensible office access from physically unprotectable network segments "out in the wilderness"
- just implementing manged switches with a bunch of used HPE 1810 / 1820 to get my LAN segmented

So may be, FreeRADIUS is an overkill, but nevertheless, I'd strive for segmentation of my LAN and WLAN into distinct realms.
Is this a use case begging for more attention?

brada4, 2024-11-25:
For last bullet list yll have to go to forums.

first is duplicate report, other two means invalid default configuration, for which @neheb can do something.

predators46, 2024-11-26:
if freeradius is not maintained and is useless why would i update freeradius in openwrt? Mistakes are the key to success

wolfgangr, 2024-11-27:
> For last bullet list yll have to go to forums.

Agree. See here:  
https://forum.openwrt.org/t/how-to-setup-vlan-infrastructure-with-wpa-enterprise-radius-friends/216722

Sorry for pouring my RADIUS novice frustration upon you.  
After digging throu freeRADIUS pages & friends, I think that there may be use cases to run RADIUS on openWRT - but not all of them. And may be, openWRT is not the best platform to _learn_ RADIUS...

waddras, 2025-12-28:
for me it just worked just needed 
```
opkg install libopenssl-legacy
opkg install freeradius3-default
/etc/init.d/radiusd stop
radiusd -X
```