openwrt/packages #25456: apk-tools: compilation failure

gonciarz, 2024-11-27:
Maintainer(s): Robert Marko <robimarko@gmail.com>, Jonas Gorski <jonas.gorski@gmail.com>, Christian Marangi <ansuelsmth@gmail.com>, Petr Štetiar <ynezz@true.cz>, Felix Fietkau <nbd@nbd.name>, Paul Spooren <mail@aparcar.org>
Environment: local dev env, built on ArchLinux arm64 (against FriendlyArm NanoPc-T6)

Description:

I try to build an image from main branch (run against: eed39d4):

```
git clone https://git.openwrt.org/openwrt/openwrt.git
cd openwrt
./scripts/feeds update -a
./scripts/feeds install -a

cat << EOF > .config
CONFIG_TARGET_rockchip=y
CONFIG_TARGET_rockchip_armv8=y
CONFIG_TARGET_rockchip_armv8_DEVICE_friendlyarm_nanopc-t6=y
EOF

make -j1 V=s defconfig download clean world
```

The script stops on compiling apk-tools:
```
make[3] -C package/system/apk compile
    ERROR: package/system/apk failed to build (build variant: mbedtls).
make -r world: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
make: *** [/sync/workspace/openwrt3/openwrt/include/toplevel.mk:248: world] Error 1
```

It seems like a root cause:
```
/usr/include/stdio.h:53:9: error: unknown type name '__gnuc_va_list'
   53 | typedef __gnuc_va_list va_list;
      |         ^~~~~~~~~~~~~~
/usr/include/stdio.h:53:24: error: conflicting types for 'va_list'; have 'int'
   53 | typedef __gnuc_va_list va_list;
      |                        ^~~~~~~
```

Full log (with V=s):
```
make[3]: Entering directory '/sync/workspace/openwrt/package/system/apk'
touch /sync/workspace/openwrt/build_dir/target-aarch64_generic_musl/apk-mbedtls/apk-3.0.0_pre20241113/.prepared_951970ef98e51ce9898a417ff2e5cf7c_18f1e190c5d53547fed41a3eaa76e9e9_check
. /sync/workspace/openwrt/include/shell.sh; zstdcat /sync/workspace/openwrt/dl/apk-3.0.0_pre20241113.tar.zst | tar -C /sync/workspace/openwrt/build_dir/target-aarch64_generic_musl/apk-mbedtls/apk-3.0.0_pre20241113/.. -xf -
[ ! -d ./src/ ] || cp -fpR ./src/. /sync/workspace/openwrt/build_dir/target-aarch64_generic_musl/apk-mbedtls/apk-3.0.0_pre20241113

Applying /sync/workspace/openwrt/package/system/apk/patches/0001-openwrt-move-layer-db-to-temp-folder.patch using plaintext: 
patching file src/database.c

Applying /sync/workspace/openwrt/package/system/apk/patches/0010-app_list-add-full-print.patch using plaintext: 
patching file src/app_list.c
touch /sync/workspace/openwrt/build_dir/target-aarch64_generic_musl/apk-mbedtls/apk-3.0.0_pre20241113/.prepared_951970ef98e51ce9898a417ff2e5cf7c_18f1e190c5d53547fed41a3eaa76e9e9
rm -f /sync/workspace/openwrt/build_dir/target-aarch64_generic_musl/apk-mbedtls/apk-3.0.0_pre20241113/.configured_*
rm -f /sync/workspace/openwrt/staging_dir/target-aarch64_generic_musl/stamp/.apk.mbedtls_installed
/sync/workspace/openwrt/staging_dir/host/bin/sed -e "s|@CC@|'/sync/workspace/openwrt/staging_dir/host/bin/gcc',|" -e "s|@CXX@|'/sync/workspace/openwrt/staging_dir/host/bin/g++',|" -e "s|@PKGCONFIG@|/sync/workspace/openwrt/staging_dir/host/bin/pkg-config|" -e "s|@CMAKE@|/sync/workspace/openwrt/staging_dir/host/bin/cmake|" -e "s|@PYTHON@|/sync/workspace/openwrt/staging_dir/host/bin/python3|" -e "s|@CFLAGS@|'-O2', '-I/sync/workspace/openwrt/staging_dir/host/include', '-I/sync/workspace/openwrt/staging_dir/hostpkg/include', '-I/sync/workspace/openwrt/staging_dir/target-aarch64_generic_musl/host/include', '-I/sync/workspace/openwrt/staging_dir/host/include', '-I/sync/workspace/openwrt/staging_dir/hostpkg/include', '-I/sync/workspace/openwrt/staging_dir/target-aarch64_generic_musl/host/include',|" -e "s|@CXXFLAGS@|'-O2', '-I/sync/workspace/openwrt/staging_dir/host/include', '-I/sync/workspace/openwrt/staging_dir/hostpkg/include', '-I/sync/workspace/openwrt/staging_dir/target-aarch64_generic_musl/host/include', '-I/sync/workspace/openwrt/staging_dir/host/include', '-I/sync/workspace/openwrt/staging_dir/hostpkg/include', '-I/sync/workspace/openwrt/staging_dir/target-aarch64_generic_musl/host/include',|" -e "s|@LDFLAGS@|'-L/sync/workspace/openwrt/staging_dir/host/lib', '-L/sync/workspace/openwrt/staging_dir/hostpkg/lib', '-L/sync/workspace/openwrt/staging_dir/target-aarch64_generic_musl/host/lib',|" -e "s|@PREFIX@|/sync/workspace/openwrt/staging_dir/host|" < /sync/workspace/openwrt/staging_dir/host/lib/meson/openwrt-native.txt.in > /sync/workspace/openwrt/build_dir/target-aarch64_generic_musl/apk-mbedtls/apk-3.0.0_pre20241113/openwrt-native.txt
/sync/workspace/openwrt/staging_dir/host/bin/sed -e "s|@CC@|'aarch64-openwrt-linux-musl-gcc',|" -e "s|@CXX@|'aarch64-openwrt-linux-musl-g++',|" -e "s|@LD@|'bfd',|" -e "s|@AR@|aarch64-openwrt-linux-musl-gcc-ar|" -e "s|@STRIP@|aarch64-openwrt-linux-musl-strip|" -e "s|@NM@|aarch64-openwrt-linux-musl-gcc-nm|" -e "s|@PKGCONFIG@|/sync/workspace/openwrt/staging_dir/host/bin/pkg-config|" -e "s|@CMAKE@|/sync/workspace/openwrt/staging_dir/host/bin/cmake|" -e "s|@PYTHON@|/sync/workspace/openwrt/staging_dir/host/bin/python3|" -e "s|@CFLAGS@|'-Os', '-pipe', '-mcpu=generic', '-fno-caller-saves', '-fno-plt', '-fhonour-copts', '-ffile-prefix-map=/sync/workspace/openwrt/build_dir/target-aarch64_generic_musl/apk-mbedtls/apk-3.0.0_pre20241113=apk-3.0.0_pre20241113', '-Wformat', '-Werror=format-security', '-fstack-protector', '-D_FORTIFY_SOURCE=1', '-Wl,-z,now', '-Wl,-z,relro', '-I/sync/workspace/openwrt/staging_dir/toolchain-aarch64_generic_gcc-13.3.0_musl/usr/include', '-I/sync/workspace/openwrt/staging_dir/toolchain-aarch64_generic_gcc-13.3.0_musl/include', '-I/sync/workspace/openwrt/staging_dir/toolchain-aarch64_generic_gcc-13.3.0_musl/include/fortify',|" -e "s|@CXXFLAGS@|'-Os', '-pipe', '-mcpu=generic', '-fno-caller-saves', '-fno-plt', '-fhonour-copts', '-ffile-prefix-map=/sync/workspace/openwrt/build_dir/target-aarch64_generic_musl/apk-mbedtls/apk-3.0.0_pre20241113=apk-3.0.0_pre20241113', '-Wformat', '-Werror=format-security', '-fstack-protector', '-D_FORTIFY_SOURCE=1', '-Wl,-z,now', '-Wl,-z,relro', '-I/sync/workspace/openwrt/staging_dir/toolchain-aarch64_generic_gcc-13.3.0_musl/usr/include', '-I/sync/workspace/openwrt/staging_dir/toolchain-aarch64_generic_gcc-13.3.0_musl/include', '-I/sync/workspace/openwrt/staging_dir/toolchain-aarch64_generic_gcc-13.3.0_musl/include/fortify',|" -e "s|@LDFLAGS@|'-L/sync/workspace/openwrt/staging_dir/toolchain-aarch64_generic_gcc-13.3.0_musl/usr/lib', '-L/sync/workspace/openwrt/staging_dir/toolchain-aarch64_generic_gcc-13.3.0_musl/lib', '-fuse-ld=bfd', '-znow', '-zrelro',|" -e "s|@ARCH@|"aarch64"|" -e "s|@CPU@|"generic"|" -e "s|@ENDIAN@|little|" < /sync/workspace/openwrt/staging_dir/host/lib/meson/openwrt-cross.txt.in > /sync/workspace/openwrt/build_dir/target-aarch64_generic_musl/apk-mbedtls/apk-3.0.0_pre20241113/openwrt-cross.txt
VERSION=3.0.0_pre20241113 /sync/workspace/openwrt/staging_dir/host/bin/python3 /sync/workspace/openwrt/staging_dir/host/bin/meson.py  setup --buildtype plain --native-file /sync/workspace/openwrt/build_dir/target-aarch64_generic_musl/apk-mbedtls/apk-3.0.0_pre20241113/openwrt-native.txt --cross-file /sync/workspace/openwrt/build_dir/target-aarch64_generic_musl/apk-mbedtls/apk-3.0.0_pre20241113/openwrt-cross.txt -Ddefault_library=both -Dcompressed-help=false -Ddocs=disabled -Dhelp=enabled -Dlua_version=5.1 -Durl_backend=wget -Dzstd=false -Dcrypto_backend=mbedtls /sync/workspace/openwrt/build_dir/target-aarch64_generic_musl/apk-mbedtls/apk-3.0.0_pre20241113/openwrt-build /sync/workspace/openwrt/build_dir/target-aarch64_generic_musl/apk-mbedtls/apk-3.0.0_pre20241113/openwrt-build/..
The Meson build system
Version: 1.5.1
Source dir: /sync/workspace/openwrt/build_dir/target-aarch64_generic_musl/apk-mbedtls/apk-3.0.0_pre20241113
Build dir: /sync/workspace/openwrt/build_dir/target-aarch64_generic_musl/apk-mbedtls/apk-3.0.0_pre20241113/openwrt-build
Build type: cross build
Project name: apk-tools
Project version: 3.0.0_pre20241113
C compiler for the host machine: aarch64-openwrt-linux-musl-gcc (gcc 13.3.0 "aarch64-openwrt-linux-musl-gcc (OpenWrt GCC 13.3.0 r28180-b8b658bc0d) 13.3.0")
C linker for the host machine: aarch64-openwrt-linux-musl-gcc ld.bfd 2.42
C compiler for the build machine: /sync/workspace/openwrt/staging_dir/host/bin/gcc (gcc 14.2.1 "gcc (GCC) 14.2.1 20240910")
C linker for the build machine: /sync/workspace/openwrt/staging_dir/host/bin/gcc ld.bfd 2.43.0
Build machine cpu family: x86_64
[... 160 lines cut ...]
      |                             ^~~~~~~~~~~~~~
In file included from /usr/include/stdio.h:954:
[... cut ...]
time: package/system/apk/mbedtls/compile#4.77#0.87#5.46
    ERROR: package/system/apk failed to build (build variant: mbedtls).
make[2]: *** [package/Makefile:185: package/system/apk/compile] Error 1
make[2]: Leaving directory '/sync/workspace/openwrt'
make[1]: *** [package/Makefile:179: /sync/workspace/openwrt/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/sync/workspace/openwrt'
make: *** [/sync/workspace/openwrt/include/toplevel.mk:248: world] Error 2
```

esaaprillia, 2024-12-03:
@neheb

neheb, 2024-12-03:
Something is leaking host paths. Most likely mbedtls

rkakrik, 2024-12-15:
The problem still exists(

neheb, 2024-12-15:
indeed

esaaprillia, 2024-12-15:
I've had this problem when building other packages. maybe this problem is caused when building the package he chose the path /usr/include and /usr/lib, namely the machine path used, when building the package the path should have been chosen (STANGING_DIR)/usr/include and (STANGING_DIR)/usr/lib . this is a problem when cross compiling

closed 2024-12-18 by openwrt-bot