openwrt/packages #25471: dockerd containers cannot connect to each other

slavb18, 2024-12-01:
Maintainer: @p-w-p 
Environment:   `OpenWrt SNAPSHOT, r28214-5a4eb56a7b` 
`Linux OpenWrt 6.6.63 #0 SMP Thu Nov 28 21:33:21 2024 aarch64 GNU/Linux`

Description:
Connection refused between containers
Example `docker-compose.yml`
```
version: '3'

name: dockertest

services:
  nginx:
    container_name: nginx
    image: nginx:latest
    ports:
      - "9999:80"
    volumes:
      - ./html:/usr/share/nginx/html
    restart: always
  test:
    container_name: test
    image: alpine/curl
    command: curl -v http://nginx:80
    depends_on:
      - nginx
    healthcheck:
      test: ping -c 1 nginx || exit 1
    restart: always
```

On ubuntu it works, but on openwrt i see `connect to 172.19.0.2 port 80 from 172.19.0.3 port 57500 failed: Connection refused`

Maybe this is firewall related problem

Here is docker part of `/etc/config/network`
```
config interface 'docker'
        option device 'docker0'
        option proto 'none'
        option auto '0'

config device
        option type 'bridge'
        option name 'docker0'
```

And `/etc/config/firewall`

```
config zone 'docker'
        option input 'ACCEPT'
        option output 'ACCEPT'
        option forward 'ACCEPT'
        option name 'docker'
        list network 'docker'
        option log '1'

config forwarding
        option src 'docker'
        option dest 'lan'

config forwarding
        option src 'lan'
        option dest 'docker'
```

hudra0, 2024-12-02:
This is a known behavior related to how Docker's default bridge network handles container communication, which becomes more apparent in OpenWrt's network stack.

The root cause:
- OpenWrt by default uses Docker's default bridge network
- In the default bridge network, containers can only access each other by IP addresses
- DNS name resolution doesn't work between containers in the default bridge
- That's why your `curl http://nginx:80` fails (DNS resolution of "nginx" fails)

You have two options to solve this:

1. Using Default Bridge with Links (Legacy Solution):
```yaml
version: '3'

name: dockertest

services:
  nginx:
    container_name: nginx
    image: nginx:latest
    network_mode: bridge
    ports:
      - "9999:80"
    volumes:
      - ./html:/usr/share/nginx/html
    restart: always
    
  test:
    container_name: test
    image: alpine/curl
    network_mode: bridge
    links:
      - 'nginx:nginx'
    command: curl -v http://nginx:80
    depends_on:
      - nginx
    healthcheck:
      test: ping -c 1 nginx || exit 1
    restart: always
```
2. Using Custom Network (Recommended Solution):

```yaml
version: '3'

name: dockertest

services:
  nginx:
    container_name: nginx
    image: nginx:latest
    networks:
      - custom-bridge
    ports:
      - "9999:80"
    volumes:
      - ./html:/usr/share/nginx/html
    restart: always
    
  test:
    container_name: test
    image: alpine/curl
    networks:
      - custom-bridge
    command: curl -v http://nginx:80
    depends_on:
      - nginx
    healthcheck:
      test: ping -c 1 nginx || exit 1
    restart: always

networks:
  custom-bridge:
    driver: bridge
    driver_opts:
      com.docker.network.bridge.name: docker1
```
For the custom network solution, you'll need these additional steps in OpenWrt:

**Create a new interface in LuCI:**

- Network > Interfaces > Add new interface
- Set protocol to "Unmanaged"
- Set device to "docker1" (matching the bridge name in docker-compose)
- Uncheck "Bring up on boot"

**Configure firewall:**

- Assign the new interface to the docker firewall zone (or a new zone)
- Ensure the docker zone has proper settings

3. Restart Docker (probably better to restart) :
```
service dockerd restart
```
The custom network solution is recommended because:

- It provides automatic DNS resolution between containers
- Containers can resolve each other by name or alias
- Better isolation from other containers
- Each user-defined network can have its own settings

The reason this works on Ubuntu but not on OpenWrt by default is that OpenWrt always uses the default bridge, while on Ubuntu a user-defined network might be created automatically depending on your configuration.

closed 2024-12-03 by slavb18

namchuai, 2025-05-02:
@hudra0 Could you please provide step by step for this? I'm stuck.
```
Configure firewall:

Assign the new interface to the docker firewall zone (or a new zone)
Ensure the docker zone has proper settings
```

hudra0, 2025-05-07:
When you create a new interface you should assign an existing firewall zone or create a new one:

[image]

hudra0, 2025-05-14:
Please show your compose/run command

namchuai, 2025-05-15:
Thank you for your dedication! Here's my docker-compose.yml.
```yaml
services:
  app:
    image: 'jc21/nginx-proxy-manager:latest'
    restart: unless-stopped
    ports:
      - '80:80'
      - '443:443'
      - '81:81'
    networks:
      - custom-bridge
    environment:
      DB_MYSQL_HOST: "db"
      DB_MYSQL_PORT: 3306
      DB_MYSQL_USER: "npm"
      DB_MYSQL_PASSWORD: "npm"
      DB_MYSQL_NAME: "npm"
      DISABLE_IPV6: 'true'
    volumes:
      - ./data:/data
      - ./letsencrypt:/etc/letsencrypt
    depends_on:
      - db

  db:
    image: 'jc21/mariadb-aria:latest'
    restart: unless-stopped
    networks:
      - custom-bridge
    environment:
      MYSQL_ROOT_PASSWORD: 'npm'
      MYSQL_DATABASE: 'npm'
      MYSQL_USER: 'npm'
      MYSQL_PASSWORD: 'npm'
      MARIADB_AUTO_UPGRADE: '1'
    volumes:
      - ./mysql:/var/lib/mysql

networks:
  custom-bridge:
    driver: bridge
    driver_opts:
      com.docker.network.bridge.name: docker1
```

hudra0, 2025-05-17:
In the compose file you shared the ports used by the stack conflict with OpenWrt’s Luci interface:

Luci (via uhttpd) listens on the router’s IP on ports 80 (HTTP) and 443 (HTTPS).

Your compose file tries to bind those exact ports to the Nginx Proxy Manager (NPM) container:

```
ports:
  - "80:80"
  - "443:443"
  - "81:81"

```
When Docker attempts to bind to ports 80/443, they’re already in use by uhttpd so either luci fails or the container won't start.

There are three possible solutions:

1. Move luci to different ports (let NPM keep 80/443) (not tested)

```
uci set uhttpd.main.listen_http='0.0.0.0:8080'
uci set uhttpd.main.listen_https='0.0.0.0:8443'
uci commit uhttpd
service uhttpd restart
```
Luci will now be accessible at ports 8080 (HTTP) and 8443 (HTTPS).
You can leave your Docker Compose file as-is and npm will be able to use ports 80/443 for let’s encrypt.

2. Keep luci on ports 80/443, and move npm to different host ports

```
services:
  app:
    image: jc21/nginx-proxy-manager:latest
    ports:
      - "8080:80"
      - "8443:443"
      - "81:81"
```
This avoids port conflicts, but let’s encrypt challenges on port 80/443 may not work. You’d need a workaround, such as DNS-based validation.

Alternative: Use a macvlan network
This gives your npm container its own IP address on the LAN, avoiding any port conflicts entirely:

```
services:
  app:
    image: jc21/nginx-proxy-manager:latest
    container_name: npm
    restart: unless-stopped
    networks:
      npm_net:
        ipv4_address: 192.168.1.10
    environment:
      DB_MYSQL_HOST: "192.168.1.11"
      DB_MYSQL_PORT: 3306
      DB_MYSQL_USER: "npm"
      DB_MYSQL_PASSWORD: "npm"
      DB_MYSQL_NAME: "npm"
      DISABLE_IPV6: "true"
    volumes:
      - ./data:/data
      - ./letsencrypt:/etc/letsencrypt
    depends_on:
      - db

  db:
    image: jc21/mariadb-aria:latest
    container_name: npm-db
    restart: unless-stopped
    networks:
      npm_net:
        ipv4_address: 192.168.1.11
    environment:
      MYSQL_ROOT_PASSWORD: "npm"
      MYSQL_DATABASE: "npm"
      MYSQL_USER: "npm"
      MYSQL_PASSWORD: "npm"
      MARIADB_AUTO_UPGRADE: "1"
    volumes:
      - ./mysql:/var/lib/mysql

networks:
  npm_net:
    driver: macvlan
    driver_opts:
      parent: br-lan   # ← your LAN bridge interface
    ipam:
      config:
        - subnet: 192.168.1.0/24
          gateway: 192.168.1.1
```
Now npm gets its own IP (e.g., 192.168.1.10) and can use ports 80/443 independently of the router.

namchuai, 2025-05-20:
I forgot to mention, I was moved luci to port 88 and disabled 443 for luci. Thanks so much for your help, I moved away from this and running docker in a mini computer instead.