openwrt/packages #25483: node: undefined reference to absl::lts_20230802

vortexilation, 2024-12-03:
Maintainer: @nxhack 
Environment: x86_64

Description:

Posting it in here also.

It seems that node will failed to compile if there is already compiled abseil-cpp in $(STAGING_DIR_HOSTPKG)/lib dir with [the following error](https://pastebin.com/EWhEmzTh).

Abseil-cpp was compiled from bundled in protobuf 25.3, here is the draft Makefile : 
```
#
# Copyright (C) 2007-2015 OpenWrt.org
#
# This is free software, licensed under the GNU General Public License v2.
# See /LICENSE for more information.
#

include $(TOPDIR)/rules.mk

PKG_NAME:=protobuf
PKG_VERSION:=25.3
PKG_RELEASE:=1

PKG_SOURCE_PROTO:=git
PKG_SOURCE_URL:=https://github.com/protocolbuffers/protobuf.git
PKG_MIRROR_HASH:=63ddc2978bc7cdd357f66ed70f273a787960e990a328887b7f65d119094c7353
PKG_SOURCE_DATE:=2024-11-08
PKG_SOURCE_VERSION:=4a2aef570deb2bfb8927426558701e8bfc26f2a4

PKG_MAINTAINER:=Ken Keys <kkeys@caida.org>
PKG_LICENSE:=BSD-3-Clause
PKG_LICENSE_FILES:=LICENSE
PKG_CPE_ID:=cpe:/a:google:protobuf

#CMAKE_BINARY_SUBDIR:=build

include $(INCLUDE_DIR)/package.mk
include $(INCLUDE_DIR)/host-build.mk
include $(INCLUDE_DIR)/cmake.mk

define Package/protobuf/Default
  SECTION:=libs
  CATEGORY:=Libraries
  TITLE:=A structured data encoding library
  URL:=https://github.com/google/protobuf
  DEPENDS:=+zlib +libpthread +libatomic +libstdcpp
endef

define Package/protobuf
  $(call Package/protobuf/Default)
  DEPENDS+=+protobuf-lite
endef

define Package/protobuf-lite
  $(call Package/protobuf/Default)
endef

define Package/protobuf/description/Default
Protocol Buffers are a way of encoding structured data in an efficient
yet extensible format. Google uses Protocol Buffers for almost all
of its internal RPC protocols and file formats.
endef

define Package/protobuf/description
$(call Package/protobuf/description/Default)

This package provides the libprotoc, libprotobuf, and libprotobuf-lite
libraries.  For a much smaller protobuf package, see "protobuf-lite".

endef

define Package/protobuf-lite/description
$(call Package/protobuf/description/Default)

This package provides the libprotobuf-lite library.

endef

#ifdef CONFIG_USE_MUSL
#TARGET_LDFLAGS += \
#	-Wl,-rpath $(STAGING_DIR_ROOT)/lib:$(STAGING_DIR_ROOT)/usr/lib
#endif

CMAKE_HOST_OPTIONS += \
	-Dprotobuf_BUILD_PROTOC_BINARIES=ON \
	-Dprotobuf_BUILD_TESTS=OFF \
	-DBUILD_SHARED_LIBS=OFF \
	-DCMAKE_INSTALL_LIBDIR=lib \
	-Dprotobuf_ABSL_PROVIDER=module \
#	-Dabsl_DIR=$(STAGING_DIR_HOSTPKG)/lib/cmake/absl \
	-DCMAKE_INSTALL_PREFIX=$(STAGING_DIR_HOSTPKG)

CMAKE_OPTIONS += \
	-Dprotobuf_BUILD_PROTOC_BINARIES=ON \
	-Dprotobuf_BUILD_TESTS=OFF \
	-Dprotobuf_WITH_ZLIB=ON \
	-DBUILD_SHARED_LIBS=ON \
	-Dprotobuf_ABSL_PROVIDER=module \
#	-Dabsl_DIR=$(STAGING_DIR)/usr/lib/cmake/absl

TARGET_LDFLAGS += -latomic
TARGET_CFLAGS += -fPIC

#define Host/Install
#	$(INSTALL_DIR) $(STAGING_DIR_HOSTPKG)/bin
#	$(CP) $(HOST_BUILD_DIR)/build/protoc* $(1)/bin/
#	$(INSTALL_DIR) $(STAGING_DIR_HOSTPKG)/lib
#	$(CP) $(HOST_BUILD_DIR)/build/lib* $(1)/lib/
#	$(CP) $(HOST_BUILD_DIR)/build//cmake $(1)/lib
#endef

define Build/InstallDev
	$(call Build/InstallDev/cmake,$(1))
	$(SED) 's,/usr/include,$$$${prefix}/include,g' $(1)/usr/lib/pkgconfig/protobuf.pc
	$(SED) 's,/usr/lib,$$$${exec_prefix}/lib,g' $(1)/usr/lib/pkgconfig/protobuf.pc
	$(SED) 's,/usr/include,$$$${prefix}/include,g' $(1)/usr/lib/pkgconfig/protobuf-lite.pc
	$(SED) 's,/usr/lib,$$$${exec_prefix}/lib,g' $(1)/usr/lib/pkgconfig/protobuf-lite.pc
	$(INSTALL_DIR) $(1)/usr/lib $(1)/usr/include
	$(CP) $(PKG_INSTALL_DIR)/usr/include/* $(1)/usr/include/
	$(CP) $(PKG_INSTALL_DIR)/usr/lib/* $(1)/usr/lib/
	$(CP) $(PKG_INSTALL_DIR)/usr/lib/cmake $(1)/usr/lib
	$(CP) $(PKG_INSTALL_DIR)/usr/lib/pkgconfig $(1)/usr/lib/pkgconfig
endef

define Package/protobuf-lite/install
	$(INSTALL_DIR) $(1)/usr/lib
	$(CP) $(PKG_INSTALL_DIR)/usr/lib/libutf8_* $(1)/usr/lib/
	$(CP) $(PKG_INSTALL_DIR)/usr/lib/libprotobuf-lite.so* $(1)/usr/lib/
	$(CP) $(PKG_INSTALL_DIR)/usr/lib/libabsl_* $(1)/usr/lib
endef

define Package/protobuf/install
	$(INSTALL_DIR) $(1)/usr/lib
	$(CP) $(PKG_INSTALL_DIR)/usr/lib/libprotoc.so* $(1)/usr/lib/
	$(CP) $(PKG_INSTALL_DIR)/usr/lib/libprotobuf.so* $(1)/usr/lib/
endef

$(eval $(call BuildPackage,protobuf))
$(eval $(call BuildPackage,protobuf-lite))
$(eval $(call HostBuild,protobuf))
```

Node 22.11.0 will able to compile with the following patch : 
```
diff -Naur a/tools/v8_gypfiles/v8.gyp b/tools/v8_gypfiles/v8.gyp
--- a/tools/v8_gypfiles/v8.gyp
+++ b/tools/v8_gypfiles/v8.gyp
@@ -1026,6 +1026,17 @@
           '<(SHARED_INTERMEDIATE_DIR)',
         ],
       },
+      'link_settings': {
+         'libraries': [
+              '-labsl_hash',
+              '-labsl_city',
+              '-labsl_flags_reflection',
+              '-labsl_raw_hash_set',
+              '-labsl_low_level_hash',
+              '-labsl_throw_delegate'
+            ],
+          'library_dirs':[ '../../../../staging_dir/hostpkg/lib'],
+        },
       'sources': [
         '<(generate_bytecode_builtins_list_output)',
 
```

but will segmentation fault when running npm or compiling node-yarn : 
```
/home/user/works/openwrt/staging_dir/hostpkg/bin/npm i -g --production /home/user/works/openwrt/build_dir/hostpkg/node-yarn-1.22.22

Segmentation fault
```

vortexilation, 2024-12-03:
Yes, of course, the problem is when compiling node-yarn, node/npm segmentatation fault.

Thanks for joining in this issue.

nxhack, 2024-12-03:
(It's just a guess...) Perhaps you should prepare separate host-build and target-build versions of the linking library.
You may also need to devise a way to build obj.host and obj.target.

nxhack, 2024-12-03:
If the target is not x86_64, it seems like you can clearly see where the error is.

vortexilation, 2024-12-03:
> (It's just a guess...) Perhaps you should prepare separate host-build and target-build versions of the linking library.
> You may also need to devise a way to build obj.host and obj.target.

Your suggestion is for protobuf?, can you elaborate more or give an example?.

> If the target is not x86_64, it seems like you can clearly see where the error is.

The target is x86_64, just like the build host.

vortexilation, 2024-12-19:
More proper fixes : 
```
diff -Naur a/tools/v8_gypfiles/v8.gyp b/tools/v8_gypfiles/v8.gyp
--- a/tools/v8_gypfiles/v8.gyp
+++ b/tools/v8_gypfiles/v8.gyp
@@ -1034,6 +1034,12 @@
         '<@(inspector_all_sources)',
       ],
       'conditions': [
+        ['OS != "aix" and OS != "os400" and OS != "mac" and OS != "ios"', {
+          'ldflags': [
+            '-Wl,--whole-archive',
+            '<(obj_dir)/tools/v8_gypfiles/<(STATIC_LIB_PREFIX)v8_abseil<(STATIC_LIB_SUFFIX)',
+          ],
+        }],
         ['v8_enable_snapshot_compression==1', {
           'sources': [
             '<!@pymod_do_main(GN-scraper "<(V8_ROOT)/BUILD.gn"  "\\"v8_base_without_compiler.*?v8_enable_snapshot_compression.*?sources \\+= ")',
@@ -2254,8 +2260,10 @@
         'ABSEIL_ROOT': '../../deps/v8/third_party/abseil-cpp',
       },
       'direct_dependent_settings': {
+        'include_dirs!': [ '<!@(echo "$STAGING_DIR"/usr/include)' ],
         'include_dirs': [
           '<(ABSEIL_ROOT)',
+          '<!@(echo "$STAGING_DIR"/usr/../usr/include)',
         ],
       },
       'include_dirs': [
```

Now it doesn't collide anymore with other compiled abseil-cpp.

vidplace7, 2024-12-19:
@vortexilation 
```diff
+        ['OS != "aix" and OS != "os400" and OS != "mac" and OS != "ios"', {
+          'ldflags': [
+            '-Wl,--whole-archive',
+            '<(obj_dir)/tools/v8_gypfiles/<(STATIC_LIB_PREFIX)v8_abseil<(STATIC_LIB_SUFFIX)',
+          ],
+        }],
```
This part didn't seem to be necessary in my testing. Submitted a slightly modified patch-change as #25582 :+1: for @nxhack's review.

closed 2024-12-24 by commodo