openwrt/packages #25487: <mwan3> MWAN3 does not detect connection failure when receives ICMP destination unreachable.

palonso73, 2024-12-04:
We have notices that MWAN3 does not detect that one of the WAN connection is failing if it receives ICMP destination unreachable, it considers that the ICMP ping is working and the status of the interface is considered UP and running.

Is there any way to avoid this behaviour?

We have another router with a cellular connection connected to the WAN interface.

brada4, 2024-12-05:
Who is we? 
What behaviour are you refering to? ICMP type 3 ois generated by most of internet like on closed udp port etc.

palonso73, 2024-12-05:
Sorry, WE is I.

I have another router with a cellular connection connected to the WAN, the other WAN is the WWAN interface of the own router. It has a 5G cellular module from Quectel.

To simulate the loss of connection through the WAN, I set the cellular interface of the second router to shutdown state. Then the pings no longer reach the 8.8.8.8 destination, but the second router sends ICMP unreachable messages through the WAN. MWAN3 looks to understand this packets as ICMP responses and the WAN interface, the one with lower metric is shown as having connection to 8.8.8.8 when it is not real.