openwrt/packages #25491: iperf3: segmentation fault on mediatek/filogic

jkugler, 2024-12-04:
Maintainer: @\<github-user> (find it by checking history of the package Makefile)
Environment: aarch64, GL.iNet GL-MT6000, OpenWrt 23.05.5 r24106-10cc5fcd00

Description:

Running `iperf3 iperf3 -4 -c -P 1 speedtest.anc.acsalaska.net` generates a segmentation fault, with any value of `-P`. Removing `-P` results in a successful run.

```
root@router:~# iperf3 -4 -c -P 1 speedtest.anc.acsalaska.net
Segmentation fault
root@router:~# iperf3 -4 -c speedtest.anc.acsalaska.net
Connecting to host speedtest.anc.acsalaska.net, port 5201
[  5] local 63.140.92.173 port 37936 connected to 216.67.109.97 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec   115 MBytes   963 Mbits/sec   68    986 KBytes       
[  5]   1.00-2.00   sec   116 MBytes   975 Mbits/sec    0   1.05 MBytes       
[  5]   2.00-3.00   sec   116 MBytes   975 Mbits/sec    0   1.13 MBytes       
[  5]   3.00-4.00   sec   116 MBytes   977 Mbits/sec    0   1.20 MBytes       
[  5]   4.00-5.00   sec   117 MBytes   980 Mbits/sec    0   1.27 MBytes       
[  5]   5.00-6.00   sec   117 MBytes   979 Mbits/sec    0   1.34 MBytes       
[  5]   6.00-7.00   sec   116 MBytes   976 Mbits/sec    0   1.40 MBytes       
[  5]   7.00-8.00   sec   116 MBytes   973 Mbits/sec    0   1.46 MBytes       
[  5]   8.00-9.00   sec   117 MBytes   979 Mbits/sec    0   1.51 MBytes       
[  5]   9.00-10.00  sec   117 MBytes   980 Mbits/sec    0   1.57 MBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec  1.14 GBytes   976 Mbits/sec   68             sender
[  5]   0.00-10.02  sec  1.13 GBytes   973 Mbits/sec                  receiver

iperf Done.
```

neheb, 2024-12-05:
Backported fix. Update should arrive in a day.

closed 2024-12-05 by neheb

jkugler, 2024-12-05:
Excellent, thank you!