openwrt/packages #25520: banip: writing to the filesystem needlessly

andrew-2e128, 2024-12-09:
Maintainer: @dibdot
Environment: ramips/mt7621, TP-Link Archer C6 v3, 23.05.5

Description:

I was using inotifywait to watch for file writes (modify) to /overlay/ (mount point of my router's flash)
here is the exact command I used:
```
inotifywait --monitor -e modify,close_write,create,delete,delete_self -r /overlay/
```
and I got this output
```
/overlay/upper/etc/banip/ MODIFY banip.blocklist
/overlay/upper/etc/banip/ MODIFY banip.blocklist
```
the files are empty...

also
everytime at boot/restart it does:
```
/overlay/upper/etc/config/ CREATE .banip.uci-<random 6 chars>
```

note, I do have luci (and banip-luci) installed (it's behavior is strange, it does not have a save and apply buttons like the rest of luci apps and default services btw)

here is the config
```
config banip 'global'
	option ban_enabled '1'
	option ban_debug '0'
	option ban_autodetect '1'
	list ban_logterm 'Exit before auth from'
	list ban_logterm 'luci: failed login'
	list ban_logterm 'error: maximum authentication attempts exceeded'
	list ban_logterm 'sshd.*Connection closed by.*\[preauth\]'
	list ban_logterm 'SecurityEvent=\"InvalidAccountID\".*RemoteAddress='
	list ban_logterm 'received a suspicious remote IP '\''.*'\'''
	option ban_deduplicate '1'
	option ban_loginput '1'
	option ban_logforwardwan '1'
	option ban_logforwardlan '0'
	list ban_feed 'feodo'
	list ban_feed 'sslbl'
	list ban_feed 'threat'
	list ban_feed 'threatview'
	option ban_autoallowlist '0'
	option ban_autoblocklist '0'
	option ban_allowlistonly '0'
	option ban_nftpolicy 'memory'
	list ban_trigger 'pppoe'
	option ban_fetchcmd 'uclient-fetch'
	option ban_protov4 '1'
	list ban_ifv4 'pppoe'
	option ban_fetchretry '5'
	option ban_nicelimit '0'
	option ban_filelimit '1024'
	option ban_nftpriority '-100'
	option ban_icmplimit '10'
	option ban_synlimit '10'
	option ban_udplimit '100'
	option ban_blocktype 'drop'
	option ban_nftloglevel 'warn'
	option ban_logprerouting '0'
	option ban_loglimit '100'
	list ban_dev 'pppoe-pppoe'
```

andrew-2e128, 2024-12-09:
just updated to check with the latest version
```
/overlay/upper/etc/config/ CREATE .banip.uci-<random 6 chars>
```
no longer happens, but

```
/overlay/upper/etc/banip/ MODIFY banip.blocklist
/overlay/upper/etc/banip/ MODIFY banip.blocklist
```
still happens

dibdot, 2024-12-09:
Not reproducible. When and how often does this happen?

andrew-2e128, 2024-12-09:
every time it updates or reloads

andrew-2e128, 2024-12-09:
I can retest it in the next openwrt release (24.10) with a fresh build/install

andrew-2e128, 2024-12-12:
@dibdot I apologize for not working on making it reproducible on other systems, but I figured out the issue,
basically for whatever reason on line 956 and 971 it writes ${tmp_raw} to ${ban_blocklist} even if it is empty, I changed both lines to
```sh
if [ -s "${tmp_raw}" ]; then
	"${ban_catcmd}" "${tmp_raw}" 2>/dev/null >"${ban_blocklist}"
fi
```
and it stopped writing the file;

I am still debugging the issue myself, just wanted to give you a notice on what I found

andrew-2e128, 2024-12-12:
I assume the enter if statement is just for de-duplication of the file `${ban_blocklist}` then maybe in this case it should be just skipped entirely if the file `${ban_blocklist}` is empty instead?

dibdot, 2024-12-12:
@andrew-2e128 Thanks, that explains why I can't reproduce the issue with my forthcoming dev version, cause I've already removed this section ... ;-) So this will be fixed with the next release.

Thanks again!

dibdot, 2025-01-17:
fixed with this release: https://github.com/openwrt/packages/commit/80768dfdae3e0db273cf2a69f7911da44102f8c2 (will be backported to 24.10 in the next couple of weeks).

closed 2025-01-17 by dibdot