openwrt/packages #25522: https_dns_proxy - consider increasing max number of curl concurent connections in https_client.c

Mile-Lile, 2024-12-09:
# Issue template (remove lines from top till here)

Maintainer: @stangri 
Environment: (mediatek/filogic, Xiaomi ax3000t, OpenWrt latest SNAPSHOT)

Description:
I saw this error in my logs:

```
[F] 1733741993.412721 https_client.c:601 curl needed more event, than max connections: 8
```

stangri, 2024-12-09:
When does this happen? How often does this happen?

krant, 2024-12-09:
> Sounds like a curl restriction

It is not - https://github.com/aarond10/https_dns_proxy/blob/484bd153bb85a51df1c5bede1b091be76537e0a7/src/https_client.h#L9

Mile-Lile, 2024-12-09:
I installed it with apk.... version is Version 2023.12.26-1  and looks like this:

[image]

on snapshot I installed this morning version:

[image]

Mile-Lile, 2024-12-09:
it does not happen often... once today

[image]

vincejv, 2024-12-11:
can we make this configurable instead?? There are some connections/internet connections that limits the number of outgoing connections especially those on CGNAT setups

stangri, 2024-12-11:
@Mile-Lile could you please post the output of:
```sh
curl -V
cat /etc/config/https-dns-proxy
```

@vincejv -- it's a compilation-time setting.

Mile-Lile, 2024-12-12:
here it is. Sorry for late response, I don't have remote access to the unit....

`root@AX3000T:~# curl -V
curl 8.10.1 (aarch64-openwrt-linux-gnu) libcurl/8.10.1 mbedTLS/3.6.2 nghttp2/1.63.0
Release-Date: 2024-09-18
Protocols: file ftp ftps http https ipfs ipns mqtt
Features: alt-svc HSTS HTTP2 HTTPS-proxy IPv6 Largefile SSL threadsafe UnixSockets

root@AX3000T:~# cat /etc/config/https-dns-proxy

config main 'config'
        option canary_domains_icloud '1'
        option canary_domains_mozilla '1'
        option dnsmasq_config_update '*'
        option force_dns '1'
        list force_dns_port '53'
        list force_dns_port '853'
        option procd_trigger_wan6 '0'

config https-dns-proxy
        option resolver_url 'https://dns11.quad9.net/dns-query'
        option bootstrap_dns '9.9.9.9`

baranyaib90, 2024-12-29:
Hi, the issue happens, when a large burst of DNS querries (more than 8 request) came into the proxy after a long time of no querries at all (in my logs I see 342 seconds idle time).
If someone needs a **work around**: send a querry to the proxy every 1 minute. That will keep the HTTP/2 connection alive.

closed 2025-04-14 by stangri