openwrt/packages #25552: Nut_server wrong driver option

philippe9999, 2024-12-14:
Maintainer: common packages maintainers
Environment: ARMv8 Processor rev 0, Bananapi BPI-R4, OpenWrt 24.10.0-rc2 r28161-ea17e958b9

Description:

Use luci nut panel to create /etc/config/nut_server file.  Try to start nut-server without success.

After some investigation, found that the driver option in config file nut_server was not properly written.

Luci give this driver option  /lib/nut/usbhid-ups and send this to the file. 

After change the config file manualy from 
option driver '/lib/nut/usbhid-ups' 
to
option driver 'usbhid-ups' 
erverything  work as expected.

Salutations.

systemcrash, 2025-01-09:
We're fixing this on the luci side. Thank you for the report @philippe9999

systemcrash, 2025-01-09:
Fixed (in luci)