openwrt/packages #25555: dropbear: Consider using option 'DirectInterface' to bind directly to interface

Mile-Lile, 2024-12-15:
Maintainer: @Rondom 
Environment: (OpenWrt SNAPSHOT r28327-502916468e / LuCI Master 24.346.53951~31955d6)

Description:

When I go to System/Administration/SSH Access and choose dropbear to listen to only my lan interface I get this message:

```
dropbear Option 'Interface' binds to address(es) but not to interface
dropbear Consider using option 'DirectInterface' to bind directly to interface
```

So, why don`t we use DirectInterface option or just bind it to local IP of the router instead?

Here is my config:

[image]

Mile-Lile, 2024-12-15:
OK. I'll add @systemcrash in the ticket... I think he has most commits in the luci section...

systemcrash, 2024-12-15:
Is it correct that `DirectInterface` is a dropbear only property? I don't know how sshd reacts to that property present, but it may be possible to add it, given that the luci ssh handles `/etc/dropbear/` anyway. The caution here is being highly dependent on specific options and losing compatibility with sshd.

The `dropbear.init` script seems to contradict itself, however:
https://github.com/openwrt/openwrt/blob/5a24850de1418510c421d45eb2436bdcb2b243c6/package/network/services/dropbear/files/dropbear.init#L195-L213

Both take an interface?

Which is in the openwrt repo, and not packages repo.

See also https://github.com/openwrt/openwrt/issues/9666

Mile-Lile, 2024-12-16:
I tried to google it but there are so many  sources... it can be compiled in busybox, [linux man page](https://linux.die.net/man/8/dropbear) doesn't know about `listen to specific interface` option. On the other [other variation of dropbear](https://www.mankier.com/8/dropbear) has `-l interface
 Listen on the specified interface` probably when you run it from command line... [official page fo dropbear](https://matt.ucc.asn.au/dropbear/dropbear.html) doesn't have man page... 
when I saw option in luci I thought it works like dnsmasq... dnsmasq has option to listen to specific interface or subnet... or to restrict service to specific interface (usefull to e.g. guest interface)... I  don't know... seems this option in luci is not what I think is for...

stangri, 2024-12-16:
I've noticed some inconsistencies within the init script when using `DirectInterface` instead of `Interface`:
1. When using `Interface` on the non-suitable (non-netifd for example) interface, I do get a warning in the console: `dropbear: Network interface '***' has no suitable IP address(es)!`, however when switching option name to `DirectInterface`, the warning is no longer in the console, but only in the log: `10:53:57 dropbear: Network interface '***' is not available!`. If the instance cannot be started, IMHO the warning should be displayed in the console, not just in the log.
2. This might be the user error, but I figured I'd mention it in case it isn't: I use the `RemoteForward` directive in the ssh config file on the client (so I can edit router files with the text editor on my desktop OS) and with the `Interface` option in the dropbear config it works, with the `DirectInterface` option in the dropbear config, I get an error: 
  ```
  /usr/bin/rmate: connect: Network unreachable
  /usr/bin/rmate: line 412: /dev/tcp/localhost/52698: Network unreachable
  Unable to connect to TextMate on localhost:52698
  ```