openwrt/packages #25575: adblock - fetarure proposal - time based filter of blocked domains

Mile-Lile, 2024-12-18:
@dibdot 
I know it is to much to ask but I'll still try ask for this option... time based filter for blocked domains...

in some softwares it is called `family time` and during this hours filter block some websites because it is time for family:) on other softwares this feature is called `happy hours` and some websites are allowed only during these hours (e.g. you can play browlstars.com only from 18:00 till 20:00 on friday)... There is app luci-app-wifischedule and maybe this luci logic can be borowed from there?

Thx if you consider this FR...

dibdot, 2024-12-18:
Time based functions are already available via cron. Just cconfigure suspend/resume of adblock in the cron scheduler.

Mile-Lile, 2024-12-18:
Ok. Thx...

closed 2024-12-18 by Mile-Lile

Mile-Lile, 2024-12-19:
your suggestion will susped whole adblock service not just blocked domains... it would be preferably if filetring of ads to be active all the time but still have option of `family time`...