openwrt/packages #25578: prometheus-node-exporter-lua: `tonumber` of empty string causing error and dropped metrics

shikasta-net, 2024-12-19:
Maintainer: @champtar
Environment: RTL8382, D-Link DGS-1210-16, Openwrt 23.05.4; and RTL8380, D-Link DGS-1210-10MP F, Openwrt 23.05.5

Description:
Log noise and missing metrics caused at:
https://github.com/openwrt/packages/blob/bbc02fa8685b6fc4f8c3174ce49c29aca474b6aa/utils/prometheus-node-exporter-lua/files/usr/lib/lua/prometheus-collectors/netclass.lua#L36-L38
```if speed ~= nil and tonumber(speed) >= 0 then``` causes
> /usr/lib/lua/prometheus-collectors/netclass.lua:36: attempt to compare number with nil

when `speed` is an empty string, because `tonumber('') = nil` and `nil >= 0` is invalid.  Is only an issue on these d-link switches, not linksys wrt3200 router or BT homehub5 accesspoint.

shikasta-net, 2024-12-19:
I tried the obvious thing of adding `and speed ~= ''` to the conditional but that fails with
>/usr/bin/prometheus-node-exporter-lua:42: bad argument #4 to 'format' (string expected, got nil)

sanist1, 2025-03-02:
I used 

and #speed > 0 

which at the very stopped the infinite error logs.

canon-cmre-kym-eden, 2025-03-03:
Following helpful comment from @sanist1 (which I now can't see) I needed to add `and #speed > 0` to both the speed and the file metrics parts of the `get_metric` function.
/usr/lib/lua/prometheus-collectors/netclass.lua:36-38
```
  if speed ~= nil and #speed > 0 and tonumber(speed) >= 0 then
    metric_node_network.speed_bytes({device = device}, tonumber(speed)*1000*1000/8)
  end
```
/usr/lib/lua/prometheus-collectors/netclass.lua:59-61
```
    if value ~= nil and #value > 0 then
      metric_node_network[metric]({device = device}, tonumber(value))
    end
```

shikasta-net, 2025-03-03:
Following helpful comment from @sanist1 (which I now can't see) I needed to add `and #speed > 0` to both the speed and the file metrics parts of the `get_metric` function.
/usr/lib/lua/prometheus-collectors/netclass.lua:36-38
```
  if speed ~= nil and #speed > 0 and tonumber(speed) >= 0 then
    metric_node_network.speed_bytes({device = device}, tonumber(speed)*1000*1000/8)
  end
```
/usr/lib/lua/prometheus-collectors/netclass.lua:59-61
```
    if value ~= nil and #value > 0 then
      metric_node_network[metric]({device = device}, tonumber(value))
    end
```

sanist1, 2025-03-07:
I accidentally deleted the comment, thanks for picking up on it! I have applied your changes and it seems to fix the error.