openwrt/packages #25588: atlas-sw-probe is heavily broken

LegendaryYbarro2005, 2024-12-21:
Maintainer: @ja-pa  and @BKPepe
Environment: All
Software: OpenWrt 24.10.0-rc2 r28161-ea17e958b9 and SNAPSHOT (as of 2024-12-21)

Description:
I installed the packages `atlas-sw-probe`. Then setup via instruction in [manual guide](https://parkercs.tech/running-a-ripe-atlas-probe-on-openwrt). That worked well with OpenWrt 23.05 and 22.03. OpenWrt 24.10.0-rc2 (as well in snapshots releases) on x86_64 VM gave me the error while creating the key:

```
root@OpenWrt:~# /etc/init.d/atlas create_key
Unknown argument -b
Usage: ssh-keygen -t <type> -f <filename> [-s bits]
-t type Type of key to generate. One of:
                rsa
                ed25519
-f filename    Use filename for the secret key.
               ~/.ssh/id_dropbear is recommended for client keys.
-s bits Key size in bits, should be a multiple of 8 (optional)
           Ed25519 has a fixed size of 256 bits
-y              Just print the publickey and fingerprint for the
                private key in <filename>.
-C              Specify the key comment (email).
sed: /etc/atlas/probe_key.pub: No such file or directory
ln: /usr/libexec/atlas-probe-scripts/etc/probe_key: File exists
ln: /usr/libexec/atlas-probe-scripts/etc/probe_key.pub: File exists
chown: /etc/atlas/probe_key: No such file or directory
chown: /etc/atlas/probe_key.pub: No such file or directory
chgrp: /etc/atlas/probe_key: No such file or directory
chgrp: /etc/atlas/probe_key.pub: No such file or directory
chmod: /etc/atlas/probe_key: No such file or directory
chmod: /etc/atlas/probe_key.pub: No such file or directory
Key generated successfully. Use the get_key command to show the public key and get instruction on how to register your probe.
root@OpenWrt:~#
```

Running /etc/init.d/atlas get_key to print the key in the terminal and doesn't work:

```
root@OpenWrt:~# /etc/init.d/atlas get_key
Error! Pub. key not found
root@OpenWrt:~#
```

The `atlas-sw-probe` package is out of date. I have tried to update it but unfortunately, I don't have enough knowledge to do it myself, and as i'm trying to update the package, it needs the latest version.

Not sure this is the right place to post this issue, please redirect me on the right place.

Thanks!

LegendaryYbarro2005, 2025-02-18:
Thanks for reporting us. I'm running into this issue with my x86_64 on 24.04.0 so it still seems to be a problem in the latest version. Switching to `openssh-keygen` solution works!

P.S. I am wondering if this is still an issue in the latest OpenWrt release.

So we can hopefully get it working on 24.xx? (since 24.10.0 is out now)

Immanuel31, 2025-02-23:
Yes, it's still in issue in OpenWrt 24.10.0 - installing [`openssh-keygen`](https://openwrt.org/packages/pkgdata/openssh-keygen) seems to be the quickest workaround/fix:
```
opkg update
opkg install openssh-keygen
```

phillipremaker, 2025-11-07:
You MUST openssh-keygen for the script to work, or modify the dropbear generated private key to the right format.

See, it FINALLY  worked once I installed openssh-keygen, which should be a dependency

(Dropbear ssh-keygen)
```
11/07/25 17:58:32 net-ok
11/07/25 17:58:32 reg-init
11/07/25 17:58:34 reginit-fail
11/07/25 18:01:37 net-try
11/07/25 18:01:40 net-ok
11/07/25 18:01:41 reg-init
11/07/25 18:01:43 reginit-fail
```
 

(OpenSSH ssh-Keygen)
```
11/07/25 18:04:46 net-try
11/07/25 18:04:49 net-ok
11/07/25 18:04:49 reg-init
11/07/25 18:04:52 ctrl-init
11/07/25 18:04:55 keep-start
```

For the `/etc/atlas/probe_key` file, drobrear writes a binary file and OpenSSH writes a textfile starting iwth
`-----BEGIN OPENSSH PRIVATE KEY-----`

The probe software cannot use the dropbear key.