openwrt/packages #25589: perl: userelocatableinc causes Alpine builds to fail

egorksv, 2024-12-21:
# Issue template (remove lines from top till here)

Maintainer: @Naoir 
Environment: Alpine Linux 3.18, OpenWRT 23.05.5

Description:
During cross-compilation of perl module on Alpine specifically, execution of miniperl fails, as been noted in multiple issues:
https://github.com/openwrt/packages/issues/10598

as an example:
```
make[3]: Entering directory '/root/openwrt-19.07/build_dir/target-mips_24kc_musl/perl/perl-5.28.1'

/bin/ln -s /root/openwrt-19.07/staging_dir/hostpkg/usr/bin/perl miniperl
LD_LIBRARY_PATH=/root/openwrt-19.07/build_dir/target-mips_24kc_musl/perl/perl-5.28.1 ./miniperl -w -Ilib -I. -Idist/Exporter/lib -MExporter -e '<?>' || sh -c 'echo >&2 Failed to build miniperl.  Please run make minitest; exit 1'

Can't locate File/Glob.pm in @INC (you may need to install the File::Glob module) (@INC contains: lib . dist/Exporter/lib ./../lib/perl5/site_perl/5.28.1/x86_64-linux ./../lib/perl5/site_perl/5.28.1 ./../lib/perl5/5.28.1/x86_64-linux ./../lib/perl5/5.28.1) at -e line 1.
```

This happens due to `miniperl` being called from perl source directory being cross-compiled, and `./../lib` resolves to `/root/openwrt-19.07/build_dir/target-mips_24kc_musl/perl/perl-5.28.1/./../lib` which at this point does not contain Glob.pm

Removing `-Duserelocatableinc` from Makefile seems to solve the issue

egorksv, 2024-12-21:
Also https://github.com/openwrt/packages/issues/6700 - as that needs to be resolved first, unfortunately am unable to reopen that one

Ts1ng, 2025-03-08:
I use almalinux build my openwrt bin. I have encountered this issue as well.

Ts1ng, 2025-03-08:
I have encountered this issue as well.I have encountered this issue as well.

Ts1ng, 2025-03-08:
I have encountered this issue as well and I built my openwrt in Almalinux. I found perl's configure script would using "ls -l 
 /proc/self/exe" check dos "/proc/self/exe" command exist.In Almalinux /proc/self/exe provides by coreutils-single.The check conditions of the configure script were not met by doing “ls -l /proc/self/exe” result. And I replace Almalinux's coreutils-single package by coreutils package. Finally the problem I met is solved. Hope it will help you.