openwrt/packages #25605: samba4, various error/warning messages

peterb99, 2024-12-24:
# Issue template (remove lines from top till here)

Maintainer: @\<github-user> (find it by checking history of the package Makefile)
Environment: (put here arch, model, OpenWrt version)

Description: Login via SONOS audio system needs extra parameters in template
Please add following lines to the default template under global
```
## Note next two setting need to be enabled for Sonos system to be able to login samba!!
##        server min protocol = NT1
##        ntlm auth = yes
```

The following error messages fill up the log files

in log.smbd
[2024/12/24 10:03:41.064876,  1] ../../source3/smbd/smb1_reply.c:1902(reply_ulogoffX)
  reply_ulogoffX: ulogoff, vuser id 36096 does not map to user.
[2024/12/24 10:04:14.767137,  1] ../../source3/smbd/smb1_reply.c:1902(reply_ulogoffX)
  reply_ulogoffX: ulogoff, vuser id 54832 does not map to user.
[2024/12/24 10:04:47.576173,  1] ../../source3/smbd/smb1_reply.c:1902(reply_ulogoffX)
  reply_ulogoffX: ulogoff, vuser id 41707 does not map to user.
[2024/12/24 10:07:02.553629,  1] ../../source3/smbd/smb1_reply.c:1902(reply_ulogoffX)
  reply_ulogoffX: ulogoff, vuser id 53559 does not map to user.
[2024/12/24 10:07:19.102549,  1] ../../source3/smbd/smb1_reply.c:1902(reply_ulogoffX)
  reply_ulogoffX: ulogoff, vuser id 9961 does not map to user.

tail log.rpcd_classic
  lpcfg_do_global_parameter: WARNING: The "null passwords" option is deprecated
[2024/12/23 23:24:44,  0] ../../source3/rpc_server/rpc_worker.c:1125(rpc_worker_main)
  rpcd_classic version 4.18.8 started.
  Copyright Andrew Tridgell and the Samba Team 1992-2023
[2024/12/23 23:24:44,  1] ../../lib/param/loadparm.c:1909(lpcfg_do_global_parameter)
  lpcfg_do_global_parameter: WARNING: The "null passwords" option is deprecated
[2024/12/23 23:24:44,  1] ../../lib/param/loadparm.c:1909(lpcfg_do_global_parameter)

Params used:
/var/log# testparm  /etc/samba/smb.conf
Load smb config files from /etc/samba/smb.conf
lpcfg_do_global_parameter: WARNING: The "null passwords" option is deprecated
Loaded services file OK.
Weak crypto is allowed by GnuTLS (e.g. NTLM as a compatibility fallback)

Server role: ROLE_STANDALONE

Press enter to see a dump of your service definitions

# Global parameters
[global]
	bind interfaces only = Yes
	deadtime = 15
	disable spoolss = Yes
	domain master = Yes
	enable core files = No
	interfaces = br-lan
	load printers = No
	map to guest = Bad User
	mdns name = mdns
	ntlm auth = ntlmv1-permitted
	null passwords = Yes
	os level = 99
	passdb backend = smbpasswd
	preferred master = Yes
	printcap name = /dev/null
	security = USER
	server min protocol = NT1
	server string = Samba on OpenWRT
	socket options = IPTOS_LOWDELAY TCP_NODELAY
	wins support = Yes
	workgroup = PBHOME
	idmap config * : backend = tdb
	delete veto files = Yes
	invalid users = root
	veto files = /Thumbs.db/.DS_Store/._.DS_Store/.apdisk/

[MyMusic]
	create mask = 0666
	directory mask = 0777
	path = /mnt/nvme0n1/MyMusic
	read only = No
	vfs objects = catia fruit streams_xattr io_uring
	fruit:veto_appledouble = no
	fruit:metadata = stream
	fruit:encoding = native

```
Format code blocks by wrapping them with pairs of ```
```

peterb99, 2024-12-24:
Also remove the following line from the template
'''
## allow client access to accounts that have null passwords. 
	## null passwords = yes
'''
as this option is not supported anymore

Neustradamus, 2024-12-31:
@peterb99: Have you tested ksmbd?

peterb99, 2024-12-31:
should I?

Neustradamus, 2024-12-31:
It replaces Samba, but your ticket is not bad, it must be solved too :)