openwrt/packages #25616: pbr: starts before VPN interface is up

mikhirev, 2024-12-25:
Maintainer: @stangri 
Environment: ath79/nand/netgear_wndr3700-v4 OpenWrt 24.10.0-rc4 (r28211-d55754ce0d)

Description:

After reboot routing to the `wireguard` interface does not work correctly. I found the following line in system log:

```
Thu Dec 26 03:04:18 2024 user.notice pbr [2621]: Setting up routing for 'wireguard/0.0.0.0' [✗]
```

After I restart pbr, everything works fine. In the log I see:

```
Thu Dec 26 03:07:53 2024 user.notice pbr [9955]: Setting up routing for 'wireguard/10.11.0.2' [✓]
```

But after reboot the problem reproduces. Seems that pbr starts too early, before the `wireguard` interface is set up.

mikhirev, 2024-12-26:
Well, my assumption that pbr should wait for interfaces was incorrect. It seems that interface triggers don't work as expected. I see the following trigger registered:

```
                        [
                                "interface.*",
                                [
                                        "if",
                                        [
                                                "eq",
                                                "interface",
                                                "wireguard"
                                        ],
                                        [
                                                "run_script",
                                                "/etc/init.d/pbr",
                                                "on_interface_reload",
                                                "wireguard"
                                        ]
                                ]
                        ]
```

However it does not work properly at startup.

mikhirev, 2024-12-26:
If I run `/etc/init.d/pbr on_interface_reload wireguard` in command line, it is reloaded successfully. So my guess is that the trigger does not take action because the `wireguard` interface gets up while `pbr` is starting (it takes more than 30 secs and there is a line `firewall: Reloading firewall due to ifup of wireguard (wireguard)` between messages about setting up routing from `pbr` in log).

mikhirev, 2024-12-26:
As I see, when starting with `on_boot`, `pbr` waits for the default gateway interface but does not wait for other interfaces.

mikhirev, 2024-12-29:
As a workaroud, I did

```
uci set pbr.@pbr[0].procd_boot_delay=30
uci commit
```

However everything should work out-of-the-box.

sinancetinkaya, 2025-01-08:
> As a workaroud, I did
> 
> ```
> uci set pbr.@pbr[0].procd_boot_delay=30
> uci commit
> ```
> 
> However everything should work out-of-the-box.

I was having the same problem, thank you!

stangri, 2025-01-09:
> Seems that pbr starts too early, before the `wireguard` interface is set up.

Depends on a specific use case (hardware and uplink connection). README has information on various options which could be used if your uplink connection is slow to set up.

closed 2025-01-09 by stangri

mikhirev, 2025-01-09:
> Depends on a specific use case

Yes. But it shouldn't depend. There's quite a long time while pbr is starting when triggers are not set up and changes in the interface status don't affect routing. It's a bug, however I'm unsure if it should be fixed in pbr itself or in init scripts that should set up triggers earlier.

BTW why pbr startup takes so long time? More than 30 seconds in my case.

stangri, 2025-01-10:
> > Depends on a specific use case
> 
> Yes. But it shouldn't depend. 

Yes, I'm sure when there's a world peace, pbr boot timing wouldn't depend on slow-starting wans.

> It's a bug, however I'm unsure if it should be fixed in pbr itself or in init scripts that should set up triggers earlier.

If you feel so strongly about it, I hope it inspires you to implement a working solution.

> BTW why pbr startup takes so long time? More than 30 seconds in my case.

Depends on a specific use case.

sinancetinkaya, 2025-01-10:
> > > Depends on a specific use case
> > 
> > 
> > Yes. But it shouldn't depend.
> 
> Yes, I'm sure when there's a world peace, pbr boot timing wouldn't depend on slow-starting wans.
> 
🙂
> > It's a bug, however I'm unsure if it should be fixed in pbr itself or in init scripts that should set up triggers earlier.
> 
> If you feel so strongly about it, I hope it inspires you to implement a working solution.
>
It's not a bug, it's a feature 🙂
> > BTW why pbr startup takes so long time? More than 30 seconds in my case.
> 
> Depends on a specific use case.

Can you explain? My be we can avoid it 🤔

mikhirev, 2025-01-10:
> If you feel so strongly about it, I hope it inspires you to implement a working solution.

If I were, I would do that. But I'd like to know your opinion on how to do that better, at least. Something other than "that's your problem, I don't care".

Well, I understand that you may not want to waste your time on this. That's your right. What I don't understand is why you closed this issue as completed? There is a real problem that is not solved.

stangri, 2025-01-12:
> What I don't understand is why you closed this issue as completed? There is a real problem that is not solved.

Solved by perusing README and setting `procd_boot_delay` to an appropriate value.

mikhirev, 2025-01-13:
Not solved, worked around. Please keep this issue open until the proper solution will be found.

omnicron-llc, 2025-01-19:
@stangri  An ugly hack for a non-working component is not a solution and attempting to lampshade it is a sign of flagrant negligence and/or incompetence.

cardboardaardvark, 2025-01-19:
This issue should not be closed. This is obviously a race condition and the notion of accepting a known race condition as being ok is not really good for a project that should be considered production quality.

As well I came across this problem because it showed up for me after an upgrade from 22.03 to 23.05. I kept the original configs as a part of the upgrade. Everything was fine before the upgrade. I get the exact problem here after the upgrade.

The issue should remain in the open state until the race condition is solved.

stangri, 2025-01-20:
> Not solved, worked around. Please keep this issue open until the proper solution will be found.

Workaround would have been needing to use rc.local to schedule restart, giving a boot delay option/code is a solution here.

> But I'd like to know your opinion on how to do that better, at least.

If you feel like experimenting to test your theory see if you can use an external script in the trigger (instead of calling /etc/init.d/pbr on interface reload) which would set a flag and attempt to call pbr on_interface_reload directly. You'd need to add a code to pbr to check/reset the flag on start and at the end of loading to call service start again if the flag was set halfway thru. Alternatively, if you can find any procd documentation with suggestions on how to address this, I'd be all ears.

cardboardaardvark, 2025-01-20:
@stangri I do not understand the rational here for closing the ticket as resolved. Can you please let me know why it is seemingly more important to optimize for a reduced set of open bug reports than it is to leave bug reports open so interested parties can find things to improve?

> if you can find any procd documentation with suggestions on how to address this

You seem to understand that the behavior is at least not optimal. I would say the behavior is correctly characterized as a malfunction especially in light of my report where the problem showed up for me with an upgrade from 22 to 23.

This is a race condition. It's not just a race condition it looks like a new race condition introduced in version 23 of OpenWRT. Race conditions that do not need to exist, ie a race condition that can be solved with out an arbitrary delay introduced, are traditionally considered to be malfunctions. In fact, out of my entire 30 year career with computers and networking I can't think of a single instance where anyone has felt that arbitrary delays have ever fixed a race condition instead of working around it.

> Workaround would have been needing to use rc.local to schedule restart

The arbitrary delay, though available as an official configuration parameter, is not much better. Can it even bet set via LuCI?

I've seen you reference the PBR readme as a source for several different ways to handle an interface that is slow to come up but I'm not seeing that specific case covered at https://docs.openwrt.melmac.net/pbr/

I can assure you, having just gone through identifying this problem exists, that it is not obvious from LuCI that this is some kind of problem related to interface start ordering/timing. It's also not obvious in the least that the problem will exist during a normal setup process because the issue is only exposed after the first reboot.

Hypothetical retort: "You should always test the new system with a reboot"

This is good advice. But I also have (or had) some faith in OpenWRT doing the right thing and had learned to not expect bugs like this because it's treated me well over the last 5 years.

Is this behavior from PBR a surprise? Yes it is. Is it a bug? Yes it is.

> If you feel so strongly about it, I hope it inspires you to implement a working solution.

Well, ideally, yes, we would find a solution to this problem. Here again you seem to understand that there is something that needs to be resolved.

Slightly less ideally someone else may be interested in improving the situation. How would someone else who may be interested in improving the situation find this? By having the ticket be open.

If you are going to leave the ticket closed the more honest close reason would be "wont fix" instead of resolved.

stangri, 2025-01-21:
> But I also have (or had) some faith in OpenWRT doing the right thing and had learned to not expect bugs like this because it's treated me well over the last 5 years.

So do I, so I'm using OpenWrt-provided framework for registering for events and reacting to them. In addition to that, there's a provision to time the pbr start to avoid issues. 

> I've seen you reference the PBR readme as a source for several different ways to handle an interface that is slow to come up but I'm not seeing that specific case covered at https://docs.openwrt.melmac.net/pbr/

You're welcome to contribute to the docs, they are on github as well. My free unpaid time has already been spent this week responding to this issue.

cardboardaardvark, 2025-01-21:
@stangri You did not address many of the questions and concerns I raised.

> My free unpaid time has already been spent this week responding to this issue.

Do you understand that by closing this ticket that you are engaging in information destruction and the information that is being destroyed is information other people could use to address this problem so you don't have to? And that by leaving the ticket closed and trying to justify it that you are causing more of your time to be consumed than is needed? So, again:

> I do not understand the rational here for closing the ticket as resolved. Can you please let me know why it is seemingly more important to optimize for a reduced set of open bug reports than it is to leave bug reports open so interested parties can find things to improve?

Are you the author of PBR? If so, why are you seemingly uninterested in getting to the bottom of why the user experience with your software has obviously degraded with the version change from 22 to 23? It seems likely that either the usage of the event system is incorrect and always has been, needs to be updated to account for new changes, or the event system in OpenWRT introduced a bug that is impacting PBR.

If you are the author of PBR it becomes even more difficult to understand the motivation to keep others from discovering this open issue especially because it could lead to a reduction in your free unpaid time spent.

Much of your interaction in this thread leads me to believe you are getting frustrated with having to deal with this. That becomes even more of a reason to open the ticket again so your many calls for someone else to resolve the problem have a greater likelihood of actually come to fruition and your concern about your valuable free unpaid time being consumed can be addressed.

Just open the ticket back up. It's not only the correct action to take in terms of an open-source collaborative project you'd be doing yourself a favor too in that you won't have to keep telling other people to work on it (the ticket being open itself means someone needs to work on it) and keep explaining why the ticket should be closed.

No one here is even asking for you to spend more of your free unpaid time on this. We are asking for this issue to get the visibility that is expected for a problem impacting users of an opensource project.

mikhirev, 2025-01-21:
> If you feel like experimenting to test your theory see if you can use an external script in the trigger (instead of calling /etc/init.d/pbr on interface reload) which would set a flag and attempt to [...]

Thank you. However when I digged into startup scripts, I figured out that triggers are registered only after the service has started successfully. But I have to check one more time.

> I'm using OpenWrt-provided framework for registering for events and reacting to them.

Have you reported an issue in that framework? If no, where should I go to report it?

stangri, 2025-01-21:
> Thank you. However when I digged into startup scripts, I figured out that triggers are registered only after the service has started successfully. But I have to check one more time.

There's another way to set up triggers on boot which some other services use and I thought I had implemented it before (so my testing recommendations above assumed it was in use), maybe it was lost in post 1.1.6 transitions. I've restored it in 1.1.8-r5 available from my repo. Before or after installing it, but before rebooting do:

```
uci del pbr.config.procd_boot_delay
uci set pbr.config.procd_slow_start=1
uci commit pbr
```

This is a temporary variable for testing only, if this has the desired effect I'll remove the need for it before merging update in OpenWrt repos.

So on next reboot it will use the same style of on_boot trigger as dhcpd.

> If so, why are you seemingly uninterested in getting to the bottom of why the user experience with your software has obviously degraded with the version change from 22 to 23?

Reference for obvious degradation? With steps to reproduce and information requested in README?

cardboardaardvark, 2025-01-21:
> Reference for obvious degradation?

I'm confused by how this is not clear considering what I've already logged in this bug report but here you go, quoting myself from earlier in this ticket:

> As well I came across this problem because it showed up for me after an upgrade from 22.03 to 23.05. I kept the original configs as a part of the upgrade. Everything was fine before the upgrade. I get the exact problem here after the upgrade.

Steps to reproduce:

Short version: Upgrade from 22.03.3 to 23.05.5 and keep the configs

Long version:

1) Use OpenWRT 22.03.3 with a wireguard VPN and PBR for over a year with out any issues at all related to PBR failing to work after a reboot
2) Upgrade to 23.05.5 using the standard sysupgrade image and not one that incorporates the wireguard and PBR packages using the image builder
3) During the upgrade specify the option to keep the configuration
4) Reboot after the upgrade completes with out any reported issue and observe that everything functions as expected
5) Install the wireguard and PBR packages
6) Manually restart the network system and observe the wireguard VPN is operating
7) Manually start PBR and observe that routing through the wireguard VPN is functioning according to the policies given to PBR

I understand that keeping the configuration is unsupported when jumping major versions but that detail is actually irrelevant here as the other problem reports have not specified that they occurred during an upgrade nor have I seen any discussion related to this being an upgrade related bug.

> With steps to reproduce and information requested in README?

Reproduction steps have been supplied. I'm unsure of what additional information you are asking for here. Which README are you referring to? The wiki lists the following information needed for a bug report:

Name the tree/revision/version: OpenWRT 23.05.5 + PBR 1.1.6-22
Name the affected device: Netgear Nighthawk X4S R7800
What does it do that it should not do / what does it not do that it should do: Policy routing does not function after device is rebooted
Steps to reproduce: Configure policy routing, verify it is functional, reboot device, verify policy routing is no longer functional
What you have already done to workaround/fix the problem: Setting pbr.@pbr[0].procd_boot_delay=60 appears to avoid the issue
Any additional info you think is important: Problem was exposed after upgrade from 22.03.3

If there is other info you think would be helpful to confirm that this problem exists and is causing a degraded user experience after the upgrade then please let me know.

> maybe it was lost in post 1.1.6 transitions

So there is a possible regression in PBR. The behavior of PBR at boot is quite likely not actually as expected. 

This bug report continues to be in the closed state for what reason?

mikhirev, 2025-01-22:
> I've restored it in 1.1.8-r5 available from my repo.

Where can I find it? In https://github.com/stangri/repo.openwrt.melmac.net I see only 1.1.8-r4.

omnicron-llc, 2025-01-23:
> You're welcome to contribute to the docs, they are on github as well. My free unpaid time has already been spent this week responding to this issue.

Your unpaid time is worth less than nothing if you respond to issues incompetently.

sinancetinkaya, 2025-01-23:
One thing I don't understand is that why bash script language is used for such complex tasks? 
Can I code something complex in bash? Yes. Would I do it? Absolutely not. It's a primitive language, no debugging, no proper IDE, no break points, no variable monitoring. It's like an invitation for bugs.

cardboardaardvark, 2025-01-31:
What's the status here?

It appears there is a potential bugfix in version 1.1.8-r5 and now 1.1.8-r6 is available at https://github.com/stangri/repo.openwrt.melmac.net

I don't see any testing results from @mikhirev since he was unable to find the mentioned 1.1.8-r5 at the time it was referenced (I was also unable to find that file at the linked repo at that time).

Are we supposed to be testing 1.1.8-r6 as a potential fix?

Is this bug report going to stay in the closed state while testing for a potential fix happens?

mcdullmedia, 2025-05-27:
I have a similar issue but not the same.  My openwrt has multiple wan setting with travelmate.
priority WAN > WIFI client > WWAN (LTE dongle)
But it most case it is WIFI Client being connected and the WWAN.  WAN is rarely used.  
The pbr never started as it is always default to WAN interface.  
Any proper fix please.

omnicron-llc, 2025-08-13:
Incompetent "maintainers" concealing bugs by closing them in order to save face. What a state the open source world has reached.