openwrt/packages #25625: wget suggest using different alternative priority for wget-ssl and wget-nossl

1plus-1, 2024-12-27:
Currently wget-ssl and wget-nossl have the same alternative priority for link /usr/bin/wget. This causes an [issue](https://github.com/openwrt/openwrt/issues/17392) that the /usr/bin/wget is not created when both are selectd and then 'apk update' is not working.

The issue should be fix in openwrt repo but I do feel giving different priority for alternatives is a better practice.
https://github.com/openwrt/packages/blob/ca9c3d1baea38f0e5cedae7b74cfe8ef15a3ec83/net/wget/Makefile#L52

Maybe wget-ssl=300 and wget-nossl=400?

robimarko, 2024-12-28:
I agree, if both are present it makes sense that SSL variant has higher priority, please make a PR.

1plus-1, 2024-12-28:
PR #25633 is created.

Neustradamus, 2024-12-28:
Linked to:
- https://github.com/openwrt/packages/pull/25633
- https://github.com/openwrt/packages/issues/25625
- https://github.com/openwrt/openwrt/issues/17385
- https://github.com/openwrt/openwrt/issues/17392

PR #25633 (2024-12-31): wget: update alternative priority
Maintainer: me
Compile tested: mediatek/filogic
Run tested: Openwrt One

Description:
Fix #25625. Update wget-ssl and wget-nossl using differet alternative priorities for /usr/bin/wget.
closed 2024-12-31 by robimarko