openwrt/packages #25703: <coova-chilli>: Download Speed Capped at 40-50 Mbps with CoovaChilli on OpenWrt 23.05.5 (MT7621 + xt_coova)

aswanthk07, 2025-01-07:
# Issue: CoovaChilli Performance and Errors with xt_coova on OpenWrt 23.05.5

## Setup Details
- **CoovaChilli Version:** 1.6
- **Platform:** OpenWrt 23.05.5
- **Architecture:** MT7621
- **Firewall:** firewall3 + iptables_zz_legacy
- **Additional Support:** xt_coova
- **ISP Speed:** 100 Mbps (Real-world)

## Problem 1: Speed Capping on Download
When performing a speed test via CoovaChilli:
- **Download Speed:** 30-40 Mbps (tested using Ookla)
- **Upload Speed:** 100 Mbps (matches ISP speed)
- On the regular network (without CoovaChilli), both download and upload speeds match the ISP-provided 100 Mbps.

### Additional Testing
Performed an **iperf test**:
- **Setup:** Iperf server on one CoovaChilli client, Iperf client on another CoovaChilli client.
- **Result:** Download and upload speeds via CoovaChilli WAN were approximately **300-400 Mbps**, indicating no bottleneck in internal data flow.

---

## Problem 2: Errors When Using `kname="chilli"`

After setting `kname="chilli"` in the configuration, the following errors are observed in the logs:
```
Tue Jan  7 10:47:43 2025 daemon.err chilli[8361]: chilli[8361]: PID 8361 reloaded binary options file
Tue Jan  7 10:47:43 2025 daemon.info chilli[8361]: CoovaChilli 1.6. Copyright 2002-2005 Mondru AB. Licensed under GPL. Copyright 2006-2012 David Bird (Coova Technologies). Licensed under GPL. See http://coova.github.io/ for details.
Tue Jan  7 10:47:43 2025 daemon.info chilli[8361]: TX queue length set to 100
Tue Jan  7 10:47:43 2025 kern.info kernel: [  204.445245] device br-network4 entered promiscuous mode
Tue Jan  7 10:47:43 2025 daemon.err chilli[8361]: No such file or directory: could not open /proc/net/coova/chilli
Tue Jan  7 10:47:43 2025 kern.info kernel: [  204.605612] xt_coova: looking for chilli
Tue Jan  7 10:47:43 2025 kern.info kernel: [  204.609731] xt_coova: created chilli refcnt=1
Tue Jan  7 10:47:43 2025 kern.info kernel: [  204.614072] xt_coova: match ret=0
Tue Jan  7 10:47:43 2025 kern.info kernel: [  204.632326] xt_coova: looking for chilli
Tue Jan  7 10:47:43 2025 kern.info kernel: [  204.636280] xt_coova: found chilli refcnt=2
Tue Jan  7 10:47:43 2025 kern.info kernel: [  204.640524] xt_coova: match ret=0
Tue Jan  7 10:47:43 2025 daemon.err chilli[8361]: bad idx (-1)
Tue Jan  7 10:47:43 2025 daemon.warn chilli[8361]: RADIUS id 0 was not found in queue!
Tue Jan  7 10:47:52 2025 kern.info kernel: [  213.969309] seq_file: buggy .next function 0x835ee0d8 did not update position index
Tue Jan  7 10:47:53 2025 kern.info kernel: [  214.978726] seq_file: buggy .next function 0x835ee0d8 did not update position index
Tue Jan  7 10:47:54 2025 kern.info kernel: [  215.988028] seq_file: buggy .next function 0x835ee0d8 did not update position index
```
---

## Configuration Details

**Custom `up.sh` Script changes for `xt_coova`:**
```sh

set_xt_coova_rules() {
    case $TUNTAP in
        "tun1") NETWORK_RANGE="172.21.100.0/24" ;;
        "tun2") NETWORK_RANGE="172.22.100.0/24" ;;
        "tun3") NETWORK_RANGE="172.23.100.0/24" ;;
        "tun4") NETWORK_RANGE="172.24.100.0/24" ;;
        *) echo "Unknown TUNTAP device: $TUNTAP"; return ;;
    esac
    ipt -I FORWARD -o wan --src $NETWORK_RANGE -m coova --name chilli -j ACCEPT
    ipt -I FORWARD -i wan --dst $NETWORK_RANGE -m coova --name chilli --dest -j ACCEPT
    ipt -I FORWARD --src $NETWORK_RANGE -j ACCEPT
    ipt -I FORWARD --dst $NETWORK_RANGE -j ACCEPT
}
run_up() {
    if [ -n "$TUNTAP" ]; then
        if [ "$KNAME" != "" ]; then
            ipt -I FORWARD -i $DHCPIF -m coova --name $KNAME -j ACCEPT 
            ipt -I FORWARD -o $DHCPIF -m coova --name $KNAME --dest -j ACCEPT
        fi
        set_xt_coova_rules
    fi
    [ -e /etc/chilli/ipup.sh ] && . /etc/chilli/ipup.sh
}

```

@brada4 @sevan @pparent76 could you guys please look into this and help if you find anything wrong?

brada4, 2025-01-07:
can you show fragments from 
4 lines via `iptables-save` 
all lines with xt and related chain/table headers from `nft list ruleset`

aswanthk07, 2025-01-07:
@brada4 , sharing iptables output as well if this help in anyways. 
```sh
root@HaloWiFi:~# iptables -L -n -v
Chain INPUT (policy DROP 0 packets, 0 bytes)
 pkts bytes target     prot opt in     out     source               destination         
   23  3643 DROP       all  --  br-network4 *       0.0.0.0/0            0.0.0.0/0           
    0     0 ACCEPT     icmp --  tun4   *       0.0.0.0/0            172.24.100.1        
    0     0 ACCEPT     udp  --  tun4   *       0.0.0.0/0            172.24.100.1         udp dpt:53
    0     0 ACCEPT     udp  --  tun4   *       0.0.0.0/0            172.24.100.1         udp dpts:67:68
    0     0 ACCEPT     udp  --  tun4   *       0.0.0.0/0            255.255.255.255      udp dpts:67:68
    0     0 ACCEPT     tcp  --  tun4   *       0.0.0.0/0            172.24.100.1         tcp dpt:3991
  142 14944 ACCEPT     tcp  --  tun4   *       0.0.0.0/0            172.24.100.1         tcp dpt:3990
    0     0 DROP       all  --  tun4   *       0.0.0.0/0            172.24.100.1        
  154 13302 ACCEPT     all  --  lo     *       0.0.0.0/0            0.0.0.0/0            /* !fw3 */
 3614  291K input_rule  all  --  *      *       0.0.0.0/0            0.0.0.0/0            /* !fw3: Custom input rule chain */
 3502  269K ACCEPT     all  --  *      *       0.0.0.0/0            0.0.0.0/0            ctstate RELATED,ESTABLISHED /* !fw3 */
    0     0 syn_flood  tcp  --  *      *       0.0.0.0/0            0.0.0.0/0            tcp flags:0x17/0x02 /* !fw3 */
    1   326 zone_lan_input  all  --  br-lan *       0.0.0.0/0            0.0.0.0/0            /* !fw3 */
  107 20506 zone_wan_input  all  --  wan    *       0.0.0.0/0            0.0.0.0/0            /* !fw3 */
    0     0 zone_network3_input  all  --  br-network3 *       0.0.0.0/0            0.0.0.0/0            /* !fw3 */
    0     0 zone_network4_input  all  --  br-network4 *       0.0.0.0/0            0.0.0.0/0            /* !fw3 */
    4  1360 reject     all  --  *      *       0.0.0.0/0            0.0.0.0/0            /* !fw3 */

Chain FORWARD (policy DROP 0 packets, 0 bytes)
 pkts bytes target     prot opt in     out     source               destination         
32840   40M ACCEPT     all  --  *      *       0.0.0.0/0            172.24.100.0/24     
40408   24M ACCEPT     all  --  *      *       172.24.100.0/24      0.0.0.0/0           
    0     0 ACCEPT     all  --  wan    *       0.0.0.0/0            172.24.100.0/24     coova: name: chilli side: dest
    0     0 ACCEPT     all  --  *      wan     172.24.100.0/24      0.0.0.0/0           coova: name: chilli side: source 
    0     0 ACCEPT     all  --  tun4   wan     0.0.0.0/0            0.0.0.0/0           
    0     0 DROP       all  --  tun4   !wan    0.0.0.0/0            0.0.0.0/0           
  244 14600 TCPMSS     tcp  --  *      *       0.0.0.0/0            0.0.0.0/0            tcp flags:0x06/0x02 TCPMSS clamp to PMTU
    0     0 ACCEPT     all  --  *      tun4    0.0.0.0/0            0.0.0.0/0           
    0     0 ACCEPT     all  --  tun4   *       0.0.0.0/0            0.0.0.0/0           
    0     0 DROP       all  --  *      br-network4  0.0.0.0/0            0.0.0.0/0           
    1    80 DROP       all  --  br-network4 *       0.0.0.0/0            0.0.0.0/0           
13933 6091K forwarding_rule  all  --  *      *       0.0.0.0/0            0.0.0.0/0            /* !fw3: Custom forwarding rule chain */
12386 5970K ACCEPT     all  --  *      *       0.0.0.0/0            0.0.0.0/0            ctstate RELATED,ESTABLISHED /* !fw3 */
 1547  121K zone_lan_forward  all  --  br-lan *       0.0.0.0/0            0.0.0.0/0            /* !fw3 */
    0     0 zone_wan_forward  all  --  wan    *       0.0.0.0/0            0.0.0.0/0            /* !fw3 */
    0     0 zone_network3_forward  all  --  br-network3 *       0.0.0.0/0            0.0.0.0/0            /* !fw3 */
    0     0 zone_network4_forward  all  --  br-network4 *       0.0.0.0/0            0.0.0.0/0            /* !fw3 */
    0     0 reject     all  --  *      *       0.0.0.0/0            0.0.0.0/0            /* !fw3 */

Chain OUTPUT (policy ACCEPT 0 packets, 0 bytes)
 pkts bytes target     prot opt in     out     source               destination         
  154 13302 ACCEPT     all  --  *      lo      0.0.0.0/0            0.0.0.0/0            /* !fw3 */
 4151 1150K output_rule  all  --  *      *       0.0.0.0/0            0.0.0.0/0            /* !fw3: Custom output rule chain */
 3962 1136K ACCEPT     all  --  *      *       0.0.0.0/0            0.0.0.0/0            ctstate RELATED,ESTABLISHED /* !fw3 */
    1   341 zone_lan_output  all  --  *      br-lan  0.0.0.0/0            0.0.0.0/0            /* !fw3 */
  188 13779 zone_wan_output  all  --  *      wan     0.0.0.0/0            0.0.0.0/0            /* !fw3 */
    0     0 zone_network3_output  all  --  *      br-network3  0.0.0.0/0            0.0.0.0/0            /* !fw3 */
    0     0 zone_network4_output  all  --  *      br-network4  0.0.0.0/0            0.0.0.0/0            /* !fw3 */

Chain forwarding_lan_rule (1 references)
 pkts bytes target     prot opt in     out     source               destination         

Chain forwarding_network1_rule (1 references)
 pkts bytes target     prot opt in     out     source               destination         

Chain forwarding_network2_rule (1 references)
 pkts bytes target     prot opt in     out     source               destination         

Chain forwarding_network3_rule (1 references)
[... cut ...]
 pkts bytes target     prot opt in     out     source               destination         
  188 13779 output_wan_rule  all  --  *      *       0.0.0.0/0            0.0.0.0/0            /* !fw3: Custom wan output rule chain */
  188 13779 zone_wan_dest_ACCEPT  all  --  *      *       0.0.0.0/0            0.0.0.0/0            /* !fw3 */

Chain zone_wan_src_REJECT (1 references)
 pkts bytes target     prot opt in     out     source               destination         
   95 20122 reject     all  --  wan    *       0.0.0.0/0            0.0.0.0/0            /* !fw3 */
```

brada4, 2025-01-07:
There is space char missing.
```
-A FORWARD -s 172.24.100.0/24 -o wan -m coova--name chilli --source  -j ACCEPT
--------------------------------------------^^
```

aswanthk777, 2025-01-08:
> There is space char missing.
> 
> ```
> -A FORWARD -s 172.24.100.0/24 -o wan -m coova--name chilli --source  -j ACCEPT
> --------------------------------------------^^
> ```

Thanks @brada4 for the findings, as I mentioned above these are the commands I'm using for xt_coova to kick in. 
```
    ipt -I FORWARD -o wan --src $NETWORK_RANGE -m coova --name chilli -j ACCEPT
    ipt -I FORWARD -i wan --dst $NETWORK_RANGE -m coova --name chilli --dest -j ACCEPT
    ipt -I FORWARD --src $NETWORK_RANGE -j ACCEPT
    ipt -I FORWARD --dst $NETWORK_RANGE -j ACCEPT
```
There is already a space char between them in my up.sh and tried adding one more space char but didn't  make any change. Will this be an issue of iptables-zz-legacy version I'm using? Also can you please suggest if I can use iptables-nft instead of iptables-zz-legacy for coova if this is the issue?

aswanthk07, 2025-01-08:
> Thanks @brada4 for the findings, as I mentioned above these are the commands I'm using for xt_coova to kick in.
>
> ```
>     ipt -I FORWARD -o wan --src $NETWORK_RANGE -m coova --name chilli -j [...]

```
:~# iptables -L -v -n | grep chilli
    0     0 ACCEPT     all  --  wan    *       0.0.0.0/0            172.24.100.0/24     coova: name: chilli side: dest
    0     0 ACCEPT     all  --  *      wan     172.24.100.0/24      0.0.0.0/0           coova: name: chilli side: source 
:~# iptables-save | grep chilli
-A FORWARD -d 172.24.100.0/24 -i wan -m coova--name chilli --dest  -j ACCEPT
-A FORWARD -s 172.24.100.0/24 -o wan -m coova--name chilli --source  -j ACCEPT
```

Also looking into iptables the name and coova module initialized correctly it seems despite the space char issue still persist on iptables-save.

brada4, 2025-01-08:
You see youroutput is a new issue, try forums, or opennds first.

rlimaboldmss, 2025-04-02:
> # Issue: CoovaChilli Performance and Errors with xt_coova on OpenWrt 23.05.5
> ## Setup Details
> * **CoovaChilli Version:** 1.6
> * **Platform:** OpenWrt 23.05.5
> * **Architecture:** MT7621
> * **Firewall:** firewall3 + iptables_zz_legacy
> * **Additional Support:** xt_coova
> * **ISP Speed:** 100 Mbps (Real-world)
> 
> ## Problem 1: Speed Capping on Download
> When performing a speed test via CoovaChilli:
> 
> * **Download Speed:** 30-40 Mbps (tested using Ookla)
> * **Upload Speed:** 100 Mbps (matches ISP speed)
> * On the regular network (without CoovaChilli), both download and upload speeds match the ISP-provided 100 Mbps.
> 
> ### Additional Testing
> Performed an **iperf test**:
> 
> * **Setup:** Iperf server on one CoovaChilli client, Iperf client on another CoovaChilli client.
> * **Result:** Download and upload speeds via CoovaChilli WAN were approximately **300-400 Mbps**, indicating no bottleneck in internal data flow.
> 
> ## Problem 2: Errors When Using `kname="chilli"`
> After setting `kname="chilli"` in the configuration, the following errors are observed in the logs:
> 
> ```
> Tue Jan  7 10:47:43 2025 daemon.err chilli[8361]: chilli[8361]: PID 8361 reloaded binary options file
> Tue Jan  7 10:47:43 2025 daemon.info chilli[8361]: CoovaChilli 1.6. Copyright 2002-2005 Mondru AB. Licensed under GPL. Copyright 2006-2012 David Bird (Coova Technologies). Licensed under GPL. See http://coova.github.io/ for details.
> Tue Jan  7 10:47:43 2025 daemon.info chilli[8361]: TX queue length set to 100
> Tue Jan  7 10:47:43 2025 kern.info kernel: [  204.445245] device br-network4 entered promiscuous mode
> Tue Jan  7 10:47:43 2025 daemon.err chilli[8361]: No such file or directory: could not open /proc/net/coova/chilli
> Tue Jan  7 10:47:43 2025 kern.info kernel: [  204.605612] xt_coova: looking for chilli
> Tue Jan  7 10:47:43 2025 kern.info kernel: [  204.609731] xt_coova: created chilli refcnt=1
> Tue Jan  7 10:47:43 2025 kern.info kernel: [  204.614072] xt_coova: match ret=0
> Tue Jan  7 10:47:43 2025 kern.info kernel: [  204.632326] xt_coova: looking for chilli
> Tue Jan  7 10:47:43 2025 kern.info kernel: [  204.636280] xt_coova: found chilli refcnt=2
> Tue Jan  7 10:47:43 2025 kern.info kernel: [  204.640524] xt_coova: match ret=0
> Tue Jan  7 10:47:43 2025 daemon.err chilli[8361]: bad idx (-1)
> Tue Jan  7 10:47:43 2025 daemon.warn chilli[8361]: RADIUS id 0 was not found in queue!
> Tue Jan  7 10:47:52 2025 kern.info kernel: [  213.969309] seq_file: buggy .next function 0x835ee0d8 did not update position index
> Tue Jan  7 10:47:53 2025 kern.info kernel: [  214.978726] seq_file: buggy .next function 0x835ee0d8 did not update position index
> Tue Jan  7 10:47:54 2025 kern.info kernel: [  215.988028] seq_file: buggy .next function 0x835ee0d8 did not update position index
> ```
> 
> ## Problem 3: Upload Speed is Unlimited, Bandwidth Limitation Not Working
> * **Observation:** Upload speeds through CoovaChilli are now unlimited and match the ISP-provided speed (100 Mbps). However, any bandwidth limit (set manually or from RADIUS) for the upload is ignored.
> * This issue occurs even when download bandwidth limitations are applied correctly.
> * No errors appear in the logs specific to this problem, but the bandwidth control feature for uploads seems to be bypassed.
> 
> ## Configuration Details
> **Custom `up.sh` Script changes for `xt_coova`:**
> 
> set_xt_coova_rules() {
>     case $TUNTAP in
>         "tun1") NETWORK_RANGE="172.21.100.0/24" ;;
>         "tun2") NETWORK_RANGE="172.22.100.0/24" ;;
>         "tun3") NETWORK_RANGE="172.23.100.0/24" ;;
>         "tun4") NETWORK_RANGE="172.24.100.0/24" ;;
>         *) echo "Unknown TUNTAP device: $TUNTAP"; return ;;
>     esac
>     ipt -I FORWARD -o wan --src $NETWORK_RANGE -m coova --name chilli -j ACCEPT
>     ipt -I FORWARD -i wan --dst $NETWORK_RANGE -m coova --name chilli --dest -j ACCEPT
>     ipt -I FORWARD --src $NETWORK_RANGE -j ACCEPT
>     ipt -I FORWARD --dst $NETWORK_RANGE -j ACCEPT
> }
> run_up() {
>     if [ -n "$TUNTAP" ]; then
>         if [ "$KNAME" != "" ]; then
>             ipt -I FORWARD -i $DHCPIF -m coova --name $KNAME -j ACCEPT 
>             ipt -I FORWARD -o $DHCPIF -m coova --name $KNAME --dest -j ACCEPT
>         fi
>         set_xt_coova_rules
>     fi
>     [ -e /etc/chilli/ipup.sh ] && . /etc/chilli/ipup.sh
> }
> **Chilli Configuration:**
> 
> tundev="tun4"
> net="172.24.100.0/24"
> dynip="172.24.100.0/24"
> dns1="8.8.8.8"
> dns2="8.8.4.4"
> ipup="/etc/chilli/up.sh"
> ipdown="/etc/chilli/down.sh"
> radiusserver1="x.x.x.x"
> radiusserver2="x.x.x.x"
> radiussecret="xxxxx"
> dhcpif="br-network4"
> uamserver="https://example.com"
> uamlisten="172.24.100.1"
> uamallowed="example.com"
> uamdomain=".example.com"
> wwwdir="/etc/chilli/www"
> wwwbin="/etc/chilli/wwwsh"
> radiusnasid="x-x-x"
> kname="chilli"
> [@brada4](https://github.com/brada4) [@sevan](https://github.com/sevan) [@pparent76](https://github.com/pparent76) could you guys please look into this and help if you find anything wrong?

Hi @aswanthk07 do you have any news on this? I am facing same issue