openwrt/packages #25705: Dropbear: on lantiq target no longer supports ed25519 keys

achelon5, 2025-01-08:
Maintainer: @\<hauke@hauke-m.de>
Environment: 
Model: Netgear DM200
Architecture: xRX200 rev 1.2
Target Platform: lantiq/xrx200_legacy
Firmware Version: OpenWrt 24.10.0-rc5 r28304-6dacba30a7 / LuCI openwrt-24.10 branch 25.006.62535~c9cc773

Description:

Dropbear is missing support for ed25519 keys in  OpenWrt 24.10.0-rc5 on the DM200 build. Attempting SSH authentication with an ed25519 key results in "Permission denied" in the SSH client. Attempts to generate an  ed25519 result in "Unknown key type 'ed25519'" when following the steps in the documentation https://openwrt.org/docs/guide-user/security/dropbear.public-key.auth.

```
root@dm200:~# dropbearkey -t ed25519 -f ~/.ssh/id_dropbear
Unknown key type 'ed25519'
Usage: dropbearkey -t <type> -f <filename> [-s bits]
-t type	Type of key to generate. One of:
		rsa
-f filename    Use filename for the secret key.
               ~/.ssh/id_dropbear is recommended for client keys.
-s bits	Key size in bits, should be a multiple of 8 (optional)
-y		Just print the publickey and fingerprint for the
		private key in <filename>.
-C		Specify the key comment (email).
```

achelon5, 2025-01-13:
I did not compile anything, I just used the firmware builder and it produced a build with no ed25519 support and this has not happened pre 24.10.0. Can it be fixed?

dhewg, 2025-01-13:
It's not just his image:
e63326e26a82cb083eec602b3a2f71d4cd8e2ea4
d0f295837a03f7f52000ae6d395827bdde7996a4

This "works" as intended

achelon5, 2025-01-13:
I'm slightly at a loss how to debug this. I generated a custom image for a Raspberry Pi4B and ed25519 support works fine:

```
Usage: dropbearkey -t <type> -f <filename> [-s bits]
-t type	Type of key to generate. One of:
		rsa
		ed25519
-f filename    Use filename for the secret key.
               ~/.ssh/id_dropbear is recommended for client keys.
-s bits	Key size in bits, should be a multiple of 8 (optional)
           Ed25519 has a fixed size of 256 bits
-y		Just print the publickey and fingerprint for the
		private key in <filename>.
-C		Specify the key comment (email).
```

but an image generated for the DM200 target is missing ed25519 support:

```
Usage: dropbearkey -t <type> -f <filename> [-s bits]
-t type	Type of key to generate. One of:
		rsa
-f filename    Use filename for the secret key.
               ~/.ssh/id_dropbear is recommended for client keys.
-s bits	Key size in bits, should be a multiple of 8 (optional)
-y		Just print the publickey and fingerprint for the
		private key in <filename>.
-C		Specify the key comment (email).
```

There are no obvious issues in my package specification to the image bulder, and the same package specification worked fine against 23.05. The suggested workaround of generating the key locally does not work because support is missing entirely and authentication attempts with the key will fail with "permission denied".  Should I raise an issue against the image builder?

achelon5, 2025-01-13:
> @achelon5 See the commit description [d0f2958](https://github.com/openwrt/packages/commit/d0f295837a03f7f52000ae6d395827bdde7996a4)
> 
> The Netgear DM200 has a small flash disk that's why its image was compiled with the SMALL_FLASH build flag. When the SMALL_FLASH flag is enabled then the DROPBEAR_ED25519 config is disabled by default:
> 
> > config DROPBEAR_ED25519
> > default y if !SMALL_FLASH

This would seem to explain it. Other than setup my own build environment to recompile, ia there any other solution for this issue?

achelon5, 2025-01-13:
> why don't you use use the rsa? It's safe and the only advantage of the ed25519 is that the key negotiation during a connection is faster. A human anyway won't notice any difference.

My hardware token has an ED25519 key stored on it, not an RSA token.

t's a shame there isn't a separate package with a dropbear+ED25519 support in it because my DM200 has plenty of storage space left on it. A number of packages are not required in bridge mode, which is the only reason to use this device. (This device is one of the very few ones your software supports that has working VDSL2 and doesn't have un-needed features like WiFI. I.e. it is a perfect modem only VDSL2 device.)

closed 2025-01-13 by achelon5