openwrt/packages #25709: Yaffs2 cannot be compiled with GCC 14.2 on Main branch

sppmasterspp, 2025-01-08:
Maintainer: I cannot find it. (find it by checking history of the package Makefile)
Environment: 

Description: YAFFS2 fails to compile on Main branch with GCC 14.2

```
printenv: executable
install -d -m0755 /home/debian/openwrt-RB5009UG/bin/targets/mvebu/cortexa72/packages/tmp
mkdir -p /home/debian/openwrt-RB5009UG/build_dir/target-aarch64_cortex-a72_musl/u-boot-2024.07/ipkg-aarch64_cortex-a72/uboot-envtools/CONTROL
(cd /home/debian/openwrt-RB5009UG/build_dir/target-aarch64_cortex-a72_musl/u-boot-2024.07/ipkg-aarch64_cortex-a72/uboot-envtools/CONTROL; ( echo "$CONTROL"; printf "Description: "; echo "$DESCRIPTION" | sed -e 's,^[[:space:]]*, ,g'; ) > control; chmod 644 control; ( echo "#!/bin/sh"; echo "[ \"\${IPKG_NO_SCRIPT}\" = \"1\" ] && exit 0"; echo "[ -s "\${IPKG_INSTROOT}/lib/functions.sh" ] || exit 0"; echo ". \${IPKG_INSTROOT}/lib/functions.sh"; echo "default_postinst \$0 \$@"; ) > postinst; ( echo "#!/bin/sh"; echo "[ -s "\${IPKG_INSTROOT}/lib/functions.sh" ] || exit 0"; echo ". \${IPKG_INSTROOT}/lib/functions.sh"; echo "default_prerm \$0 \$@"; ) > prerm; chmod 0755 postinst prerm; echo "$V_Package_uboot_envtools_conffiles" > conffiles;  )
/home/debian/openwrt-RB5009UG/staging_dir/host/bin/fakeroot /home/debian/openwrt-RB5009UG/staging_dir/host/bin/bash /home/debian/openwrt-RB5009UG/scripts/ipkg-build -m "" /home/debian/openwrt-RB5009UG/build_dir/target-aarch64_cortex-a72_musl/u-boot-2024.07/ipkg-aarch64_cortex-a72/uboot-envtools /home/debian/openwrt-RB5009UG/bin/targets/mvebu/cortexa72/packages
/home/debian/openwrt-RB5009UG/staging_dir/host/bin/find: '/home/debian/openwrt-RB5009UG/build_dir/target-aarch64_cortex-a72_musl/u-boot-2024.07/ipkg-aarch64_cortex-a72/uboot-envtools/etc/config/ubootenv': No such file or directory
/home/debian/openwrt-RB5009UG/staging_dir/host/bin/find: '/home/debian/openwrt-RB5009UG/build_dir/target-aarch64_cortex-a72_musl/u-boot-2024.07/ipkg-aarch64_cortex-a72/uboot-envtools/etc/fw_env.config': No such file or directory
/home/debian/openwrt-RB5009UG/staging_dir/host/bin/find: '/home/debian/openwrt-RB5009UG/build_dir/target-aarch64_cortex-a72_musl/u-boot-2024.07/ipkg-aarch64_cortex-a72/uboot-envtools/etc/fw_sys.config': No such file or directory
Packaged contents of /home/debian/openwrt-RB5009UG/build_dir/target-aarch64_cortex-a72_musl/u-boot-2024.07/ipkg-aarch64_cortex-a72/uboot-envtools into /home/debian/openwrt-RB5009UG/bin/targets/mvebu/cortexa72/packages/uboot-envtools_2024.07-r1_aarch64_cortex-a72.ipk
echo "uboot-envtools" >> /home/debian/openwrt-RB5009UG/staging_dir/target-aarch64_cortex-a72_musl/pkginfo/uboot-envtools.default.install
make[3]: Leaving directory '/home/debian/openwrt-RB5009UG/package/boot/uboot-envtools'
time: package/boot/uboot-envtools/compile#9.02#10.53#27.29
make[3]: Entering directory '/home/debian/openwrt-RB5009UG/package/boot/uboot-mvebu'
rm -f /home/debian/openwrt-RB5009UG/build_dir/target-aarch64_cortex-a72_musl/u-boot-rb5009/u-boot-2024.10/.built
touch /home/debian/openwrt-RB5009UG/build_dir/target-aarch64_cortex-a72_musl/u-boot-rb5009/u-boot-2024.10/.built_check
make  -C /home/debian/openwrt-RB5009UG/build_dir/target-aarch64_cortex-a72_musl/u-boot-rb5009/u-boot-2024.10 CROSS_COMPILE=aarch64-openwrt-linux-musl- DTC="/home/debian/openwrt-RB5009UG/build_dir/target-aarch64_cortex-a72_musl/linux-mvebu_cortexa72/linux-6.6.58/scripts/dtc/dtc" PATH=/home/debian/openwrt-RB5009UG/staging_dir/host/bin:/home/debian/openwrt-RB5009UG/staging_dir/target-aarch64_cortex-a72_musl/host/bin:/home/debian/openwrt-RB5009UG/staging_dir/hostpkg/bin:/home/debian/openwrt-RB5009UG/staging_dir/toolchain-aarch64_cortex-a72_gcc-14.2.0_musl/bin:/home/debian/openwrt-RB5009UG/staging_dir/toolchain-aarch64_cortex-a72_gcc-14.2.0_musl/bin:/home/debian/openwrt-RB5009UG/staging_dir/host/bin:/home/debian/openwrt-RB5009UG/staging_dir/toolchain-aarch64_cortex-a72_gcc-14.2.0_musl/bin:/home/debian/openwrt-RB5009UG/staging_dir/host/bin:/home/debian/openwrt-RB5009UG/staging_dir/host/bin:/usr/local/sbin:/usr/sbin:/sbin:/usr/local/bin:/usr/bin:/bin:/usr/local/games:/usr/games:/home/debian/.dotnet/tools HOSTCC="ccache /home/debian/openwrt-RB5009UG/staging_dir/host/bin/gcc" HOSTCFLAGS="-O2 -I/home/debian/openwrt-RB5009UG/staging_dir/host/include -I/home/debian/openwrt-RB5009UG/staging_dir/hostpkg/include -I/home/debian/openwrt-RB5009UG/staging_dir/target-aarch64_cortex-a72_musl/host/include -I/home/debian/openwrt-RB5009UG/staging_dir/host/include -I/home/debian/openwrt-RB5009UG/staging_dir/hostpkg/include -I/home/debian/openwrt-RB5009UG/staging_dir/target-aarch64_cortex-a72_musl/host/include -std=gnu11" HOSTLDFLAGS="-L/home/debian/openwrt-RB5009UG/staging_dir/host/lib -L/home/debian/openwrt-RB5009UG/staging_dir/hostpkg/lib -L/home/debian/openwrt-RB5009UG/staging_dir/target-aarch64_cortex-a72_musl/host/lib" LOCALVERSION="-OpenWrt-r27903-27657050d0" STAGING_PREFIX="/home/debian/openwrt-RB5009UG/staging_dir/host" PKG_CONFIG_PATH="/home/debian/openwrt-RB5009UG/staging_dir/host/lib/pkgconfig" PKG_CONFIG_LIBDIR="/home/debian/openwrt-RB5009UG/staging_dir/host/lib/pkgconfig" PKG_CONFIG_EXTRAARGS="--static" V=1
make[4]: Entering directory '/home/debian/openwrt-RB5009UG/build_dir/target-aarch64_cortex-a72_musl/u-boot-rb5009/u-boot-2024.10'
set -e; mkdir -p include/config/;       KERNELVERSION=2024.10 /bin/bash ./scripts/setlocalversion . < include/config/auto.conf > include/config/uboot.release.tmp; if [ -r include/config/uboot.release ] && cmp -s include/config/uboot.release include/config/uboot.release.tmp; then rm -f include/config/uboot.release.tmp; else : '  UPD     include/config/uboot.release'; mv -f include/config/uboot.release.tmp include/config/uboot.release; fi
set -e; mkdir -p include/generated/;    (echo \#define PLAIN_VERSION \"2024.10-OpenWrt-r27903-27657050d0\"; echo \#define U_BOOT_VERSION \"U-Boot \" PLAIN_VERSION; echo \#define U_BOOT_VERSION_NUM 2024; echo \#define U_BOOT_VERSION_NUM_PATCH $(echo 10 | sed -e "s/^0*//"); echo \#define HOST_ARCH 0xaa64; echo \#define CC_VERSION_STRING \"$(LC_ALL=C aarch64-openwrt-linux-musl-gcc --version | head -n 1)\"; echo \#define LD_VERSION_STRING \"$(LC_ALL=C aarch64-openwrt-linux-musl-ld.bfd --version | head -n 1)\"; ) < include/config/uboot.release > include/generated/version_autogenerated.h.tmp; if [ -r include/generated/version_autogenerated.h ] && cmp -s include/generated/version_autogenerated.h include/generated/version_autogenerated.h.tmp; then rm -f include/generated/version_autogenerated.h.tmp; else : '  UPD     include/generated/version_autogenerated.h'; mv -f include/generated/version_autogenerated.h.tmp include/generated/version_autogenerated.h; fi
set -e; mkdir -p include/generated/;    (if test -n "${SOURCE_DATE_EPOCH}"; then SOURCE_DATE="@${SOURCE_DATE_EPOCH}"; DATE=""; for date in gdate date.gnu date; do ${date} -u -d "${SOURCE_DATE}" >/dev/null 2>&1 && DATE="${date}"; done; if test -n "${DATE}"; then LC_ALL=C ${DATE} -u -d "${SOURCE_DATE}" +'#define U_BOOT_DATE "%b %d %C%y"'; LC_ALL=C ${DATE} -u -d "${SOURCE_DATE}" +'#define U_BOOT_TIME "%T"'; LC_ALL=C ${DATE} -u -d "${SOURCE_DATE}" +'#define U_BOOT_TZ "%z"'; LC_ALL=C ${DATE} -u -d "${SOURCE_DATE}" +'#define U_BOOT_EPOCH %s'; else return 42; fi; else LC_ALL=C date +'#define U_BOOT_DATE "%b %d %C%y"'; LC_ALL=C date +'#define U_BOOT_TIME "%T"'; LC_ALL=C date +'#define U_BOOT_TZ "%z"'; LC_ALL=C date +'#define U_BOOT_EPOCH %s'; fi) < Makefile > include/generated/timestamp_autogenerated.h.tmp; if [ -r include/generated/timestamp_autogenerated.h ] && cmp -s include/generated/timestamp_autogenerated.h include/generated/timestamp_autogenerated.h.tmp; then rm -f include/generated/timestamp_autogenerated.h.tmp; else : '  UPD     include/generated/timestamp_autogenerated.h'; mv -f include/generated/timestamp_autogenerated.h.tmp include/generated/timestamp_autogenerated.h; fi
set -e; mkdir -p include/generated/;    (if test -n "${DEVICE_TREE}"; then echo \#define DEVICE_TREE \"\"; else echo \#define DEVICE_TREE CONFIG_DEFAULT_DEVICE_TREE; fi) < Makefile > include/generated/dt.h.tmp; if [ -r include/generated/dt.h ] && cmp -s include/generated/dt.h include/generated/dt.h.tmp; then rm -f include/generated/dt.h.tmp; else : '  UPD     include/generated/dt.h'; mv -f include/generated/dt.h.tmp include/generated/dt.h; fi
make -f ./scripts/Makefile.build obj=scripts/basic
rm -f .tmp_quiet_recordmcount
make -f ./scripts/Makefile.build obj=.
[... cut ...]
make[2]: *** [package/Makefile:177: package/boot/uboot-mvebu/compile] Error 1
make[2]: Leaving directory '/home/debian/openwrt-RB5009UG'
make[1]: *** [package/Makefile:171: /home/debian/openwrt-RB5009UG/staging_dir/target-aarch64_cortex-a72_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/debian/openwrt-RB5009UG'
make: *** [/home/debian/openwrt-RB5009UG/include/toplevel.mk:248: world] Error 2
```
For reference - https://forum.openwrt.org/t/building-kernel-6-6-for-mikrotik-rb5009ug/193213/34?u=sppmaster

neheb, 2025-01-09:
a little bit of searching shows https://github.com/u-boot/u-boot/commit/957a3b982082627ac474fd1c88a5cfc0a28543ae

Maybe relevant here.

neheb, 2025-01-09:
or https://github.com/u-boot/u-boot/commit/bd356e05cd82d7b02a09a590f497f9a2f7a01277

...

sppmasterspp, 2025-01-09:
Current workaround is to remove yaffs2.
https://github.com/openwrt/openwrt/blob/main/package/boot/uboot-mvebu/patches/102-arm-mvebu-add-support-for-MikroTik-RB5009UG-S-IN.patch#L457

```
456 +CONFIG_USB_ETHER_RTL8152=y
457 +# CONFIG_YAFFS2 is not set
458 +# CONFIG_SHA256 is not set
```