openwrt/packages #25712: glib2 host build doesn't have rpath set, causing unpredictable errors

wryun, 2025-01-08:
# Issue template (remove lines from top till here)

Maintainer: @yegorich @neheb 
Environment: OpenWrt 23.0.5.5 host build

Description:

A host build of glib2 generate binaries that don't have the rpath set, which means they rely on locally installed libraries when run (despite being built against host lib dependencies). I've noticed this because it breaks libsoup's build, which uses glib-compile-resources

For example, I don't have libffi7 installed (Ubuntu 22.04):

```
; /home/james.haggerty/a/morse_openwrt_23/staging_dir/hostpkg/bin/glib-compile-resources ../libsoup/soup.gresource.xml --sourcedir ../libsoup/. --sourcedir ../libsoup --c-name soup --internal --generate --target libsoup/soup-resources.c --dependency-file libsoup/soup-resources.c.d
/home/james.haggerty/a/morse_openwrt_23/staging_dir/hostpkg/bin/glib-compile-resources: error while loading shared libraries: libffi.so.7: cannot open shared object file: No such file or directory
```

But this works if I point it explicitly at hostpkg/lib:
```
; local (LD_LIBRARY_PATH=/home/james.haggerty/a/morse_openwrt_23/staging_dir/hostpkg/lib) /home/james.haggerty/a/morse_openwrt_23/staging_dir/hostpkg/bin/glib-compile-resources ../libsoup/soup.gresource.xml --sourcedir ../libsoup/. --sourcedir ../libsoup --c-name soup --internal --generate --target libsoup/soup-resources.c --dependency-file libsoup/soup-resources.c.d
```

This can be fixed by adding:
```
HOST_LDFLAGS+=-Wl,-rpath,$(STAGING_DIR_HOSTPKG)/lib 
```
to the glib2 Makefile, but I wasn't sure if this was the right way to do it, so I haven't raised a PR.

neheb, 2025-01-09:
Sounds like libffi static needs to be backported.

neheb, 2025-01-09:
Weird. https://github.com/openwrt/packages/commit/aba4a294ba35341c2253ff5c0f151613215878c2 is in 23.05

maybe staging_dir needs cleaning.

wryun, 2025-01-09:
Yes, my apologies. Looks like I had a staging_dir that had been around for too long which had libffi.so in it, so meson cheerfully found it and used it. Removing libffi.so solved my problem.

Thanks very much!

closed 2025-01-09 by wryun