openwrt/packages #25722: Transmission : Custom 'done' script never runs.

mic-user, 2025-01-12:
Maintainer: @kolacinskikarol @dangowrt
Environment: mvebu / cortexa9 - Marvell Armada 385 88F6820 / Linksys WRT32X / 24.10.0-rc5

Description:
The feature 'Script torrent done filename' not working which automatically run a custom script when a torrent download finished. Also not a permission problem since it runs with root:root. It wasnt a problem with OpenwWRT 19.7.10's transmission

lbernstone, 2025-01-23:
I'm guessing this is due to the script not being allowed in the procd jail.  Anybody know what needs to be opened up to allow that script (which will be available as part of the uci info) to fork a new process?

mic-user, 2025-01-26:
@lbernstone You talking about '/etc/seccomp/transmission-daemon.json' ? Damn, found it.
Commenting below in the init.d script show no problem. no crash like symptom.
`seccomp_path="/etc/seccomp/transmission-daemon.json"
#if [ -f "$seccomp_path" ]; then
#       procd_set_param seccomp "$seccomp_path"
#fi
`

Then, how to find the which call should be add into the seccomp file for normal working and precise security matter?