openwrt/packages #25771: Unstable USB-tethering connection in v23.05, fails after 1-2 hours of use

sndcrde, 2025-01-17:
Related: https://forum.openwrt.org/t/cant-get-usb-tethering-to-work-in-24-10-rc5/221685

Hi,

While USB-tethering does work for me in v23.05, the connection becomes unstable/unusable after a couple of hours of use: ping times increase to ~1s instead of ~50ms I usually get, many dropped packets and sometimes I see something like the following:

```
# ping 1.1.1.1
PING 1.1.1.1 (1.1.1.1) 56(84) bytes of data.
64 bytes from 1.1.1.1: icmp_seq=1 ttl=52 time=632 ms
64 bytes from 1.1.1.1: icmp_seq=2 ttl=52 time=472 ms
64 bytes from 1.1.1.1: icmp_seq=3 ttl=52 time=199 ms
64 bytes from 1.1.1.1: icmp_seq=4 ttl=52 time=50.8 ms
64 bytes from 1.1.1.1: icmp_seq=5 ttl=52 time=100 ms
64 bytes from 1.1.1.1: icmp_seq=6 ttl=52 time=197 ms
wrong data byte #54 should be 0x36 but was 0x76
#16     10 11 12 13 14 15 16 17 18 19 1a 1b 1c 1d 1e 1f 20 21 22 23 24 25 26 27 28 29 2a 2b 2c 2d 2e 2f
#48     30 31 32 33 34 35 76 cb
64 bytes from 1.1.1.1: icmp_seq=8 ttl=52 time=139 ms
wrong data byte #54 should be 0x36 but was 0x30
#16     10 11 12 13 14 15 16 17 18 19 1a 1b 1c 1d 1e 1f 20 21 22 23 24 25 26 27 28 29 2a 2b 2c 2d 2e 2f
...
^C
--- 1.1.1.1 ping statistics ---
16 packets transmitted, 13 received, 18.75% packet loss, time 15037ms
rtt min/avg/max/mdev = 50.796/274.266/682.831/215.647 ms

```

Installing `kmod-usb-net-cdc-ncm` doesn't seem to improve stability. Installing `kmod-usb-net-cdc-eem` in addition to `kmod-usb-net-cdc-ncm` also doesn't seem to improve stability.

The following is what I did to get USB-tething working with v23.05.02

```
# opkg update
# opkg install kmod-usb-net-rndis
# opkg install kmod-usb-core kmod-usb-ehci kmod-usb2

# opkg list-installed | grep usb
kmod-phy-ath79-usb - 5.15.167-1
kmod-usb-core - 5.15.167-1
kmod-usb-ehci - 5.15.167-1
kmod-usb-net - 5.15.167-1
kmod-usb-net-cdc-ether - 5.15.167-1
kmod-usb-net-rndis - 5.15.167-1
kmod-usb2 - 5.15.167-1
```

Any ideas? Thanks

sndcrde, 2025-01-17:
Tried USB-tethering in v24.10.rc5. The connection seems to be even less stable than in v23.05.5, becomes unusable after about 10-15 minutes (ping times increase to ~1s, considerable packet loss).

```
## Normal ping times immediately after the USB-tethering was established
# ping 1.1.1.1
PING 1.1.1.1 (1.1.1.1) 56(84) bytes of data.
64 bytes from 1.1.1.1: icmp_seq=1 ttl=52 time=34.8 ms
64 bytes from 1.1.1.1: icmp_seq=2 ttl=52 time=32.4 ms
64 bytes from 1.1.1.1: icmp_seq=3 ttl=52 time=30.8 ms
64 bytes from 1.1.1.1: icmp_seq=4 ttl=52 time=29.5 ms
64 bytes from 1.1.1.1: icmp_seq=5 ttl=52 time=32.6 ms
^C
--- 1.1.1.1 ping statistics ---
5 packets transmitted, 5 received, 0% packet loss, time 4007ms
rtt min/avg/max/mdev = 29.454/32.011/34.828/1.815 ms
```

```
## Ping times after browsing the net for 15 minutes (regular sites, github, openwrt, et)
# ping 1.1.1.1
PING 1.1.1.1 (1.1.1.1) 56(84) bytes of data.
64 bytes from 1.1.1.1: icmp_seq=2 ttl=52 time=48.0 ms
64 bytes from 1.1.1.1: icmp_seq=1 ttl=52 time=2048 ms
64 bytes from 1.1.1.1: icmp_seq=3 ttl=52 time=1047 ms
64 bytes from 1.1.1.1: icmp_seq=4 ttl=52 time=1045 ms
64 bytes from 1.1.1.1: icmp_seq=5 ttl=52 time=51.3 ms
64 bytes from 1.1.1.1: icmp_seq=6 ttl=52 time=1073 ms
64 bytes from 1.1.1.1: icmp_seq=7 ttl=52 time=1048 ms
64 bytes from 1.1.1.1: icmp_seq=8 ttl=52 time=1048 ms
64 bytes from 1.1.1.1: icmp_seq=9 ttl=52 time=1045 ms
^C
--- 1.1.1.1 ping statistics ---
10 packets transmitted, 9 received, 10% packet loss, time 9041ms
rtt min/avg/max/mdev = 48.019/939.256/2048.239/566.472 ms, pipe 3
```

sndcrde, 2025-01-20:
```
# logread -f

---> Pure openwrt 24.10.rc5, no additional modules are installed
---> The phone is connected to the router via USB

# logread -f
Sat Jan 18 06:36:13 2025 kern.info kernel: [ 1151.154383] usb 1-1: USB disconnect, device number 6
Sat Jan 18 06:36:13 2025 kern.info kernel: [ 1151.506630] usb 1-1: new high-speed USB device number 7 using ehci-platform
Sat Jan 18 06:36:26 2025 kern.info kernel: [ 1164.114268] usb 1-1: USB disconnect, device number 7
Sat Jan 18 06:36:26 2025 kern.info kernel: [ 1164.466628] usb 1-1: new high-speed USB device number 8 using ehci-platform
Sat Jan 18 06:36:39 2025 kern.info kernel: [ 1177.234131] usb 1-1: USB disconnect, device number 8
Sat Jan 18 06:36:39 2025 kern.info kernel: [ 1177.586633] usb 1-1: new high-speed USB device number 9 using ehci-platform
Sat Jan 18 06:43:00 2025 kern.info kernel: [ 1558.239632] usb 1-1: USB disconnect, device number 9
Sat Jan 18 06:43:00 2025 kern.info kernel: [ 1558.596631] usb 1-1: new high-speed USB device number 10 using ehci-platform
Sat Jan 18 06:44:57 2025 kern.info kernel: [ 1675.676503] usb 1-1: USB disconnect, device number 10
Sat Jan 18 06:44:58 2025 kern.info kernel: [ 1676.026624] usb 1-1: new high-speed USB device number 11 using ehci-platform
Sat Jan 18 06:47:04 2025 kern.info kernel: [ 1802.508125] usb 1-1: USB disconnect, device number 11
Sat Jan 18 06:47:05 2025 kern.info kernel: [ 1802.866632] usb 1-1: new high-speed USB device number 12 using ehci-platform
Sat Jan 18 06:47:05 2025 kern.err kernel: [ 1803.336649] usb 1-1: device descriptor read/64, error -71
Sat Jan 18 06:47:17 2025 kern.info kernel: [ 1815.468004] usb 1-1: USB disconnect, device number 12
Sat Jan 18 06:47:18 2025 kern.info kernel: [ 1815.816623] usb 1-1: new high-speed USB device number 13 using ehci-platform
Sat Jan 18 06:47:18 2025 kern.err kernel: [ 1816.466634] usb 1-1: device not accepting address 13, error -71
Sat Jan 18 06:47:19 2025 kern.info kernel: [ 1816.836635] usb 1-1: new high-speed USB device number 14 using ehci-platform
Sat Jan 18 06:47:30 2025 kern.info kernel: [ 1828.587882] usb 1-1: USB disconnect, device number 14
Sat Jan 18 06:47:31 2025 kern.info kernel: [ 1828.936647] usb 1-1: new high-speed USB device number 15 using ehci-platform
Sat Jan 18 06:47:31 2025 kern.err kernel: [ 1829.406625] usb 1-1: device descriptor read/64, error -71
Sat Jan 18 06:49:30 2025 kern.info kernel: [ 1948.081898] usb 1-1: USB disconnect, device number 15
Sat Jan 18 06:49:30 2025 kern.info kernel: [ 1948.436678] usb 1-1: new high-speed USB device number 16 using ehci-platform
Sat Jan 18 06:50:27 2025 kern.info kernel: [ 2005.711251] usb 1-1: USB disconnect, device number 16
Sat Jan 18 06:50:28 2025 kern.info kernel: [ 2006.066628] usb 1-1: new high-speed USB device number 17 using ehci-platform
Sat Jan 18 06:50:41 2025 kern.info kernel: [ 2018.831130] usb 1-1: USB disconnect, device number 17
Sat Jan 18 06:50:41 2025 kern.info kernel: [ 2019.186631] usb 1-1: new high-speed USB device number 18 using ehci-platform
Sat Jan 18 06:50:54 2025 kern.info kernel: [ 2031.790999] usb 1-1: USB disconnect, device number 18
Sat Jan 18 06:50:54 2025 kern.info kernel: [ 2032.146624] usb 1-1: new high-speed USB device number 19 using ehci-platform
Sat Jan 18 06:51:07 2025 kern.info kernel: [ 2044.870879] usb 1-1: USB disconnect, device number 19
Sat Jan 18 06:51:07 2025 kern.info kernel: [ 2045.226660] usb 1-1: new high-speed USB device number 20 using ehci-platform
Sat Jan 18 06:51:19 2025 kern.info kernel: [ 2057.710748] usb 1-1: USB disconnect, device number 20
Sat Jan 18 06:51:20 2025 kern.info kernel: [ 2058.066623] usb 1-1: new high-speed USB device number 21 using ehci-platform
Sat Jan 18 06:51:20 2025 kern.err kernel: [ 2058.536624] usb 1-1: device descriptor read/64, error -71
Sat Jan 18 06:51:33 2025 kern.info kernel: [ 2070.990631] usb 1-1: USB disconnect, device number 21
Sat Jan 18 06:51:33 2025 kern.info kernel: [ 2071.346625] usb 1-1: new high-speed USB device number 22 using ehci-platform
Sat Jan 18 06:51:46 2025 kern.info kernel: [ 2083.960502] usb 1-1: USB disconnect, device number 22
Sat Jan 18 06:51:46 2025 kern.info kernel: [ 2084.316631] usb 1-1: new high-speed USB device number 23 using ehci-platform
Sat Jan 18 06:51:47 2025 kern.err kernel: [ 2084.786621] usb 1-1: device descriptor read/64, error -71
Sat Jan 18 06:51:59 2025 kern.info kernel: [ 2097.120377] usb 1-1: USB disconnect, device number 23
Sat Jan 18 06:51:59 2025 kern.info kernel: [ 2097.476625] usb 1-1: new high-speed USB device number 24 using ehci-platform
Sat Jan 18 06:52:12 2025 kern.info kernel: [ 2110.200261] usb 1-1: USB disconnect, device number 24
Sat Jan 18 06:52:12 2025 kern.info kernel: [ 2110.556625] usb 1-1: new high-speed USB device number 25 using ehci-platform
Sat Jan 18 06:52:13 2025 kern.err kernel: [ 2111.026621] usb 1-1: device descriptor read/64, error -71
Sat Jan 18 06:52:25 2025 kern.info kernel: [ 2123.310131] usb 1-1: USB disconnect, device number 25
Sat Jan 18 06:52:25 2025 kern.info kernel: [ 2123.666622] usb 1-1: new high-speed USB device number 26 using ehci-platform
Sat Jan 18 06:52:30 2025 daemon.notice netifd: wwan_cell_hotspot (3460): udhcpc: sending renew to server 192.168.85.198
Sat Jan 18 06:52:30 2025 daemon.notice netifd: wwan_cell_hotspot (3460): udhcpc: lease of 192.168.85.53 obtained from 192.168.85.198, lease time 3600
[... cut ...]
Sat Jan 18 06:52:39 2025 kern.err kernel: [ 2137.086648] usb 1-1: device descriptor read/64, error -71
Sat Jan 18 06:52:51 2025 kern.info kernel: [ 2149.389883] usb 1-1: USB disconnect, device number 27
Sat Jan 18 06:52:52 2025 kern.info kernel: [ 2149.746619] usb 1-1: new high-speed USB device number 28 using ehci-platform
Sat Jan 18 06:54:02 2025 kern.info kernel: [ 2220.394130] usb 1-1: USB disconnect, device number 28
Sat Jan 18 06:54:03 2025 kern.info kernel: [ 2220.746621] usb 1-1: new high-speed USB device number 29 using ehci-platform

```

sndcrde, 2025-01-20:
See the log below, for some reason can't attache files so will paste it below.

The USB connection became unusable when I tried to open a folder containing several (~10) bookmarks. It may be a coincidence that the connection became  corrupted/unusable at this time or may be it's because the phone's USB driver implementation is poorly written and isn't thread safe.

Once the USB connection becomes corrupted, a phone reboot is required to get it usable again.

Any ideas? Thanks

```
# logread -f

---> openwrt 24.10.rc5 with kmod-usb-net-rndis installed
---> Reboot the phone, enable USB-tethering and start browsing

Sat Jan 18 05:49:13 2025 kern.info kernel: [ 1028.120980] usb 1-1: USB disconnect, device number 7
Sat Jan 18 05:49:13 2025 kern.info kernel: [ 1028.128418] rndis_host 1-1:1.0 eth1: unregister 'rndis_host' usb-1b000000.usb-1, RNDIS device
Sat Jan 18 05:49:14 2025 kern.info kernel: [ 1028.736772] usb 1-1: new high-speed USB device number 8 using ehci-platform
Sat Jan 18 05:49:14 2025 kern.info kernel: [ 1028.978733] rndis_host 1-1:1.0 eth1: register 'rndis_host' at usb-1b000000.usb-1, RNDIS device, e6:bf:f1:d6:6b:42
Sat Jan 18 05:49:14 2025 daemon.notice netifd: Interface 'wan_cell_usb' is disabled
Sat Jan 18 05:49:14 2025 daemon.notice netifd: Network device 'eth1' link is down
Sat Jan 18 05:49:14 2025 daemon.notice netifd: Interface 'wan_cell_usb' has link connectivity loss
Sat Jan 18 05:49:14 2025 daemon.notice netifd: wan_cell_usb (3183): udhcpc: received SIGTERM
Sat Jan 18 05:49:14 2025 daemon.notice netifd: wan_cell_usb (3183): udhcpc: unicasting a release of 192.168.123.25 to 192.168.123.125
Sat Jan 18 05:49:14 2025 daemon.notice netifd: wan_cell_usb (3183): udhcpc: sending release
Sat Jan 18 05:49:14 2025 daemon.notice netifd: wan_cell_usb (3183): udhcpc: bind(UDP): Address not available
Sat Jan 18 05:49:14 2025 daemon.notice netifd: wan_cell_usb (3183): udhcpc: entering released state
Sat Jan 18 05:49:14 2025 daemon.notice netifd: wan_cell_usb (3183): Command failed: ubus call network.interface notify_proto { "action": 0, "link-up": false, "keep": false, "interface": "wan_cell_usb" } (Permission denied)
Sat Jan 18 05:49:14 2025 daemon.notice netifd: Interface 'wan_cell_usb' is now down
Sat Jan 18 05:49:14 2025 daemon.info dnsmasq[1]: reading /tmp/resolv.conf.d/resolv.conf.auto
Sat Jan 18 05:49:14 2025 daemon.info dnsmasq[1]: using nameserver 67.80.155.180#53
Sat Jan 18 05:49:14 2025 daemon.info dnsmasq[1]: using only locally-known addresses for test
Sat Jan 18 05:49:14 2025 daemon.info dnsmasq[1]: using only locally-known addresses for onion
Sat Jan 18 05:49:14 2025 daemon.info dnsmasq[1]: using only locally-known addresses for localhost
Sat Jan 18 05:49:14 2025 daemon.info dnsmasq[1]: using only locally-known addresses for local
Sat Jan 18 05:49:14 2025 daemon.info dnsmasq[1]: using only locally-known addresses for invalid
Sat Jan 18 05:49:14 2025 daemon.info dnsmasq[1]: using only locally-known addresses for bind
Sat Jan 18 05:49:14 2025 daemon.info dnsmasq[1]: using only locally-known addresses for lan
Sat Jan 18 05:49:14 2025 daemon.notice netifd: Interface 'wan_cell_usb' is enabled
Sat Jan 18 05:49:14 2025 daemon.notice netifd: Network device 'eth1' link is up
Sat Jan 18 05:49:14 2025 daemon.notice netifd: Interface 'wan_cell_usb' has link connectivity
Sat Jan 18 05:49:14 2025 daemon.notice netifd: Interface 'wan_cell_usb' is setting up now
Sat Jan 18 05:49:14 2025 daemon.notice netifd: wan_cell_usb (3388): udhcpc: started, v1.36.1
Sat Jan 18 05:49:14 2025 daemon.notice netifd: wan_cell_usb (3388): udhcpc: broadcasting discover
Sat Jan 18 05:49:14 2025 daemon.notice netifd: wan_cell_usb (3388): udhcpc: broadcasting select for 192.168.123.25, server 192.168.123.125
Sat Jan 18 05:49:14 2025 daemon.notice netifd: wan_cell_usb (3388): udhcpc: lease of 192.168.123.25 obtained from 192.168.123.125, lease time 3599
Sat Jan 18 05:49:15 2025 daemon.notice netifd: Interface 'wan_cell_usb' is now up
Sat Jan 18 05:49:15 2025 daemon.info dnsmasq[1]: reading /tmp/resolv.conf.d/resolv.conf.auto
Sat Jan 18 05:49:15 2025 daemon.info dnsmasq[1]: using nameserver 67.80.155.180#53
Sat Jan 18 05:49:15 2025 daemon.info dnsmasq[1]: using only locally-known addresses for test
Sat Jan 18 05:49:15 2025 daemon.info dnsmasq[1]: using only locally-known addresses for onion
Sat Jan 18 05:49:15 2025 daemon.info dnsmasq[1]: using only locally-known addresses for localhost
Sat Jan 18 05:49:15 2025 daemon.info dnsmasq[1]: using only locally-known addresses for local
Sat Jan 18 05:49:15 2025 daemon.info dnsmasq[1]: using only locally-known addresses for invalid
Sat Jan 18 05:49:15 2025 daemon.info dnsmasq[1]: using only locally-known addresses for bind
Sat Jan 18 05:49:15 2025 daemon.info dnsmasq[1]: using only locally-known addresses for lan
Sat Jan 18 05:49:15 2025 user.notice firewall: Reloading firewall due to ifup of wan_cell_usb (eth1)
Sat Jan 18 05:52:17 2025 kern.info kernel: [ 1211.929356] usb 1-1: USB disconnect, device number 8
Sat Jan 18 05:52:17 2025 kern.info kernel: [ 1211.938096] rndis_host 1-1:1.0 eth1: unregister 'rndis_host' usb-1b000000.usb-1, RNDIS device
Sat Jan 18 05:52:17 2025 kern.info kernel: [ 1212.326751] usb 1-1: new high-speed USB device number 9 using ehci-platform
Sat Jan 18 05:52:17 2025 kern.info kernel: [ 1212.570048] rndis_host 1-1:1.0 eth1: register 'rndis_host' at usb-1b000000.usb-1, RNDIS device, e6:bf:f1:d6:6b:42
[... cut ...]
64 bytes from 1.1.1.1: icmp_seq=5 ttl=52 time=1033 ms
64 bytes from 1.1.1.1: icmp_seq=6 ttl=52 time=1034 ms
64 bytes from 1.1.1.1: icmp_seq=7 ttl=52 time=356 ms
From 10.10.11.11 icmp_seq=8 Destination Net Unreachable
From 10.10.11.11 icmp_seq=9 Destination Net Unreachable
From 10.10.11.11 icmp_seq=10 Destination Net Unreachable
^C
--- 1.1.1.1 ping statistics ---
10 packets transmitted, 5 received, +3 errors, 50% packet loss, time 9092ms
rtt min/avg/max/mdev = 166.355/698.279/1034.330/365.126 ms, pipe 2

```

ddubovoy, 2025-03-04:
I have the same problem on ASUS RT-AC58U with OpenWrt 24.10.0 r28427-6df0e3d02a

Did you find a solution for this?

sndcrde, 2025-03-06:
> I have the same problem on ASUS RT-AC58U with OpenWrt 24.10.0 r28427-6df0e3d02a
> 
> Did you find a solution for this?

I have upgraded to v24.10.0 but the issue still persists, the USB-tethered connection becomes unstable for me in about 15 minutes.

My solution was to use the phone's WiFi hotspot instead of USB tethering.