openwrt/packages #25787: fail2ban 0.11.2-8 crushing on 24.10.0-rc5 - r28304-6dacba30a7

michalperth, 2025-01-19:
Hi, I'm very new to OpenWRT and basic in Linux, so please be patient. Runs .
I installed fail2ban on the OpenWRT ONE and it runs ok if none of the jails are enabled. Even single enabled, will crush the service.
The only things I found so far is in the system log:
daemon.info procd: Instance fail2ban::instance1 s in a crash loop 7 crashes, 1 seconds since last crash
 and:
[5653]: INFO    Connected to fail2ban persistent database '/var/lib/fail2ban/fail2ban.sqlite3'
2025-01-19 12:14:55,738 fail2ban.database       [5653]: WARNING New database created. Version '4'
in the fail2ban log.

Any single one enabled, will just crush the service.
For example the single entry  in the jail.local:

[sshd]
enabled = true
mode   = normal
port    = ssh
logpath = %(sshd_log)s
backend = %(sshd_backend)s

All the settings are just taken from the standard, not altered fail2ban config files for now.

will do the crushing trick.

Thanks.

tojestzart, 2025-01-21:
try banip instead

michalperth, 2025-01-22:
Ok, thanks for the suggestion, I will certainly have a look in to it. I tried the fail2ban because I'm bit  familliar with it, it is banning hundreds of IPs on my other Raspberry CM4 based router (running Raspbian).
New challange I suppose (My Linux knowlwdge is very basic).
Cheers.

michalperth, 2025-01-22:
hmmm, does banip actually scans the logs and adds bans automatically to the iptables rules as fail2ban does, or you need to add the IPs manually?
Thanks

tojestzart, 2025-01-22:
ask on our forum [eko.one.pl ](https://eko.one.pl/forum/) as far i know it bans by filters + patterns + ranges + ????

michalperth, 2025-01-23:
Ok, thanks Mr. "to jest żart" 😉

closed 2025-05-20 by BKPepe