openwrt/packages #25792: [PBR] network is not restored after interface is down

sinancetinkaya, 2025-01-19:
@stangri Can you take a look?

`/etc/config/pbr`
```
config pbr 'config'
	option enabled '1'
	option verbosity '2'
	option strict_enforcement '0'
	option resolver_set 'none'
	list resolver_instance '*'
	option ipv6_enabled '0'
	list ignored_interface 'vpnserver'
	option boot_timeout '30'
	option rule_create_option 'add'
	option procd_boot_delay '0'
	option procd_reload_delay '1'
	option webui_show_ignore_target '0'
	option nft_rule_counter '0'
	option nft_set_auto_merge '1'
	option nft_set_counter '0'
	option nft_set_flags_interval '1'
	option nft_set_flags_timeout '0'
	option nft_set_policy 'performance'
	list webui_supported_protocol 'all'
	list webui_supported_protocol 'tcp'
	list webui_supported_protocol 'udp'
	list webui_supported_protocol 'tcp udp'
	list webui_supported_protocol 'icmp'

config include
	option path '/usr/share/pbr/pbr.user.aws'
	option enabled '0'

config include
	option path '/usr/share/pbr/pbr.user.netflix'
	option enabled '0'

config dns_policy
	option name 'Redirect Local IP DNS'
	option src_addr '192.168.1.5'
	option dest_dns '1.1.1.1'
	option enabled '0'

config policy
	option name 'Ignore Local Requests'
	option interface 'wan'
	option dest_addr '192.168.0.0/16'
	option enabled '0'

config policy
	option name 'Discord'
	option dest_addr 'discord.com discord.gg discordapp.net discordapp.com discord.media'
	option interface 'cloudflare'

config policy
	option name 'LTE'
	option interface 'wwan'
	option src_addr '192.168.1.2 192.168.1.7'

config policy
	option name 'all'
	option interface 'cloudflare'
	option dest_addr '0.0.0.0/0'
	option enabled '0'

```
```
# lets do a clean start
root@router-sinan:~# service pbr restart  
Resetting chains and sets [✓]
[... 111 lines cut ...]
Sun Jan 19 22:50:59 2025 daemon.notice wpa_supplicant[2566]: phy1-sta0: Removed BSSID XX:XX:XX:XX:XX:XX from ignore list (clear)

# network gets completely down
root@router-sinan:~# ping 8.8.8.8
PING 8.8.8.8 (8.8.8.8): 56 data bytes
ping: sendto: Network unreachable
root@router-sinan:~# ping 8.8.8.8
PING 8.8.8.8 (8.8.8.8): 56 data bytes
ping: sendto: Network unreachable

# only restarting network service fixes the problem
root@router-sinan:~# service network restart
root@router-sinan:~# ping 8.8.8.8
PING 8.8.8.8 (8.8.8.8): 56 data bytes
64 bytes from 8.8.8.8: seq=0 ttl=119 time=25.273 ms
64 bytes from 8.8.8.8: seq=1 ttl=119 time=24.927 ms
64 bytes from 8.8.8.8: seq=2 ttl=119 time=24.471 ms
64 bytes from 8.8.8.8: seq=3 ttl=119 time=24.681 ms
^C
--- 8.8.8.8 ping statistics ---
4 packets transmitted, 4 packets received, 0% packet loss
round-trip min/avg/max = 24.471/24.838/25.273 ms
root@router-sinan:~#

```

stangri, 2025-01-20:
What am I looking at?

sinancetinkaya, 2025-01-20:
> What am I looking at?

Please read `# comments` in the shell screen
After the `wwan` gets down' pbr makes the entire network down

stangri, 2025-01-21:
> After the `wwan` gets down, pbr makes the entire network down

README->Getting Help in the "entire network down" state please.

sinancetinkaya, 2025-01-21:
> > After the `wwan` gets down, pbr makes the entire network down
> 
> README->Getting Help in the "entire network down" state please.

Okay, let me simplify the first post, so you can understand.
I know you are going to mention this. No, it doesn't fix the problem https://docs.openwrt.melmac.net/pbr/1.1.8/#AWordAboutInterfaceHotplugScript