openwrt/packages #25797: HTTPS-DNS-Proxy not starting properly after reboot

ob45, 2025-01-20:
## Describe the bug

After restarting the router the HTTPS-DNS-Proxy service does not start and work properly despite it being set to "Enabled".

It does show up as "Enabled" and the service appears to be running but the service **_does not actually work_**.

I have to manually press the button "Restart" to make it run.

It is not possible that this is a RAM issue because enough RAM is available.

[image]

## Steps to reproduce

1. Install HTTPS-DNS-Proxy
2. Configure the https-DNS servers in the service tab
3. Restart the router
4. Look at the HTTPS DNS Proxy tab and check the U.I
5. Check if HTTPS over DNS is working (it is not)

I've tried waiting up to half an hour after the restart because "maybe it just takes some time" but the only thing which actually makes it work is restarting the service manually - after which all blocked sites become accessible.

## Actual behaviour
The service appears to be running in the U.I but is extremely buggy.
I have to manually restart the service after restarting the router because the service does not automatically start properly.

Stopping the service can lead to nothing happening and the U.I keep showing that it is running.

[image]

## Expected behaviour
The service should start and work automatically after a router restart and route all DNS requests to the configured DNS servers and use DNS over HTTPS when the service is set as "Enabled".

The U.I should show th ecorrect status of the service.

## Additional info

This is my device and formware:

- Device:  Xiaomi Mi Router 4A (100M International Edition V2)
- OpenWrt Version: r28304-6dacba30a7
- OpenWrt release: 24.10.0-rc5
- OpenWrt target/subtarget: ramips/mt76x8

This is a 64MB device but as shown in the screenshot enough RAM is available.

For me it is quite easy to check if it works or not because my ISP DNS blocks certain sites, which I cannot access without using DNS over HTTPS so I can just try to hit the site after a router restart and easily check it that way too.

I'm unsure if I am encountering the same issue like https://github.com/openwrt/packages/issues/23469 because it seems to be similar but in my case something gets loaded, just not the entire config.

During normal operation:

```
root@OpenWrt:~# uci show dhcp.@dnsmasq[0].server
dhcp.cfg01411c.server='127.0.0.1#5053' '127.0.0.1#5054' '/mask.icloud.com/' '/mask-h2.icloud.com/' '/use-application-dns.net/' '127.0.0.1#5050' '127.0.0.1#5051' '127.0.0.1#5052'
```

After a reboot:

```
root@OpenWrt:~# uci show dhcp.@dnsmasq[0].server
dhcp.cfg01411c.server='127.0.0.1#5053' '127.0.0.1#5054' '/mask.icloud.com/' '/mask-h2.icloud.com/' '/use-application-dns.net/'
```

Looks like **_something_** gets loaded but not the entire / my config.

After restarting the service it looks normal again:

```
root@OpenWrt:~# uci show dhcp.@dnsmasq[0].server
dhcp.cfg01411c.server='127.0.0.1#5053' '127.0.0.1#5054' '/mask.icloud.com/' '/mask-h2.icloud.com/' '/use-application-dns.net/' '127.0.0.1#5050' '127.0.0.1#5051' '127.0.0.1#5052'
```

Idk what to make of this. The op in https://github.com/openwrt/packages/issues/23469 did say he got nothing at all... meanwhile I do get something but not my entire valid config and I have to manually restart the service each time after reboot.

stangri, 2025-01-20:
@ob45: https://docs.openwrt.melmac.net/https-dns-proxy/#GettingHelp

ob45, 2025-01-21:
> [@ob45](https://github.com/ob45): https://docs.openwrt.melmac.net/https-dns-proxy/#GettingHelp

After device restart:

```
root@OpenWrt:~# ubus call system board
{
        "kernel": "6.6.69",
        "hostname": "OpenWrt",
        "system": "MediaTek MT7628AN ver:1 eco:2",
        "model": "Xiaomi Mi Router 4A (100M International Edition V2)",
        "board_name": "xiaomi,mi-router-4a-100m-intl-v2",
        "rootfs_type": "squashfs",
        "release": {
                "distribution": "OpenWrt",
                "version": "24.10.0-rc5",
                "revision": "r28304-6dacba30a7",
                "target": "ramips/mt76x8",
                "description": "OpenWrt 24.10.0-rc5 r28304-6dacba30a7",
                "builddate": "1736026537"
        }
}
root@OpenWrt:~# curl -V
curl 8.10.1 (mipsel-openwrt-linux-gnu) libcurl/8.10.1 mbedTLS/3.6.2 nghttp2/1.63.0
Release-Date: 2024-09-18
Protocols: file ftp ftps http https ipfs ipns mqtt
Features: alt-svc HSTS HTTP2 HTTPS-proxy IPv6 Largefile SSL threadsafe UnixSockets
root@OpenWrt:~# dnsmasq --version
Dnsmasq version 2.90  Copyright (c) 2000-2024 Simon Kelley
Compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP no-DHCPv6 no-Lua TFTP no-conntrack no-ipset no-nftset no-auth no-cryptohash no-DNSSEC no-ID loop-detect inotify dumpfile

This software comes with ABSOLUTELY NO WARRANTY.
Dnsmasq is free software, and you are welcome to redistribute it
under the terms of the GNU General Public License, version 2 or 3.
root@OpenWrt:~# https-dns-proxy -V
2023.12.26-1
root@OpenWrt:~# uci export dhcp
package dhcp

config dnsmasq
        option domainneeded '1'
        option boguspriv '1'
        option filterwin2k '0'
        option localise_queries '1'
        option rebind_protection '1'
        option rebind_localhost '1'
        option local '/lan/'
        option domain 'lan'
        option expandhosts '1'
        option nonegcache '0'
        option cachesize '1000'
        option authoritative '1'
        option readethers '1'
        option leasefile '/tmp/dhcp.leases'
        option resolvfile '/tmp/resolv.conf.d/resolv.conf.auto'
        option nonwildcard '1'
        option localservice '1'
        option ednspacket_max '1232'
        option filter_aaaa '0'
        option filter_a '0'
        list server '127.0.0.1#5053'
        list server '127.0.0.1#5054'
        list server '/mask.icloud.com/'
        list server '/mask-h2.icloud.com/'
        list server '/use-application-dns.net/'

config dhcp 'lan'
        option interface 'lan'
        option start '100'
        option limit '150'
[... 615 lines cut ...]
                        [
                                "config.change",
                                [
                                        "if",
                                        [
                                                "eq",
                                                "package",
                                                "https-dns-proxy"
                                        ],
                                        [
                                                "run_script",
                                                "/etc/init.d/https-dns-proxy",
                                                "reload",
                                                "on_config_change"
                                        ]
                                ],
                                1000
                        ]
                ]
        }
}
root@OpenWrt:~# nslookup google.com 127.0.0.1:5053
nslookup: write to '127.0.0.1:5053': Connection refused
;; connection timed out; no servers could be reached
```

stangri, 2025-01-21:
There may be an issue when no servers are defined for the dnsmasq instance.

If you capture the output of `logread -ehttps-dns-proxy` -- right after reboot and the once again after service restart, what does it say? The default behaviour is to stop the service if it's started too early and with no servers defined for dnsmasq your DNS would be broken then.

Angel996, 2025-01-21:
Confirmed, got exactly the same issue. Just installed _https-dns-proxy_ today.

ob45, 2025-01-21:
> There may be an issue when no servers are defined for the dnsmasq instance.
> 
> If you capture the output of `logread -ehttps-dns-proxy` -- right after reboot and the once again after service restart, what does it say?

Right after device reboot:

```
root@OpenWrt:~# logread -ehttps-dns-proxy
Tue Jan 21 17:56:34 2025 user.notice https-dns-proxy: Starting service instances on_boot ✓✓✓
Tue Jan 21 17:56:35 2025 user.notice https-dns-proxy: Updating dnsmasq config ✓
Tue Jan 21 17:56:42 2025 user.notice https-dns-proxy: Restarting dnsmasq on_config_update ✓
Tue Jan 21 17:56:43 2025 user.notice https-dns-proxy: Setting service raw_trigger for 'interface.*.up'✓
Tue Jan 21 17:56:58 2025 user.notice https-dns-proxy: Stopping service on_failed_health_check ✓
Tue Jan 21 17:56:58 2025 user.notice https-dns-proxy: Setting service raw_trigger for 'interface.*.up'✓
```

After manually restarting the service:

```
root@OpenWrt:~# logread -ehttps-dns-proxy
Tue Jan 21 17:56:34 2025 user.notice https-dns-proxy: Starting service instances on_boot ✓✓✓
Tue Jan 21 17:56:35 2025 user.notice https-dns-proxy: Updating dnsmasq config ✓
Tue Jan 21 17:56:42 2025 user.notice https-dns-proxy: Restarting dnsmasq on_config_update ✓
Tue Jan 21 17:56:43 2025 user.notice https-dns-proxy: Setting service raw_trigger for 'interface.*.up'✓
Tue Jan 21 17:56:58 2025 user.notice https-dns-proxy: Stopping service on_failed_health_check ✓
Tue Jan 21 17:56:58 2025 user.notice https-dns-proxy: Setting service raw_trigger for 'interface.*.up'✓
Tue Jan 21 18:01:33 2025 user.notice https-dns-proxy: Starting service instances ✓✓✓
Tue Jan 21 18:01:33 2025 user.notice https-dns-proxy: Updating dnsmasq config ✓
Tue Jan 21 18:01:40 2025 user.notice https-dns-proxy: Restarting dnsmasq on_config_update ✓
```

ob45, 2025-01-21:
> I don't understand why you have:
> 
> ```
>         list server '127.0.0.1#5053'
>         list server '127.0.0.1#5054'
>         list server '/mask.icloud.com/'
>         list server '/mask-h2.icloud.com/'
>         list server '/use-application-dns.net/'
> ```
> 
> In your dhcp config when the service isn't running. It doesn't look like you've added those manually, it may be an artefact from the older version not cleaning up after itself.

I don't understand either. I haven't added them and it can't be an artefact from an older version because it is a fresh OpenWrt installation, where I had to add this service again.

I had this issue on the 24.10.0-rc4 and 24.10.0-rc3 release too...

I am unsre but if I remember correctly I didn't have this issue on one of the older LuCI less snapshots of the device.

stangri, 2025-01-21:
> it can't be an artefact from an older version because it is a fresh OpenWrt installation, where I had to add this service again.

I'll have to review the code, but if it's a fresh install it's definitely a bug most likely due to the fact that no servers are defined for dnsmasq instance before the proxy starts.

ve1mi, 2025-01-21:
> I'll have to review the code, but if it's a fresh install it's definitely a bug most likely due to the fact that no servers are defined for dnsmasq instance before the proxy starts.
I have the same problem on router versions 4ac, 4c. Tested on both versions on both routers 23.05.5 and 24.10 rc5. 
The problem is with full static from the provider, as well as with dhcp from another provider

Angel996, 2025-01-21:
I have these entries too, fresh _https-dns-proxy_ install, never used it before. The lines also appear in _/etc/config/dhcp_ file.

ob45, 2025-01-22:
> The problem is full static from the provider, as well as dhcp from another provider.'

Do you mean a static IP address from the ISP? I live in a student dorm and all rooms have a static IP assigned to them so maybe that's one of the reasons?

ve1mi, 2025-01-22:
> Do you mean a static IP address from the ISP? I live in a student dorm and all rooms have a static IP assigned to them so maybe that's one of the reasons?

This is a bug in this package, it is not about isp and router settings. The same problem on different isp (static and dhcp)

sinancetinkaya, 2025-01-22:
> I don't understand either. I haven't added them and it can't be an artefact from an older version because it is a fresh OpenWrt installation, where I had to add this service again.
>
> I had this issue [...]

He adds them himself 🤦‍♀️ https://github.com/stangri/https-dns-proxy/blob/db135baeca1a9fa263ff77c66dcd56ccd0a77e5d/files/etc/init.d/https-dns-proxy#L362

ob45, 2025-01-22:
> He adds them himself 🤦‍♀️ https://github.com/stangri/https-dns-proxy/blob/db135baeca1a9fa263ff77c66dcd56ccd0a77e5d/files/etc/init.d/https-dns-proxy#L362

I just looked the entries up and it seems like you're right 😂

Now we just need to know why this default config gets loaded on every restart instead of our actual config ^_^

stangri, 2025-01-22:
> I haven't added them and it can't be an artefact from an older version because it is a fresh OpenWrt installation, where I had to add this service again.

Can you walk me thru steps how you got from default install to your current setup?

For the issue at hand -- can you try:
1. stop https-dns-proxy
2. remove the following from dhcp config
```
        list server '127.0.0.1#5053'
        list server '127.0.0.1#5054'
```
3. start https-dns-proxy, make sure it works
4. reboot your router, immediately after it boots up capture the output of `logread -ehttps`
5. check if dns resolution works

ob45, 2025-01-23:
> Can you walk me thru steps how you got from default install to your current setup?

Sure!

1. I just open LuCI, go to System > Software
2. Click on "Update Lists"
3. After it is done updating the list I search for "HTTPS-DNS-Proxy"
4. Then I download "luci-app-https-dns-proxy" because it installs the "HTTPS-DNS-Proxy" package too
5. After it is done I lose the ability to  use the normal net because this package configures google and cloudflare DNS servers and my ISP is seething at the thought of people using their own DNS servers
6. I logout & login again for luci to load the "Services tab"
7. I go to Services > HTTPS-DNS-Proxy and delete the default DNS servers, which are pre configured
8. I configure the DNS servers, which can be seen above
9. Internet works again because these DNS servers haven't been blacklisted

That's basically it. After I restart the router my internet is gone again because the default config gets loaded with ISP banned DNS servers. Then I restart the service, which loads my DNS servers mentioned above and the internet works again.

Angel996, 2025-01-25:
Manual says after installation we must run:

```
root@OpenWrt:~# /etc/init.d/https-dns-proxy enable
root@OpenWrt:~# /etc/init.d/https-dns-proxy start
```

Does this actually do anything?

stangri, 2025-01-25:
> the default config gets loaded with ISP banned DNS servers.

Are you saying that after reboot, the https-dns-proxy config contains default resolvers and not the ones you have previosuly configured?

stangri, 2025-01-25:
> Manual says after installation we must run:

Refer to support thread on the forum with your question, this is not an AMA.

ob45, 2025-01-26:
> > the default config gets loaded with ISP banned DNS servers.
> 
> Are you saying that after reboot, the https-dns-proxy config contains default resolvers and not the ones you have previosuly configured?

I guess so, because I did not put these things here anywhere:

```
        list server '127.0.0.1#5053'
        list server '127.0.0.1#5054'
        list server '/mask.icloud.com/'
        list server '/mask-h2.icloud.com/'
        list server '/use-application-dns.net/'
```

ob45, 2025-01-26:
> Manual says after installation we must run:
> 
> ```
> root@OpenWrt:~# /etc/init.d/https-dns-proxy enable
> root@OpenWrt:~# /etc/init.d/https-dns-proxy start
> ```
> 
> Does this actually do anything?

It restarts the service. 

Enable is that it starts u on restart of the device and start is actually staring the service if I understand it correctly

stangri, 2025-01-27:
> > > the default config gets loaded with ISP banned DNS servers.
> > 
> > 
> > Are you saying that after reboot, the https-dns-proxy config contains default resolvers and not the ones you have previosuly configured?
> 

Earlier in the thread you stated:

> the default config gets loaded with ISP banned DNS servers.

So let me ask you again:

> > Are you saying that after reboot, the https-dns-proxy config contains default resolvers and not the ones you have previosuly configured?

ob45, 2025-01-30:
> So let me ask you again:
> > > Are you saying that after reboot, the https-dns-proxy config contains default resolvers and not the ones you have previosuly configured?

If you mean by "default resolvers" the standard DNS servers of Google and Cloudflare, which are preconfigured on a fresh install, then no.

I meant  these
```

        list server '127.0.0.1#5053'
        list server '127.0.0.1#5054'
        list server '/mask.icloud.com/'
        list server '/mask-h2.icloud.com/'
        list server '/use-application-dns.net/'
```

p0k33m0n, 2025-02-06:
After the update on 24.10 also something strange happened with this service. It did not start when booting the router. However, after rebooting and a few more restarts, the problem disappeared (for now). I don't know what to think about it myself, however, on the previous firmware I never had such a situation. Oh, the entries in the configuration file are correct.

ob45, 2025-02-06:
> After the update on 24.10 also something strange happened with this service. It did not start when booting the router. However, after rebooting and a few more restarts, the problem disappeared (for [...]

What device are you using?

p0k33m0n, 2025-02-06:
> What device are you using?

```
Model | Netgear WNDR4300
Architecture | Atheros AR9344 rev 2
Target Platform | ath79/nand
Firmware Version | OpenWrt 24.10.0 r28427-6df0e3d02a / LuCI openwrt-24.10 branch 25.035.62793~4f7a183
Kernel Version | 6.6.73
```

But it doesn't matter, today was the update of these packages, so maybe it will change something. I will let you know.

p0k33m0n, 2025-02-07:
Unfortunately, but still after a cold start, the service does not start.

`Thu Feb  6 11:19:53 2025 kern.info kernel: [    0.084972] pid_max: default: 32768 minimum: 301
Thu Feb  6 11:19:53 2025 kern.info kernel: [    0.103048] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Thu Feb  6 11:19:53 2025 kern.info kernel: [    0.110870] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Thu Feb  6 11:19:53 2025 kern.info kernel: [    0.130328] RCU Tasks Trace: Setting shift to 0 and lim to 1 rcu_task_cb_adjust=1 rcu_task_cpu_ids=1.
Thu Feb  6 11:19:53 2025 kern.info kernel: [    0.146990] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
Thu Feb  6 11:19:53 2025 kern.info kernel: [    0.157528] futex hash table entries: 256 (order: -1, 3072 bytes, linear)
Thu Feb  6 11:19:53 2025 kern.info kernel: [    0.168239] pinctrl core: initialized pinctrl subsystem
Thu Feb  6 11:19:53 2025 kern.info kernel: [    0.177869] NET: Registered PF_NETLINK/PF_ROUTE protocol family
Thu Feb  6 11:19:53 2025 kern.info kernel: [    0.184851] thermal_sys: Registered thermal governor 'step_wise'
Thu Feb  6 11:19:53 2025 kern.info kernel: [    0.187528] /ahb/apb: Fixed dependency cycle(s) with /ahb/apb/interrupt-controller@18060010
Thu Feb  6 11:19:53 2025 kern.info kernel: [    0.219064] clocksource: Switched to clocksource MIPS
Thu Feb  6 11:19:53 2025 kern.info kernel: [    0.238639] NET: Registered PF_INET protocol family
Thu Feb  6 11:19:53 2025 kern.info kernel: [    0.244189] IP idents hash table entries: 2048 (order: 2, 16384 bytes, linear)
Thu Feb  6 11:19:53 2025 kern.info kernel: [    0.253769] tcp_listen_portaddr_hash hash table entries: 1024 (order: 0, 4096 bytes, linear)
Thu Feb  6 11:19:53 2025 kern.info kernel: [    0.262856] Table-perturb hash table entries: 65536 (order: 6, 262144 bytes, linear)
Thu Feb  6 11:19:53 2025 kern.info kernel: [    0.271130] TCP established hash table entries: 1024 (order: 0, 4096 bytes, linear)
Thu Feb  6 11:19:53 2025 kern.info kernel: [    0.279303] TCP bind hash table entries: 1024 (order: 1, 8192 bytes, linear)
Thu Feb  6 11:19:53 2025 kern.info kernel: [    0.286837] TCP: Hash tables configured (established 1024 bind 1024)
Thu Feb  6 11:19:53 2025 kern.info kernel: [    0.294498] MPTCP token hash table entries: 512 (order: 0, 6144 bytes, linear)
Thu Feb  6 11:19:53 2025 kern.info kernel: [    0.302568] UDP hash table entries: 256 (order: 0, 4096 bytes, linear)
Thu Feb  6 11:19:53 2025 kern.info kernel: [    0.309579] UDP-Lite hash table entries: 256 (order: 0, 4096 bytes, linear)
Thu Feb  6 11:19:53 2025 kern.info kernel: [    0.318244] NET: Registered PF_UNIX/PF_LOCAL protocol family
Thu Feb  6 11:19:53 2025 kern.info kernel: [    0.324425] PCI: CLS 0 bytes, default 32
Thu Feb  6 11:19:53 2025 kern.info kernel: [    0.334402] workingset: timestamp_bits=14 max_order=15 bucket_order=1
Thu Feb  6 11:19:53 2025 kern.info kernel: [    0.343525] squashfs: version 4.0 (2009/01/31) Phillip Lougher
Thu Feb  6 11:19:53 2025 kern.info kernel: [    0.349795] jffs2: version 2.2 (NAND) (SUMMARY) (LZMA) (RTIME) (CMODE_PRIORITY) (c) 2001-2006 Red Hat, Inc.
Thu Feb  6 11:19:53 2025 kern.info kernel: [    0.370340] Block layer SCSI generic (bsg) driver version 0.4 loaded (major 251)
Thu Feb  6 11:19:53 2025 kern.info kernel: [    0.382657] pinctrl-single 1804002c.pinmux: 544 pins, size 68
Thu Feb  6 11:19:53 2025 kern.info kernel: [    0.390410] Serial: 8250/16550 driver, 16 ports, IRQ sharing enabled
Thu Feb  6 11:19:53 2025 kern.info kernel: [    0.403658] printk: console [ttyS0] disabled
Thu Feb  6 11:19:53 2025 kern.info kernel: [    0.408742] 18020000.uart: ttyS0 at MMIO 0x18020000 (irq = 11, base_baud = 2500000) is a 16550A
Thu Feb  6 11:19:53 2025 kern.info kernel: [    0.418120] printk: console [ttyS0] enabled
Thu Feb  6 11:19:53 2025 kern.info kernel: [    0.427036] printk: bootconsole [early0] disabled
Thu Feb  6 11:19:53 2025 kern.info kernel: [    0.475115] nand: device found, Manufacturer ID: 0x2c, Chip ID: 0xf1
Thu Feb  6 11:19:53 2025 kern.info kernel: [    0.481635] nand: Micron NAND 128MiB 3,3V 8-bit
Thu Feb  6 11:19:53 2025 kern.info kernel: [    0.486234] nand: 128 MiB, SLC, erase size: 128 KiB, page size: 2048, OOB size: 64
Thu Feb  6 11:19:53 2025 kern.info kernel: [    0.493982] Scanning device for bad blocks
Thu Feb  6 11:19:53 2025 kern.notice kernel: [    0.581478] 10 fixed-partitions partitions found on MTD device ar934x-nand
Thu Feb  6 11:19:53 2025 kern.notice kernel: [    0.588833] Creating 10 MTD partitions on "ar934x-nand":
Thu Feb  6 11:19:53 2025 kern.notice kernel: [    0.594289] 0x000000000000-0x000000040000 : "u-boot"
Thu Feb  6 11:19:53 2025 kern.notice kernel: [    0.602851] 0x000000040000-0x000000080000 : "u-boot-env"
Thu Feb  6 11:19:53 2025 kern.notice kernel: [    0.610734] 0x000000080000-0x0000000c0000 : "caldata"
Thu Feb  6 11:19:53 2025 kern.notice kernel: [    0.618834] 0x0000000c0000-0x000000140000 : "pot"
Thu Feb  6 11:19:53 2025 kern.notice kernel: [    0.626560] 0x000000140000-0x000000340000 : "language"
Thu Feb  6 11:19:53 2025 kern.notice kernel: [    0.634214] 0x000000340000-0x0000003c0000 : "config"
Thu Feb  6 11:19:53 2025 kern.notice kernel: [    0.641730] 0x0000003c0000-0x0000006c0000 : "traffic_meter"
Thu Feb  6 11:19:53 2025 kern.notice kernel: [    0.649977] 0x0000006c0000-0x000001fc0000 : "firmware"
[... cut ...]
Thu Feb  6 11:27:28 2025 daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 2 names
Thu Feb  6 11:27:28 2025 daemon.info dnsmasq-dhcp[1]: read /etc/ethers - 0 addresses
Thu Feb  6 11:27:28 2025 user.notice https-dns-proxy [5568]: Restarting dnsmasq on_config_update ?
Thu Feb  6 11:27:28 2025 user.notice https-dns-proxy [5568]: Setting trigger for wan ??
Thu Feb  6 11:27:35 2025 user.info adblock-4.2.4-r1[3220]: blocklist with overall 0 blocked domains loaded successfully (Netgear WNDR4300, ath79/nand, OpenWrt 24.10.0 r28427-6df0e3d02a )`

stangri, 2025-02-07:
> Unfortunately, but still after a cold boot, the service does not start. Log below. Service launched manually (Thu Feb 6 11:27:23 2025). Service do not autostart after upgrade to **24.10**, on 23.xx (and older) it has been flawlessly over the years.

The log you provided says otherwise:

```
Thu Feb  6 11:20:32 2025 user.notice https-dns-proxy [2037]: Stopping https-dns-proxy 2023.12.26-r2 on_failed_health_check ✓
Thu Feb  6 11:20:32 2025 user.notice https-dns-proxy [2037]: Setting trigger (on_boot) ✓
Thu Feb  6 11:27:23 2025 user.notice https-dns-proxy [5568]: Starting https-dns-proxy 2023.12.26-r2 instances ✓
Thu Feb  6 11:27:23 2025 user.notice https-dns-proxy [5568]: Updating dnsmasq config ✓
Thu Feb  6 11:27:28 2025 user.notice https-dns-proxy [5568]: Restarting dnsmasq on_config_update ✓
Thu Feb  6 11:27:28 2025 user.notice https-dns-proxy [5568]: Setting trigger for wan ✓✓
```

stangri, 2025-02-07:
> Quote: service launched **manually** (Thu Feb 6 11:27:23 2025).

Oh, right, should have noticed the time gap.

On the next reboot, before you manually start service, can you capture the output of: `service https-dns-proxy info`?

tmcqueen-materials, 2025-02-08:
@stangri : I am also seeing this issue. On boot I see two lines related to https-dns-proxy:
```
Fri Feb  7 22:12:22 2025 user.notice https-dns-proxy [4995]: Stopping https-dns-proxy 2023.12.26-r2 on_failed_health_check ✓
Fri Feb  7 22:12:22 2025 user.notice https-dns-proxy [4995]: Setting trigger (on_boot) ✓
```
And it is not running. As soon as I issue `service https-dns-proxy restart`, it properly starts and everything works fine.

It does look like there is some problem with the start script, since I see these errors when doing so:
```
Starting https-dns-proxy 2023.12.26-r2 instances /sbin/uci: Invalid argument
/sbin/uci: Invalid argument
/sbin/uci: Invalid argument
/sbin/uci: Invalid argument
/sbin/uci: Invalid argument
/sbin/uci: Invalid argument
/sbin/uci: Invalid argument
/sbin/uci: Invalid argument
/sbin/uci: Invalid argument
/sbin/uci: Invalid argument
✓/sbin/uci: Invalid argument
/sbin/uci: Invalid argument
/sbin/uci: Invalid argument
/sbin/uci: Invalid argument
✓
Setting trigger for wan ✓✓
```

While my uci configuration for it is pretty vanilla:
```
https-dns-proxy.config=main
https-dns-proxy.config.canary_domains_icloud='1'
https-dns-proxy.config.canary_domains_mozilla='1'
https-dns-proxy.config.dnsmasq_config_update='-'
https-dns-proxy.config.force_dns='1'
https-dns-proxy.config.force_dns_port='53' '853'
https-dns-proxy.config.procd_trigger_wan6='0'
https-dns-proxy.@https-dns-proxy[0]=https-dns-proxy
https-dns-proxy.@https-dns-proxy[0].bootstrap_dns='1.1.1.1,1.0.0.1'
https-dns-proxy.@https-dns-proxy[0].resolver_url='https://mozilla.cloudflare-dns.com/dns-query'
https-dns-proxy.@https-dns-proxy[0].listen_addr='127.0.0.1'
https-dns-proxy.@https-dns-proxy[0].listen_port='5053'
https-dns-proxy.@https-dns-proxy[0].user='nobody'
https-dns-proxy.@https-dns-proxy[0].group='nogroup'
https-dns-proxy.@https-dns-proxy[1]=https-dns-proxy
https-dns-proxy.@https-dns-proxy[1].bootstrap_dns='9.9.9.9'
https-dns-proxy.@https-dns-proxy[1].resolver_url='https://private.canadianshield.cira.ca/dns-query'
https-dns-proxy.@https-dns-proxy[1].listen_addr='127.0.0.1'
https-dns-proxy.@https-dns-proxy[1].listen_port='5054'
https-dns-proxy.@https-dns-proxy[1].user='nobody'
https-dns-proxy.@https-dns-proxy[1].group='nogroup'

```

As with others, adding sleep 10 && service https-dns-proxy restart to rc.local works around the issue.

tmcqueen-materials, 2025-02-08:
So I think I know what is going on. The problems arise if the system resolver (`dnsmasq/unbound/whatever`) is manually configured to only use the `https-dns-proxy`s as upstreams. The reason is that then this line:

https://github.com/openwrt/packages/blob/master/net/https-dns-proxy/files/etc/init.d/https-dns-proxy#L138

is always false because 

https://github.com/openwrt/packages/blob/master/net/https-dns-proxy/files/etc/init.d/https-dns-proxy#L40

will never work because https-dns-proxy is not yet up and any dns resolution relies on https-dns-proxy being up. 

I suspect the reason this sometimes works and sometimes doesnt across reboots is a race between interfaces coming up and the initial boot call for `https-dns-proxy` completing -- if the initial boot call for https-dns-proxy completes before all interfaces are up, then https-dns-proxy is restarted by the trigger once another interface is up, and that brings up https-dns-proxy. However, if all interfaces are up before the boot call completes, then https-dns-proxy never gets automatically restarted, and therefore gets left in the stopped state.

So the fix seems pretty straightforward: remove the resolveip check in the boot block, because it is not needed to bring up https-dns-proxy (as https-dns-proxy does its own bootstrap queries to bring things up...)

p0k33m0n, 2025-02-08:
> service https-dns-proxy info

```
{
        "https-dns-proxy": {
                "triggers": [
                        [
                                "interface.*.up",
                                [
                                        [
                                                "run_script",
                                                "/etc/init.d/https-dns-proxy",
                                                "restart",
                                                "on_interface_up"
                                        ]
                                ],
                                5000
                        ]
                ]
        }
}
```

stangri, 2025-02-09:
> It does look like there is some problem with the start script, since I see these errors when doing so:
>
> ```
> Starting https-dns-proxy 2023.12.26-r2 instances /sbin/uci: Invalid argument
> /sbin/uci: [...]

What's your `/etc/config/dhcp` file and what does the `for i in /etc/config/*; do if ! uci show ${i##*/} > /dev/null 2>&1; then echo -e -n "$i: "; uci show ${i##*/} > /dev/null; fi; done;` output?

> if the initial boot call for https-dns-proxy completes before all interfaces are up, then https-dns-proxy is restarted by the trigger once another interface is up, and that brings up [...]

This commit can be used to test this theory: https://github.com/stangri/https-dns-proxy/commit/853fe4903da87cf58e11270614f2192e1be958e7 or you can grab the updated init file here: https://raw.githubusercontent.com/stangri/https-dns-proxy/853fe4903da87cf58e11270614f2192e1be958e7/files/etc/init.d/https-dns-proxy

@p0k33m0n @tmcqueen-materials please test if the boot up start works with the init script linked above so I could merge the changes to the OpenWrt repos.

b1ackbeat, 2025-02-09:
I have the same problem with https-dns-proxy crash during router startup on OpenWrt 24.10, and I found the updated init file above, its seems the updated init file solves the problem.

tmcqueen-materials, 2025-02-09:
@stangri : 

Thanks! I won't know for sure until trying things many more times, but the updated init script *appears* to solve the non-start-at-boot problem.

The command `for i in /etc/config/*; do if ! uci show ${i##*/} > /dev/null 2>&1; then echo -e -n "$i: "; uci show ${i##*/} > /dev/null; fi; done;` outputs precisely nothing. 

Here is a redacted version of `/etc/config/dhcp`:
```
config dnsmasq
	option domainneeded '0'
	option boguspriv '1'
	option filterwin2k '0'
	option localise_queries '1'
	option rebind_protection '1'
	option rebind_localhost '1'
	option local '/lan/'
	option domain '(redacted-isp).net'
	option expandhosts '1'
	option nonegcache '1'
	option cachesize '2000'
	option authoritative '1'
	option readethers '1'
	option leasefile '/tmp/dhcp.leases'
	option resolvfile '/tmp/resolv.conf.d/resolv.conf.auto'
	option nonwildcard '1'
	option localservice '1'
	option ednspacket_max '65535'
	option filter_aaaa '0'
	option filter_a '0'
	list addnhosts '/tmp/hosts/pihole-hosts'
	list addnhosts '/tmp/hosts/pihole-hosts-ipv6'
	list addnhosts '/etc/extra-block-hosts'
	option noresolv '1'
	option port '5153'
	list rev_server '172.217.0.0/16,127.0.0.1'
	option ubus '0'
	option dnsforwardmax '1024'
	option confdir '/etc/dnsmasq.custom'
	option queryport '0'
	list interface 'lo'
	list interface 'br-guest'
	list interface 'br-lan'
	option allservers '0'
	list server '127.0.0.1#5053'
	list server '127.0.0.1#5054'
	list server '/mask.icloud.com/'
	list server '/mask-h2.icloud.com/'
	list server '/use-application-dns.net/'

config dhcp 'lan'
	option interface 'lan'
	option start '2'
	option limit '252'
	option leasetime '60m'
	option dhcpv4 'server'
	list dhcp_option '3,192.168.2.1'
	list dhcp_option '6,192.168.2.1'
	list dhcp_option 'option6:dns-server,[::]'
	option dhcpv6 'disabled'
	option dns_service '0'
	option ra 'disabled'

config dhcp 'wan'
	option interface 'wan'
	option ignore '1'

config host
	option mac '11:22:33:44:55:66'
	option name 'device-name-one'
	option ip '192.168.2.3'
	option hostid '3'
	option dns '1'

config host
	option mac '77:88:99:aa:bb:cc'
	option name 'device-name-two'
	option ip '192.168.2.4'
	option hostid '4'
	option dns '1'

...(a bunch more just like these with different macs, names, ips)...

config dhcp 'guest'
	option interface 'guest'
	option start '2'
	option limit '244'
	option leasetime '60m'
	option dhcpv4 'server'
	list dhcp_option '3,192.168.0.1'
	list dhcp_option '6,192.168.0.1'
	list dhcp_option 'option6:dns-server,[::]'
	option dhcpv6 'disabled'
	option dns_service '0'
	option ra 'disabled'
```

stangri, 2025-02-09:
@tmcqueen-materials @b1ackbeat thank you for testing the new init script so quickly, I'll merge changes to snapshots and 24.10 asap.

> The command `for i in /etc/config/*; do if ! uci show ${i##*/} > /dev/null 2>&1; then echo -e -n "$i: "; uci show ${i##*/} > /dev/null; fi; done;` outputs precisely nothing.

Uhm, that does seem to indicate that your configs are healthy and it is indeed an issue with the https-dns-proxy init script. The only direct call to /sbin/uci is from the uci_changes() function. Do you use a custom path for uci configs? Does the `uci changes dhcp` produce any errors when invoked directly?

tmcqueen-materials, 2025-02-09:
No problem -- `uci changes dhcp` is clean (outputs nothing). I am not using a custom path for uci configs or anything either.

stangri, 2025-02-10:
@tmcqueen-materials not sure how to approach the invalid arguments when calling uci, I can't reproduce that. If you're willing to do some more legwork and add debugging output to `uci_changes()` function it may help.

The PRs for fixed boot up start are: https://github.com/openwrt/packages/pull/25941 https://github.com/openwrt/packages/pull/25942

erayrafet, 2025-02-15:
https://github.com/openwrt/packages/issues/25949#issuecomment-2660754468

Nrth9, 2025-02-16:
Too often serious problems arise with this package. There are doubts about the competence of maintainers.

Shine-, 2025-02-17:
> This is the correct way to have `dnsmasq_config_update` set (so that it doesn't apply the default of change all instances) yet have it be a list of zero items and thus have no instances changed.

Not correct, since if you look at the code, if the value is empty, it will then default to "*" and try to mess with your configuration. You'll have to set it to an empty string that is **not** zero-length, e.g. ' ' (one blank).

```
uci set https-dns-proxy.config.dnsmasq_config_update=' '
uci commit https-dns-proxy
```

Shine-, 2025-02-17:
The value gets loaded using `config_get dnsmasq_config_update 'config' 'dnsmasq_config_update' '*'` [here](https://github.com/openwrt/packages/blob/0f9bcfa8d8f3117a2a4932e384791c0df343d790/net/https-dns-proxy/files/etc/init.d/https-dns-proxy#L244C2-L244C73).

Replicating this with your method:

```
. /lib/functions.sh
uci delete https-dns-proxy.config.dnsmasq_config_update
uci add_list https-dns-proxy.config.dnsmasq_config_update=''
uci commit https-dns-proxy
config_load "https-dns-proxy"
config_get dnsmasq_config_update 'config' 'dnsmasq_config_update' '*'
echo Value is \"${dnsmasq_config_update}\"
```
Output:
`Value is "*"`

My method:
```
. /lib/functions.sh
uci delete https-dns-proxy.config.dnsmasq_config_update
uci set https-dns-proxy.config.dnsmasq_config_update=' '
uci commit https-dns-proxy
config_load "https-dns-proxy"
config_get dnsmasq_config_update 'config' 'dnsmasq_config_update' '*'
echo Value is \"${dnsmasq_config_update}\"
```
Output:
`Value is " "`

Q.E.D.

"*" likely works for you because you don't have a dnsmasq instance it can loop through in the first place.

tmcqueen-materials, 2025-02-17:
@Shine- : Hmm, that's interesting, in part because I did exactly the same tests and got a different result. Maybe I had a list with a single space in the name ... hmm, yep, that's exactly it. You are correct.

@stangri : is there some specific way to turn off the modification of dnsmasq by this script? Other than providing a fake value that corresponds to no

aarond10, 2025-02-17:
> Too often serious problems arise with this package. There are doubts about the competence of maintainers.

@Nrth9, I wrote the original version of this and there have been a small but highly appreciated set of contributors to the code since then. @stangri does a great job of maintaining the binary and init scripts for OpenWRT. None of us get paid for this and we all have our own lives to attend to. Please stick to constructive criticism. You're being highly disrespectful.

stangri, 2025-02-17:
> [@stangri](https://github.com/stangri) : I had a chance to look at those uci invalid argument errors. They _do not_ come from `uci_changes()`.

Thanks, since no separate issue was opened for the `/sbin/uci: Invalid argument` output I can't reference it in the commit, but it was fixed in https://github.com/stangri/https-dns-proxy/commit/a656c5e48468f99eb125e6aa172b700e05fd9de9.

> [@stangri](https://github.com/stangri) : is there some specific way to turn off the modification of dnsmasq by this script? Other than providing a fake value that corresponds to no actual instance?

`-` as per README/luci app.

It's getting impossible to track the the original boot up start issue in here.

tmcqueen-materials, 2025-02-17:
> Thanks, since no separate issue was opened for the `/sbin/uci: Invalid argument` output I can't reference it in the commit, but it was fixed in [stangri/https-dns-proxy@a656c5e](https://github.com/stangri/https-dns-proxy/commit/a656c5e48468f99eb125e6aa172b700e05fd9de9).

Great, thank you! I will test it out and let you know.

> 
> > [@stangri](https://github.com/stangri) : is there some specific way to turn off the modification of dnsmasq by this script? Other than providing a fake value that corresponds to no actual instance?
> 
> `-` as per README/luci app.

Right, except that was causing the `invalid argument` messages. I'll check out the commit above to make sure it fixes it when set this way.

> It's getting impossible to track the the original boot up start issue in here.

Would you like a separate issue opened?

stangri, 2025-02-18:
> Would you like a separate issue opened?

no, I'm confident it's fixed thanks to your investigation. In retrospect I should have asked for it earlier. ;)

tmcqueen-materials, 2025-02-18:
> > Would you like a separate issue opened?
> 
> no, I'm confident it's fixed thanks to your investigation. In retrospect I should have asked for it earlier. ;)

OK. I can confirm that https://github.com/stangri/https-dns-proxy/commit/a656c5e48468f99eb125e6aa172b700e05fd9de9 fixes the invalid arguments errors when `https-dns-proxy.config.dnsmasq_config_update='-'`. Thank you!

Nrth9, 2025-02-18:
@aarond10 https-dns-proxy - It's not a photo editing program. This package is responsible for access to the Internet, especially it should be noted that it protects privacy. Constantly experiencing the same serious failures indicates that the code is poorly maintained.

ob45, 2025-02-18:
> OK. I can confirm that [stangri/https-dns-proxy@a656c5e](https://github.com/stangri/https-dns-proxy/commit/a656c5e48468f99eb125e6aa172b700e05fd9de9) fixes the invalid arguments errors when `https-dns-proxy.config.dnsmasq_config_update='-'`. Thank you!

Did this one get merged also?
If yes I could reflash my router with the latest snapshot to check if https-dns-proxy works again

stangri, 2025-02-18:
@ob45 @tmcqueen-materials: https://github.com/openwrt/packages/pull/26001 I just merged the changes to snapshots, if you're looking for a binary it may take some time, if you're building from source it should be there.

tmcqueen-materials, 2025-02-18:
@stangri : Great! It is probably important to make sure it is applied to the 24.10 branches (if not already done?) so that whenever 24.10.1 is minted these updates are included. Thank you!!

stangri, 2025-02-19:
> [@stangri](https://github.com/stangri) : Great! It is probably important to make sure it is applied to the 24.10 branches (if not already done?) so that whenever 24.10.1 is minted these updates are included. Thank you!!

This time decided to wait on reports from snapshots before cherry-picking the fix for release.

ob45, 2025-02-21:
> This time decided to wait on reports from snapshots before cherry-picking the fix for release.

Unfortunately I can't test it because the snapshot for my device seems to be extremely unstable. 

The very first attempt I tried to install the module my router gave me some odd error code in the yellow box and then restarted (without any installed modules)

The real "first" attempt and second attempt I got these segfault errors after which the router stopped providing me any internet or router access:

[image]

Either there is something wrong with the package manager or maybe even the kernel. I noticed the snapshot has a higher Kernel version than the official 24.10 release.

stangri, 2025-02-22:
> The real "first" attempt and second attempt I got these segfault errors after which the router stopped providing me any internet or router access:

That's from OpenWrt repo or my repo? I may need to update the code for the builder used for creating APKs for my repo.

stangri, 2025-02-23:
> > The real "first" attempt and second attempt I got these segfault errors after which the router stopped providing me any internet or router access:
> 
> That's from OpenWrt repo or my repo? I may need to update the code for the builder used for creating APKs for my repo.

Either way I've rebuilt the binaries in my APK repo with the most recent source checkout. For testing you can also just grab the init-script from PR and replace the existing `/etc/init.d/https-dns-proxy` with it. Apart from the version being `dev-test` it should work.

ob45, 2025-02-24:
> That's from OpenWrt repo or my repo? I may need to update the code for the builder used for creating APKs for my repo.

That was from the OpenWrt Firmware Selector

minax007, 2025-04-04:
I have tested the master-branch and the 24.10-branch versions on different devices and the issue still persists. 

In 24.10.rc7 the issue was not there. 

As a **workaround** I have added a cron job under LuCi "Scheduled Tasks":

`@reboot sleep 180 && /etc/init.d/https-dns-proxy restart`

By this the https-dns-proxy service restarts three minutes after the router has started ... and it works correctly.

stangri, 2025-04-05:
> I have tested the master-branch and the 24.10-branch versions on different devices and the issue still persists.

What's in the logs from https-dns-proxy on master branch and 24.10 branch on different devices?

minax007, 2025-04-05:
> What's in the logs from https-dns-proxy on master branch and 24.10 branch on different devices?

Sat Apr  5 09:38:52 2025 user.notice https-dns-proxy [2038]: Stopping https-dns-proxy 2023.12.26-r5 on_failed_health_check ✓
Sat Apr  5 09:38:52 2025 user.notice https-dns-proxy [2038]: Setting trigger (on_boot) ✓

minax007, 2025-04-05:
I see that the workaround that I have mentioned yesterday by setting the cron job ...

`@reboot sleep 180 && /etc/init.d/https-dns-proxy restart`

... is also no solution when the router was stopped for a longer time. 
The only way to get https-dns-proxy working again is to give it a manual restart.

[19 more replies]

closed 2025-11-18 by ob45