openwrt/packages #25800: uci : breaking a package(the firewall) luci ui.

mic-user, 2025-01-22:
Environment: mvebu / cortexa9 - Marvell Armada 385 88F6820 / Linksys WRT32X / 24.10.0-rc5

Description:
After doing uci configuration for firewall to turn on/off as below,
web ui broke.

Repro steps
> uci set firewall.@rule[$_index_number].enabled='0'
> uci commit firewall
>  service firewall reload
then

[image]

brada4, 2025-01-25:
Neither is from this repo.

Any output via `fw4 check`
... fw3?

Content in config/firewall?

Can you format so called repeater as code so that others can copy it too?

mic-user, 2025-01-25:
> Neither is from this repo.
> 
> Any output via `fw4 check` ... fw3?
> 
> Content in config/firewall?
> 
> Can you format so called repeater as code so that others can copy it too?

That is... actually hard to reproduce. At this moment no idea how to repeat.
(I saw this trouble only 2~3 times since openwrt 19.x past years)
But here is information that I can give you at best.

1. Using 'firewall 4'.
2. At this moment, `fw4 check` saying `Section @include[0] option 'reload' is not supported by fw4`
3. One of kmod is `ntfs3` which first time use within 24.10.x
4. Right after the symptom, the feature of firewall seems working with no problem.
5. There is lots of packages combination are working(samba fail2ban transmission etc, not heavy load, but concurrently).
6. And sysupgraded after the trouble to make normal.
7. For now, not able to reproduce. Do not have exact clue for the issue.

This is it. If you guys want to close, then do close. but mind it please for the future stablity.
(The reason why keep delivery a problem, even though it cannot be reproducible, is
you guys are guru of the openwrt who knows openwrt internal by sencing symptom
based on deep understanding so may have chance to fix. If I had a clue or exact log, I would deliver.)