openwrt/packages #25801: strongswan: dhcp plugin not issuing addresses to clients

lowjoel, 2025-01-22:
After upgrading to OpenWrt 24.10-rc4, using the exact same swanctl configuration as 23.05 my clients are no longer getting a DHCP address.

I can tell that the updown scripts in /etc/hotplug.d/ipsec are only being called with IPv6 events, i.e. `PLUTO_VERB='up-client-v6'` and `PLUTO_VERB='down-client-v6'`, I'm not sure if there's some configuration I'd need to change or if there's something more nefarious going on here.

tobiasbrunner, 2025-01-23:
No idea what the DHCP issue is about (you probably have to provide more details, config, logs, packet captures). But that the updown script is only called for IPv6 makes sense if no IPv4 address is assigned to the client.

lowjoel, 2025-01-23:
Yeah, I'm trying to understand what's going wrong, I have some digging in to do myself. At this point I can't figure out if I'm the only one impacted (=I did something stupid) or if others are seeing it. I'm currently working to distill my setup to a minimal reproduceable case (I couldn't yet yesterday)

lowjoel, 2025-01-24:
I notice that 23.05 used Strongswan 5.9.10, but 24.10 uses 5.9.14; 5.9.14 added an eBPF-based DHCP packet filter; let me see if compiling 5.9.13 works around this.

tobiasbrunner, 2025-01-24:
The packet filtering as such hasn't changed on Linux, it was just refactored to a different file so it can be shared with the `farp` plugin and to support FreeBSD. You haven't shared your config, logs, or packet captures, so really no idea what's going on. But if you are using a unicast server address, it could be related to the change  I pushed as response to **your** bug report [here](https://github.com/strongswan/strongswan/issues/1573), which was released with 5.9.11.

tobiasbrunner, 2025-01-24:
Oh, sorry, haven't checked there. That log is from after you changed stuff in the config you posted, right? So there does seem to be an offer, but it does not hit the socket. Maybe some firewall filtering? Or maybe it hits a different interface (i.e. not the one bound to)?

tobiasbrunner, 2025-01-24:
> unless there's something special about vlans on a bridge.

Could be, was that used on 23.05 as well? Not sure how well e.g. `SO_BINDTODEVICE` works with such interfaces (or with packet sockets as there is actually a note in the man page for `socket(7)` stating: "It is not supported for packet sockets (use normal bind(2) there)", not sure what that means exactly as the call does not seem to fail), or whether there is something regarding packet sockets and VLANs that could be an issue.

lowjoel, 2025-01-24:
> Could be, was that used on 23.05 as well

Yes, no change on that front. The vlans on that bridge are for DSA. The firewall and network config between the two releases as far as I can tell are the same (and I'm testing this on two different devices of the same model, but unfortunately I upgraded them both before I realised that this was a problem.)

tobiasbrunner, 2025-01-24:
Do you have a pcap file that contains the DHCPOFFER? Maybe our packet filter drops it for some reason.

lowjoel, 2025-01-24:
Ah. I see a difference in the pcap between the two configs I used. The logs from dnsmasq/strongswan are the same, but the pcap is different.

In both captures, I use `tcpdump -n -i br-lan.xxxx -w /tmp/dhcp.pcap --print '(udp port 66)' or '(udp port 67)'` which should capture all DHCP traffic.

Initial config: `dhcp.interface` unset, `dhcp.server = br-lan.xxxx broadcast` (i.e. dhcp.server = 192.168.1.255)
```
21:59:24.697491 IP router-addr.68 > subnet-broadcast-addr.67: BOOTP/DHCP, Request from mac1, length 248
21:59:25.698394 IP router-addr.68 > subnet-broadcast-addr.67: BOOTP/DHCP, Request from mac1, length 248
21:59:27.698950 IP router-addr.68 > subnet-broadcast-addr.67: BOOTP/DHCP, Request from mac1, length 248
21:59:30.699510 IP router-addr.68 > subnet-broadcast-addr.67: BOOTP/DHCP, Request from mac1, length 248
21:59:34.700028 IP router-addr.68 > subnet-broadcast-addr.67: BOOTP/DHCP, Request from mac1, length 248
```

Modified config: `dhcp.interface = br-lan.xxxx`, `dhcp.server` unset
```
22:10:20.880723 IP router-addr.68 > 255.255.255.255.67: BOOTP/DHCP, Request from router-addr, length 248
22:10:21.889733 IP router-addr.68 > 255.255.255.255.67: BOOTP/DHCP, Request from router-addr, length 248
22:10:23.890633 IP router-addr.68 > 255.255.255.255.67: BOOTP/DHCP, Request from router-addr, length 248
22:10:24.030407 IP router-addr.67 > 255.255.255.255.68: BOOTP/DHCP, Reply, length 334
22:10:24.032360 IP router-addr.67 > 255.255.255.255.68: BOOTP/DHCP, Reply, length 334
22:10:24.033676 IP router-addr.67 > 255.255.255.255.68: BOOTP/DHCP, Reply, length 334
22:10:26.891285 IP router-addr.68 > 255.255.255.255.67: BOOTP/DHCP, Request from router-addr, length 248
22:10:26.892651 IP router-addr.67 > 255.255.255.255.68: BOOTP/DHCP, Reply, length 334
22:10:30.891928 IP router-addr.68 > 255.255.255.255.67: BOOTP/DHCP, Request from router-addr, length 248
22:10:30.893305 IP router-addr.67 > 255.255.255.255.68: BOOTP/DHCP, Reply, length 334
```

tobiasbrunner, 2025-01-24:
> Modified config: `dhcp.interface = br-lan.xxxx`, `dhcp.server` unset

Can you bind to `br-lan`? If so, does that make a difference?

lowjoel, 2025-01-24:
Modified config: `dhcp.interface = br-lan`, `dhcp.server` unset:

```
Fri Jan 24 22:32:15 2025 authpriv.info ipsec: 05[CFG] sending DHCP DISCOVER for mac1 to 255.255.255.255
Fri Jan 24 22:32:16 2025 authpriv.info ipsec: 05[CFG] sending DHCP DISCOVER for mac1 to 255.255.255.255
Fri Jan 24 22:32:18 2025 authpriv.info ipsec: 11[MGR] ignoring request with ID 12, already processing
Fri Jan 24 22:32:18 2025 authpriv.info ipsec: 14[NET] received packet: from remote[4500] to local[4500] (57 bytes)
Fri Jan 24 22:32:18 2025 authpriv.info ipsec: 14[ENC] parsed INFORMATIONAL request 1 [ ]
Fri Jan 24 22:32:18 2025 authpriv.info ipsec: 14[ENC] generating INFORMATIONAL response 1 [ ]
Fri Jan 24 22:32:18 2025 authpriv.info ipsec: 14[NET] sending packet: from local[4500] to remote[4500] (57 bytes)
Fri Jan 24 22:32:18 2025 authpriv.info ipsec: 05[CFG] sending DHCP DISCOVER for mac1 to 255.255.255.255
Fri Jan 24 22:32:20 2025 authpriv.info ipsec: 08[MGR] ignoring request with ID 12, already processing
Fri Jan 24 22:32:21 2025 authpriv.info ipsec: 05[CFG] sending DHCP DISCOVER for mac1 to 255.255.255.255
Fri Jan 24 22:32:25 2025 authpriv.info ipsec: 02[MGR] ignoring request with ID 12, already processing
Fri Jan 24 22:32:25 2025 authpriv.info ipsec: 05[CFG] sending DHCP DISCOVER for mac1 to 255.255.255.255
Fri Jan 24 22:32:30 2025 authpriv.info ipsec: 05[CFG] DHCP DISCOVER timed out
```

tcpdump shows nothing if I listen on the vlan that is for my LAN interface, instead the source IP is my public WAN address (!!)

```
22:35:22.261182 IP local-wan-ip.68 > 255.255.255.255.67: BOOTP/DHCP, Request from mac1, length 248
22:35:23.262209 IP local-wan-ip.68 > 255.255.255.255.67: BOOTP/DHCP, Request from mac1, length 248
22:35:25.262732 IP local-wan-ip.68 > 255.255.255.255.67: BOOTP/DHCP, Request from mac1, length 248
22:35:28.263218 IP local-wan-ip.68 > 255.255.255.255.67: BOOTP/DHCP, Request from mac1, length 248
22:35:32.263685 IP local-wan-ip.68 > 255.255.255.255.67: BOOTP/DHCP, Request from mac1, length 248
```

(times don't match because I captured the logs while tcpdump-ing br-lan.xxxx, ran a second time to capture br-lan directly.)

tobiasbrunner, 2025-01-24:
Interesting choice of source address.  Why is the WAN interface even attached to the bridge interface?

Anyway, what you could try, if you are able to build a new version of strongSwan, is maybe only bind the send socket and let the packet socket listen on any interface. To do so, replace `iface` with `NULL` [here](https://github.com/strongswan/strongswan/blob/d38eaa6dd71d65a894e3914d46eeb6d221937a7c/src/libcharon/plugins/dhcp/dhcp_socket.c#L876-L877).

Alternatively, the following patch uses a different approach to bind the socket to an interface:

```diff
diff --git a/src/libcharon/network/pf_handler.c b/src/libcharon/network/pf_handler.c
index 43ef432ba607..60fc7ff6b426 100644
--- a/src/libcharon/network/pf_handler.c
+++ b/src/libcharon/network/pf_handler.c
@@ -225,6 +225,30 @@ METHOD(pf_handler_t, destroy, void,
 	free(this);
 }
 
+/**
+ * Bind the given packet socket to the a named device
+ */
+static bool bind_packet_socket_to_device(int fd, char *iface)
+{
+	struct sockaddr_ll addr = {
+		.sll_family = AF_PACKET,
+		.sll_ifindex = if_nametoindex(iface),
+	};
+
+	if (!addr.sll_ifindex)
+	{
+		DBG1(DBG_CFG, "unable to bind socket to '%s': not found", iface);
+		return FALSE;
+	}
+	if (bind(fd, (struct sockaddr*)&addr, sizeof(addr)) == -1)
+	{
+		DBG1(DBG_CFG, "binding socket to '%s' failed: %s",
+			 iface, strerror(errno));
+		return FALSE;
+	}
+	return TRUE;
+}
+
 /**
  * Setup capturing via AF_PACKET socket
  */
@@ -247,7 +271,7 @@ static bool setup_internal(private_pf_handler_t *this, char *iface,
 			 this->name, strerror(errno));
 		return FALSE;
 	}
-	if (iface && !bind_to_device(this->receive, iface))
+	if (iface && !bind_packet_socket_to_device(this->receive, iface))
 	{
 		return FALSE;
 	}
```

lowjoel, 2025-01-26:
I've diagnosed (hur hur, that's an overstatement, BPF was a fun distraction and I was fumbling around a lot) this to the point that:

Using `dhcp.interface` unset, `dhcp.server = subnet-broadcast-addr`, none of your binding patches applied:

1. the packets are being received after the BPF packet filter (which matches the observation that tcpdump shows the packets.) I do have a patch that you may consider applying to handle the case where the IPv4 header has a non-default IHL -- I can open a PR for you if you'd like.
2. the pf_handler `receive_packet` is being called (which means that the polling/async jobs are working as expected).
3. but when `receive_packet` is called, `recvfrom` succeeds. However at the next branch calling `find_interface`, it returns NULL.
4. The socket that's being read from is `lo`. dnsmasq is on the localhost of the Strongswan server.

Using `dhcp.interface = br-lan.xxx`, `dhcp.server` unset, none of your binding patches applied:

1. can't find the packets being received, though it appears in tcpdump.

tobiasbrunner, 2025-01-27:
Thanks for investigating this further.

> 1. I do have a patch that you may consider applying to handle the case where the IPv4 header has a non-default IHL

Not sure if that's really necessary because I doubt many DHCP servers are using IP options (they are rare for IPv4 to begin with).

> 3\. but when `receive_packet` is called, `recvfrom` succeeds. However at the next branch calling `find_interface`, it returns NULL.

Good catch! This part of the code is actually "new", i.e. was not part of the `dhcp` plugin before the `pf_handler` refactoring. It's from and for the `farp` plugin, which needs to know the MAC address of the interface on which a packet was received to send back a fake ARP response. The `dhcp` plugin does not care for the address or the interface name and index.

I've pushed a slightly different fix to the `dhcp-receive` branch.

> 4\. dnsmasq is on the localhost of the Strongswan server.

Is that the default or a special config? What happens if you set `dhcp.interface = lo`?

> Using `dhcp.interface = br-lan.xxx`, `dhcp.server` unset, none of your binding patches applied:

I feel this should work (with the loopback patch) if you'd remove the interface from the `pf_handler` constructor.

> Using `dhcp.interface = br-lan.xxx`, `dhcp.server` unset, bind using `bind`:

Makes sense as packets from `lo` will be ignored. I wonder if something like `dhcp.interface_receive` would be useful to specify a separate interface to bind the receive socket to.

lowjoel, 2025-01-27:
> I've pushed a slightly different fix to the dhcp-receive branch.

Let me check it out and compile it and give it a shot, thanks.

> Is that the default or a special config? What happens if you set dhcp.interface = lo?

I'm sorry. I think I was a bit imprecise with that comment. I meant that dnsmasq is on the same host as strongswan; in that sense, dnsmasq is localhost relative to charon.

To be precise, my dnsmasq.conf currently has these network settings set:
```
bind-interfaces
local-service
listen-address=router-address-v4 (e.g 192.168.1.1)
listen-address=router-address-v6 (e.g. 2001:...::1)
listen-address=::1
interface=br-lan.xxxx
no-dhcp-interface=wan
```

among others (which I think are uninteresting wrt this conversation.) Why is it that when I set the `dhcp.server` to the router address subnet's broadcast address dnsmasq replies on `lo`, I do not know. I have a setup in which I have different instances of dnsmasq running on different interfaces because each interface gets a different DNS suffix list.

> I feel this should work (with the loopback patch) if you'd remove the interface from the pf_handler constructor.

Let me try this on the weekend.

> I wonder if something like dhcp.interface_receive would be useful to specify a separate interface to bind the receive socket to.

It depends on how dnsmasq sends the reply or if this is a bug somewhere else in the stack.

DocMAX, 2025-01-28:
Same problem here. Getting a "DHCP DISCOVER timed out" although dnsmasq does offer an IP. I am on 24.10, too.

lowjoel, 2025-01-29:
> I've pushed a slightly different fix to the dhcp-receive branch.

It seems to work, thanks. I'm not sure what kind of MAC address is going to be copied, hence I did the more conservative `00:00:00:00:00:00`.

> At least it would allow binding both sockets (or only one)

Correct; but it feels messy to me. I'm not sure if this is a one-off thing or if Linux has been doing this since forever. I'm also not sure how this would play with scenarios in which we may use e.g. RADIUS to decide which DHCP server to get addresses from. For example, I may want to assign different groups of users different IP address ranges and they have to get their addresses from different DHCP servers (not sure if that's possible; I'm pulling examples out of thin air.) In this specific case, allowing addresses from either ethernet/loopback seems to work here. @DocMAX do you want to try the patch for yourself too?

tobiasbrunner, 2025-01-29:
> > I've pushed a slightly different fix to the dhcp-receive branch.
> 
> It seems to work, thanks.

Thanks for testing. I'll wait for some additional feedback from @DocMAX (see [here](https://github.com/strongswan/strongswan/commit/a7341cf23bc2fd01af7eff729894ff2a7014263b)) and then will line it up for the next release.

> I'm not sure what kind of MAC address is going to be copied, hence I did the more conservative `00:00:00:00:00:00`.

That's already the default address if nothing else is configured (see e.g. `ip link`). But  you can freely set a different address on `lo` (`ip link set dev lo address ...`, no idea what purpose or side-effects that could have), so passing the actual address along can't hurt.

> I'm not sure if this is a one-off thing or if Linux has been doing this since forever.

I'm pretty sure it has been like that forever. This is, of course, only a problem when running the DHCP server on the same host as the DHCP client, which is kind of a special setup. For instance, the `force_server_address` was also added because of that. Without it, the client would send the DHCP ACK message to the (local) unicast address from which the DHCP OFFER was sent, which would also go via `lo`, on which the DHCP server might not be listening. So by enabling that option and configuring the subnet's broadcast address, the DHCP ACK is again sent via the interface on which the server is listening  (see [the first footnote in the docs](https://docs.strongswan.org/docs/latest/plugins/dhcp.html#_footnoteref_1)).

I guess binding to an interface is more useful when the server is not running on the same host. But having the option to configure the interface for the receive socket separately might allow more generic configs (e.g. without having to know the subnet's local broadcast address) even when running on the same host.

I've pushed a small change that implements this to the branch. Let me know what you think.

> I'm also not sure how this would play with scenarios in which we may use e.g. RADIUS to decide which DHCP server to get addresses from. For example, I may want to assign different groups of users [...]

That's currently not possible. But you can let the RADIUS server [assign the virtual IP](https://docs.strongswan.org/docs/latest/plugins/eap-radius.html#_radius_attribute_forwarding) directly.

DocMAX, 2025-01-29:
```
dhcp {
    identity_lease = yes
    load = yes
    force_server_address = yes
    server = 192.168.1.255
}
```
Patch from https://github.com/strongswan/strongswan/commit/a7341cf23bc2fd01af7eff729894ff2a7014263b doesn't work with this config.

tobiasbrunner, 2025-01-29:
Do you actually have the patch applied correctly? And please post the log.

DocMAX, 2025-01-29:
```
Applying /home/docmax/openwrt/sources/snapshot/feeds/packages/net/strongswan/patches/0901-uci-verbatim-patch-from-openwrt-package-sources.patch using plaintext:
patching file src/libcharon/plugins/uci/uci_parser.c

Applying /home/docmax/openwrt/sources/snapshot/feeds/packages/net/strongswan/patches/0903-updown-Call-sbin-hotplug-call-ipsec-1-in-updown-scri.patch using plaintext:
patching file src/_updown/_updown.in

Applying /home/docmax/openwrt/sources/snapshot/feeds/packages/net/strongswan/patches/0904-gmpdh-Plugin-that-implements-gmp-DH-functions-in-an-.patch using plaintext:
patching file configure.ac
patching file src/libstrongswan/Makefile.am
patching file src/libstrongswan/plugins/gmpdh/Makefile.am
patching file src/libstrongswan/plugins/gmpdh/gmpdh_plugin.c
patching file src/libstrongswan/plugins/gmpdh/gmpdh_plugin.h

**Applying /home/docmax/openwrt/sources/snapshot/feeds/packages/net/strongswan/patches/0905-dhcp.patch using plaintext:
patching file src/libcharon/network/pf_handler.c**
touch /home/docmax/openwrt/sources/snapshot/build_dir/target-mips_24kc_musl/strongswan-5.9.14/.prepared_e9647582c88dfa58f8fddbb0c5df408e_6664517399ebbbc92a37c5bb081b5c53
rm -f /home/docmax/openwrt/sources/snapshot/build_dir/target-mips_24kc_musl/strongswan-5.9.14/.configured_*
rm -f /home/docmax/openwrt/sources/snapshot/staging_dir/target-mips_24kc_musl/stamp/.strongswan_installed
(cd /home/docmax/openwrt/sources/snapshot/build_dir/target-mips_24kc_musl/strongswan-5.9.14; rm -f aclocal.m4; if [ -f ./configure.ac ] || [ -f ./configure.in ]; then [ -d ./autom4te.cache ] && rm -rf ./a
utom4te.cache; [ -e ./config.rpath ] || ln -s /home/docmax/openwrt/sources/snapshot/scripts/config.rpath ./config.rpath; touch NEWS AUTHORS COPYING ABOUT-NLS ChangeLog; AUTOM4TE=/home/docmax/openwrt/sourc
es/snapshot/staging_dir/host/bin/autom4te AUTOCONF=/home/docmax/openwrt/sources/snapshot/staging_dir/host/bin/autoconf AUTOMAKE=/home/docmax/openwrt/sources/snapshot/staging_dir/host/bin/automake ACLOCAL=
/home/docmax/openwrt/sources/snapshot/staging_dir/host/bin/aclocal AUTOHEADER=/home/docmax/openwrt/sources/snapshot/staging_dir/host/bin/autoheader LIBTOOLIZE=/home/docmax/openwrt/sources/snapshot/staging
_dir/host/bin/libtoolize LIBTOOL=/home/docmax/openwrt/sources/snapshot/staging_dir/host/bin/libtool M4=/home/docmax/openwrt/sources/snapshot/staging_dir/host/bin/m4 AUTOPOINT=true GTKDOCIZE=true LIBTOOLIZ
E='/home/docmax/openwrt/sources/snapshot/staging_dir/host/bin/libtoolize --install' /home/docmax/openwrt/sources/snapshot/staging_dir/host/bin/autoreconf -v -f -i -B /home/docmax/openwrt/sources/snapshot/
staging_dir/host/share/aclocal -I /home/docmax/openwrt/sources/snapshot/staging_dir/target-mips_24kc_musl/host/share/aclocal -I /home/docmax/openwrt/sources/snapshot/staging_dir/hostpkg/share/aclocal -I /
home/docmax/openwrt/sources/snapshot/staging_dir/target-mips_24kc_musl/usr/share/aclocal -I m4 -I . . || true; fi; );
OpenWrt-libtoolize: putting auxiliary files in '.'.
OpenWrt-libtoolize: copying file './config.guess'
OpenWrt-libtoolize: copying file './config.sub'
OpenWrt-libtoolize: copying file './install-sh'
```
This is the building process for OpenWrt. "patching file src/libcharon/network/pf_handler.c" returns no error so i think its applied correctly.

And the ipsec log is:
```
Jan 29 19:34:02 router ipsec: 09[NET] received packet: from 109.42.112.171[11000] to 95.119.147.36[500] (948 bytes)
Jan 29 19:34:02 router ipsec: 09[ENC] parsed IKE_SA_INIT request 0 [ SA KE No N(NATD_S_IP) N(NATD_D_IP) N(FRAG_SUP) N(HASH_ALG) N(REDIR_SUP) ]
Jan 29 19:34:02 router ipsec: 09[IKE] 109.42.112.171 is initiating an IKE_SA
Jan 29 19:34:02 router ipsec: 09[CFG] selected proposal: IKE:AES_CBC_256/HMAC_SHA2_384_192/PRF_HMAC_SHA2_384/MODP_3072
Jan 29 19:34:02 router ipsec: 09[IKE] remote host is behind NAT
Jan 29 19:34:02 router ipsec: 09[IKE] DH group ECP_256 unacceptable, requesting MODP_3072
Jan 29 19:34:02 router ipsec: 09[ENC] generating IKE_SA_INIT response 0 [ N(INVAL_KE) ]
Jan 29 19:34:02 router ipsec: 09[NET] sending packet: from 95.119.147.36[500] to 109.42.112.171[11000] (38 bytes)
Jan 29 19:34:02 router ipsec: 01[NET] received packet: from 109.42.112.171[11000] to 95.119.147.36[500] (1268 bytes)
Jan 29 19:34:02 router ipsec: 01[ENC] parsed IKE_SA_INIT request 0 [ SA KE No N(NATD_S_IP) N(NATD_D_IP) N(FRAG_SUP) N(HASH_ALG) N(REDIR_SUP) ]
Jan 29 19:34:02 router ipsec: 01[IKE] 109.42.112.171 is initiating an IKE_SA
Jan 29 19:34:02 router ipsec: 01[CFG] selected proposal: IKE:AES_CBC_256/HMAC_SHA2_384_192/PRF_HMAC_SHA2_384/MODP_3072
Jan 29 19:34:02 router ipsec: 01[IKE] remote host is behind NAT
Jan 29 19:34:02 router ipsec: 01[ENC] generating IKE_SA_INIT response 0 [ SA KE No N(NATD_S_IP) N(NATD_D_IP) N(FRAG_SUP) N(HASH_ALG) N(CHDLESS_SUP) N(MULT_AUTH) ]
Jan 29 19:34:02 router ipsec: 01[NET] sending packet: from 95.119.147.36[500] to 109.42.112.171[11000] (600 bytes)
Jan 29 19:34:03 router ipsec: 15[NET] received packet: from 109.42.112.171[3326] to 95.119.147.36[4500] (1356 bytes)
Jan 29 19:34:03 router ipsec: 15[ENC] parsed IKE_AUTH request 1 [ EF(1/3) ]
Jan 29 19:34:03 router ipsec: 15[ENC] received fragment #1 of 3, waiting for complete IKE message
Jan 29 19:34:03 router ipsec: 07[NET] received packet: from 109.42.112.171[3326] to 95.119.147.36[4500] (1356 bytes)
Jan 29 19:34:03 router ipsec: 07[ENC] parsed IKE_AUTH request 1 [ EF(2/3) ]
[... cut ...]
Jan 29 20:34:07 router dnsmasq-dhcp[9377]: DHCPDISCOVER(br-lan) 7a:a7:0c:c0:aa:64
Jan 29 20:34:07 router dnsmasq-dhcp[9377]: DHCPOFFER(br-lan) 192.168.1.201 7a:a7:0c:c0:aa:64
Jan 29 20:34:10 router dnsmasq-dhcp[9377]: DHCPDISCOVER(br-lan) 7a:a7:0c:c0:aa:64
Jan 29 20:34:10 router dnsmasq-dhcp[9377]: DHCPOFFER(br-lan) 192.168.1.201 7a:a7:0c:c0:aa:64
Jan 29 20:34:14 router dnsmasq-dhcp[9377]: DHCPDISCOVER(br-lan) 7a:a7:0c:c0:aa:64
Jan 29 20:34:14 router dnsmasq-dhcp[9377]: DHCPOFFER(br-lan) 192.168.1.201 7a:a7:0c:c0:aa:64
```

tobiasbrunner, 2025-01-30:
Thanks for the details. As far as I can tell, this looks fine. But just to clarify, the settings you posted above are the only place where the `dhcp` plugin is configured? For instance, there is nothing in `strongswan.conf` itself or any of the other files that are included by it? If so, the question is what difference there is to @lowjoel's setup that could affect this. Maybe some firewall rules? Or some `sysctl` settings?

DocMAX, 2025-01-30:
Like @lowjoel my config used to work on v23.05. Nothing changed here. I try to tinker a bit more but right now no idea what to do now.

strongswan.conf looks like this:

```
charon {
    load_modular = yes
    dns1 = 10.0.0.1
    nbns1 = 10.0.0.1
    plugins {
        include strongswan.d/charon/*.conf
    }
}

include strongswan.d/*.conf
include /var/ipsec/strongswan.conf
```

lowjoel, 2025-01-30:
Another thing to check is to bump the logging for NET because that's what the pf_handler messages will appear as AFAIK. I also added logging in the patches I made to make sure the right binary is loaded. To be clear, pf_handler changes need an updated `libcharon.so`.

DocMAX, 2025-01-30:
OK, it works. My fault. I didn't flash the correct rom file. Applied the last 2 patches of the "dhcp-receive" strongswan branch.
So not sure which one is responsible for the fix.
So please add those patches officially to the openwrt strongswan package.
Thank you.

lowjoel, 2025-01-30:
@tobiasbrunner, since @DocMAX has tested it and the second patch looks reasonable to me (I haven't had a chance to test), perhaps you can merge it first and then I can open PRs to queue these for 5.9.14 in OpenWrt. Or else @DocMAX can open the PRs too, that's fine.

tobiasbrunner, 2025-01-31:
> OK, it works.

Great, thanks for testing.

> Applied the last 2 patches of the "dhcp-receive" strongswan branch.
> So not sure which one is responsible for the fix.

Only the first one is necessary to fix the issue (i.e. to make the existing configuration work again).

The other did actually not have an effect at all. Because some testing showed that binding packet sockets via `SO_BINDTODEVICE` really doesn't work at all. So the patch I posted here before, which uses `bind()`, is necessary to enforce any binding on the receive socket. I've now merged that and the other commits (https://github.com/strongswan/strongswan/commit/c2e5c00df336ea45bdfede66499688abd5ec6c3e, or just the fix https://github.com/strongswan/strongswan/commit/abbf9d28b0032cf80b79bcacea3146a60800a6dd).

commit eb91f914 (2025-04-07): strongswan: DHCP on lo fixes backport

Fixes #25801. Adds the following commits to fix DHCP behaviour on
Strongswan 5.9.14:

 - https://github.com/strongswan/strongswan/commit/abbf9d28b0032cf80b79bcacea3146a60800a6dd
 - https://github.com/strongswan/strongswan/commit/00d8c36d6fdf9e8ee99b9f92a64e7e81dbfa4432
 - https://github.com/strongswan/strongswan/commit/a50ed3006e8152eb2cf20e9f92f088ecc18081b0
PR #26274 (2025-04-09): strongswan: DHCP on lo fixes backport
Fixes #25801. Adds the following commits to fix DHCP behaviour on Strongswan 5.9.14:

 - https://github.com/strongswan/strongswan/commit/abbf9d28b0032cf80b79bcacea3146a60800a6dd
 - https://github.com/strongswan/strongswan/commit/00d8c36d6fdf9e8ee99b9f92a64e7e81dbfa4432
 - https://github.com/strongswan/strongswan/commit/a50ed3006e8152eb2cf20e9f92f088ecc18081b0

(cherry picked from commit eb91f914b9eaca8886fd69361602fa4c882d2061 #25870)

Maintainer: @pprindeville @Thermi 
Compile tested: aarch64, Linksys E8450, OpenWrt 24.10
Run tested:
commit 4c3068cc (2025-04-09): strongswan: DHCP on lo fixes backport

Fixes #25801. Adds the following commits to fix DHCP behaviour on
Strongswan 5.9.14:

 - https://github.com/strongswan/strongswan/commit/abbf9d28b0032cf80b79bcacea3146a60800a6dd
 - https://github.com/strongswan/strongswan/commit/00d8c36d6fdf9e8ee99b9f92a64e7e81dbfa4432
 - https://github.com/strongswan/strongswan/commit/a50ed3006e8152eb2cf20e9f92f088ecc18081b0

(cherry picked from commit eb91f914b9eaca8886fd69361602fa4c882d2061)
closed 2025-04-07 by pprindeville