openwrt/packages #25820: cros-vboot: can I get this package merged?

computersforpeace, 2025-01-23:
I assume this is getting buried because I opened it years ago:

https://github.com/openwrt/packages/pull/12829

It's still useful to people tinkering on OnHub and Google WiFi devices. For example, this thread: https://forum.openwrt.org/t/tp-link-1900-onhub-wont-boot-with-a-usb-flash-connected/219682/18?u=bnorris

BKPepe, 2025-05-20:
Hmmm... :) For now, let's close this issue otherwise we will be burried with issues for ages. 🙂 There are better ways how you can ask for review.

closed 2025-05-20 by BKPepe

computersforpeace, 2025-05-20:
> Hmmm... :) For now, let's close this issue otherwise we will be burried with issues for ages. 🙂

Ack :)

> There are better ways how you can ask for review.

@BKPepe If you could humor me: what are those ways? I've been trying for years now with no luck. Do people care about packages.git on IRC? openwr-devel? Forums?