openwrt/packages #25864: dnsproxy v0.75.0 r-1

PerkelSimon, 2025-01-31:
@1715173329 

I've upgraded the package for myself as usual to the last version, and I always keep the same setting, but now, error, does not work.

I change nothing on my setting and we upgraded to 0.75 and here the error:

`parsing options: positional arguments are not allowed, please check your command line arguments; detected positional arguments: [false] Thu Jan 30 21:04:30 2025 [daemon.info](http://daemon.info/) dnsproxy[26630]: jail: jail (26631) exited with exit: 2`

1715173329, 2025-01-31:
Please paste your configuration.

PerkelSimon, 2025-01-31:
> Please paste your configuration.

thanks,. as I said it is the same since a while. BTW when I re install the 0.74.1 version, that config works.
Also you can check the second config, it is the default one, even with that one, it is not working.

`config dnsproxy 'global'
	option enabled '1'
	list listen_addr '127.0.0.1'
#	list listen_addr '::1'
	list listen_port '5353'
	option log_file ''  
	option http3 '1'
	option insecure '0'
	option ipv6_disabled '1'
	option timeout ''
	option max_go_routines ''
	option rate_limit ''
	option refuse_any '0'
	option udp_buf_size ''
	option upstream_mode 'parallel'
#	option verbose '1'

config dnsproxy 'bogus_nxdomain'
	list ip_addr ''

config dnsproxy 'cache'
	option enabled '1'
	option cache_optimistic '1'
	option size '6553500'
	option min_ttl ''
	option max_ttl ''

config dnsproxy 'dns64'
	option enabled '0'
	option dns64_prefix '64:ff9b::'

config dnsproxy 'edns'
	option enabled '0'
	option edns_addr ''

config dnsproxy 'hosts'
	option enabled '0'
	list hosts_files ''

config dnsproxy 'private_rdns'
	option enabled '0'
	list upstream '127.0.0.1:53'

config dnsproxy 'servers'
        list bootstrap '45.90.28.233'
        list bootstrap '45.90.30.233'
	list fallback 'https://dns.cloudflare.com/dns-query'
#       list upstream 'https://dns.controld.com/1ex485tpr7o'
        list upstream 'quic://ninjanoir-myID.dns.nextdns.io' 
#	list upstream 'h3://security.Cloudflare-dns.com/dns-query'
#        list upstream '[/pool.ntp.org/]1.1.1.1'
#        list upstream 'https://dns9.quad9.net/dns-query'

config dnsproxy 'tls'
	option enabled '0'
	option tls_crt ''
	option tls_key ''
	option https_port '8443'
	option tls_port '853'
	option quic_port '853'`

`# For documents, please see https://github.com/AdguardTeam/dnsproxy#usage

config dnsproxy 'global'
	option enabled '1'
	list listen_addr '127.0.0.1'
#	list listen_addr '::1'
	list listen_port '5353'
	option log_file ''
	option http3 '1'
	option insecure '0'
	option ipv6_disabled '1'
	option timeout ''
	option max_go_routines ''
	option rate_limit ''
	option refuse_any '0'
	option udp_buf_size ''
	option upstream_mode ''
	option verbose '0'

config dnsproxy 'bogus_nxdomain'
	list ip_addr ''

config dnsproxy 'cache'
	option enabled '1'
	option cache_optimistic '1'
	option size '6553500'
	option min_ttl ''
	option max_ttl ''

config dnsproxy 'dns64'
	option enabled '0'
	option dns64_prefix '64:ff9b::'

config dnsproxy 'edns'
	option enabled '0'
	option edns_addr ''

config dnsproxy 'hosts'
	option enabled '0'
	list hosts_files ''

config dnsproxy 'private_rdns'
	option enabled '0'
	list upstream '127.0.0.1:53'

config dnsproxy 'servers'
	list bootstrap 'tls://8.8.8.8'
	list fallback 'tls://9.9.9.9'
	list upstream 'tls://1.1.1.1'

config dnsproxy 'tls'
	option enabled '0'
	option tls_crt ''
	option tls_key ''
	option https_port '8443'
	option tls_port '853'
	option quic_port '853'
`

1715173329, 2025-02-03:
```patch
diff --git a/net/dnsproxy/files/dnsproxy.init b/net/dnsproxy/files/dnsproxy.init
index 86062d142a6a..71a95f24f207 100644
--- a/net/dnsproxy/files/dnsproxy.init
+++ b/net/dnsproxy/files/dnsproxy.init
@@ -128,9 +128,9 @@ start_service() {
 
 	# This must be set at last, all other options set after this will be ignored
 	is_enabled "hosts" "enabled" && {
-		append_param "--hosts-file-enabled" "true"
+		append_param "--hosts-file-enabled=true"
 		config_list_foreach "hosts" "hosts_files" "append_param '--hosts-files'"
-	} || append_param "--hosts-file-enabled" "false"
+	} || append_param "--hosts-file-enabled=false"
 
 	procd_set_param respawn
 	procd_set_param stdout 1
```

apply this patch and it should work then

Idesmi, 2025-02-03:
From the release notes:

> Positional (non-flag) arguments now cause an error instead of being ignored

Introduced by commit https://github.com/AdguardTeam/dnsproxy/commit/429c98ccd1d290af702bcc647a5372b193b9fe8b

I believe your error is due to the value "true" not being recognised as bool, rather as a positional argument.

PerkelSimon, 2025-02-03:
> ```patch
> diff --git a/net/dnsproxy/files/dnsproxy.init b/net/dnsproxy/files/dnsproxy.init
> index 86062d142a6a..71a95f24f207 100644
> --- a/net/dnsproxy/files/dnsproxy.init
> +++ [...]

Works great thanks

closed 2025-02-09 by 1715173329