openwrt/packages #25875: package frr-vtysh is missing from 23.* repositories

asvdvl, 2025-02-01:
Good day, I encountered a problem that I can not install the frr-vtysh package on my mips_24kc router with firmware version 23.05.4. Understanding the problem, I discovered the absence of the package in all repositories of version 23 for any architecture.

NNET-ELi, 2025-02-04:
@asvdvl Hi The latest version of FRR no longer includes this package. 
If using the ripd and staticd functions, it is recommended to wait for frr_10.2.1-2 version

asvdvl, 2025-02-12:
@NNET-ELi Thanks! Unfortunately I can't find any mention of this, but I found that the `vtysh` command is already available for calling, and the file belongs to the main package:
```bash
root@internet-gw:~# opkg search /usr/bin/vtysh
frr - 10.2.1-1
```

closed 2025-02-12 by asvdvl