openwrt/packages #25886: freeradius3-mod-sql-sqlite broken on 24.10-rc7

luziferius, 2025-02-02:
# Symptom

Starting radiusd fails with
```
/etc/freeradius3/mods-config/sql/main/sqlite/queries.conf[368]: Failed parsing expanded string:
/etc/freeradius3/mods-config/sql/main/sqlite/queries.conf[368]: ...efix}', '%{Framed-Interface-Id}', '%{Delegated-IPv6-Prefix}' )
/etc/freeradius3/mods-config/sql/main/sqlite/queries.conf[368]:                                         ^ Unknown attribute
```
if freeradius3-mod-sql-sqlite is installed and freeradius configured to read users from an SQLite3 database.

# Needed fix

File `/usr/share/freeradius3/dictionary.rfc4818` is missing as a file and also needs to be uncommented in `/usr/share/freeradius3/dictionary`:

```diff
--- a/dictionary
+++ b/dictionary
@@ -98,7 +98,7 @@ $INCLUDE dictionary.rfc4372
 #$INCLUDE dictionary.rfc4603
 $INCLUDE dictionary.rfc4675
 $INCLUDE dictionary.rfc4679
-#$INCLUDE dictionary.rfc4818
+$INCLUDE dictionary.rfc4818
 #$INCLUDE dictionary.rfc4849
 #$INCLUDE dictionary.rfc5176
 #$INCLUDE dictionary.rfc5447
```

sinancetinkaya, 2025-04-09:
Has anyone, for real, got freeradius working on OpenWRT? (with `Simultaneous-Use := "1"` using radutmp or `freeradius3-mod-sql-sqlite`)

sinancetinkaya, 2025-04-09:
I have an external driver for that. But I can't get `Simultaneous-Use := "1"` to work with neither `freeradius3-mod-radutmp` nor `freeradius3-mod-sql-sqlite`.

sinancetinkaya, 2025-04-09:
The same problem was mentioned here too https://forum.openwrt.org/t/freeradius3-simultaneous-use-not-working-for-concurrent-connections/149760

BKPepe, 2025-05-23:
I see that freeradius3 was recently updated by @systemcrash . Maybe he can look at it as freeradius3 does not have currently maintainer in OpenWrt.