openwrt/packages #25906: frr: fails to start due to missing frr-watchfrr package

brianjmurrell, 2025-02-05:
It seems that I can ask an imagebuilder recipe to install `frr`, `frr-ospfd`, and `frr-zebra` and only those packages.  However when it comes time to starting _frr_ the following error is encountered:

```
Mon Feb  3 18:10:32 2025 daemon.notice procd: /etc/rc.d/S95frr: cannot start staticd: daemon binary not installed
Mon Feb  3 18:10:32 2025 daemon.notice procd: /etc/rc.d/S95frr: Starting watchfrr with command: '  /usr/sbin/watchfrr  -d  -F traditional   zebra mgmtd'
Mon Feb  3 18:10:32 2025 daemon.notice procd: /etc/rc.d/S95frr: /etc/rc.common: eval: line 220: /usr/sbin/watchfrr: not found
Mon Feb  3 18:10:32 2025 daemon.notice procd: /etc/rc.d/S95frr: Failed to start watchfrr!
Mon Feb  3 18:10:33 2025 daemon.info procd: - init complete -
```

Clearly `frr-watchfrr` is a prerequisite here and should be installed.  But the user should not have to know and specify this.  Rather, one of the packages I requested be installed, `frr`, `frr-ospfd`, and `frr-zebra` should have a dependency on `frr-watchfrr` if it's needed to start up _frr_.

lucize, 2025-02-07:
yeah, you could start them by hand, but yes, I guess we should make it enabled by default

brianjmurrell, 2025-02-07:
I agree.  Particularly given that it's the initscript that is wanting `watchfrr`.

> yeah, you could start them by hand, but yes, I guess we should make it enabled by default

Indeed.  If you wanted to modify the initscript to start them individually start them without `watchfrr`, then I'd agree, but ultimately the initscript should just work and package dependencies should be set up to make sure that happens IMO.

brianjmurrell, 2025-04-16:
Any progress resolving this one?

closed 2025-06-30 by 1715173329