openwrt/packages #25907: frr: Failed to open commit history "/var/lib/frr/commit-index.dat" for writing: No such file or directory

brianjmurrell, 2025-02-06:
When I start _frr_ with `/etc/init.d/frr start` I see errors in the system log:

```
Wed Feb  5 18:50:56 2025 daemon.err mgmtd[3257]: libyang: Failed to open file "/var/lib/frr/commit-20250205185056384052348.json" (No such file or directory).
Wed Feb  5 18:50:56 2025 daemon.err mgmtd[3257]: [HAWY9-CP261] Failed to open commit history "/var/lib/frr/commit-index.dat" for writing: No such file or directory
```

Any idea what those are about?  They seem alarming no matter what they are.  on a 24.10.0 target there doesn't seem to be even a `/var/lib/` never mind `/var/lib/frr/`.  Should that path exist?  Or should `mgmtd` be told somewhere else to be trying to write that data?

lucize, 2025-02-07:
do you have only one config file?
didn't try 24.10, but tested on master and 23.05 release, will test 24 sometimes

can you post your config?

brianjmurrell, 2025-02-07:
I'm not sure I am reproducing this any more.  Let's close it if and until it's see again.

brianjmurrell, 2025-04-16:
Still seeing this.  These are all of the messages from boot until just shortly after the boot has completed:

```
Apr 16 15:09:27 masterbedroom_ap watchfrr[1933]: [QDG3Y-BY5TN] zebra state -> up : connect succeeded                                                                               
Apr 16 15:09:27 masterbedroom_ap watchfrr[1933]: [YFT0P-5Q5YX] Forked background command [pid 2130]: /usr/sbin/watchfrr.sh restart ospfd                                           
Apr 16 15:09:27 masterbedroom_ap watchfrr[1933]: [YFT0P-5Q5YX] Forked background command [pid 2131]: /usr/sbin/watchfrr.sh restart mgmtd                                           
Apr 16 15:09:27 masterbedroom_ap procd: /etc/rc.d/S95frr: Cannot stop ospfd: pid file not found                                                                                    
Apr 16 15:09:27 masterbedroom_ap procd: /etc/rc.d/S95frr: Cannot stop mgmtd: pid file not found                                                                                    
Apr 16 15:09:27 masterbedroom_ap watchfrr.sh: Cannot stop mgmtd: pid file not found
Apr 16 15:09:27 masterbedroom_ap watchfrr.sh: Cannot stop ospfd: pid file not found
Apr 16 15:09:29 masterbedroom_ap procd: /etc/rc.d/S95frr: 2025/04/16 15:09:29 MGMTD: [QYS7B-8K2EV][EC 100663303] failed to mkdir "/var/lib/frr": No such file or directory         
Apr 16 15:09:29 masterbedroom_ap procd: /etc/rc.d/S95frr: 2025/04/16 15:09:29 OSPF: [QYS7B-8K2EV][EC 100663303] failed to mkdir "/var/lib/frr": No such file or directory          
Apr 16 15:09:31 masterbedroom_ap procd: /etc/rc.d/S95frr: 2025/04/16 15:09:31 MGMTD: [QYS7B-8K2EV][EC 100663303] failed to mkdir "/var/lib/frr": No such file or directory         
Apr 16 15:09:33 masterbedroom_ap watchfrr[1933]: [QDG3Y-BY5TN] ospfd state -> up : connect succeeded                                                                               
Apr 16 15:09:34 masterbedroom_ap mgmtd[2233]: libyang: Failed to open file "/var/lib/frr/commit-20250416150934758629911.json" (No such file or directory).                         
Apr 16 15:09:34 masterbedroom_ap mgmtd[2233]: [HAWY9-CP261] Failed to open commit history "/var/lib/frr/commit-index.dat" for writing: No such file or directory                   
Apr 16 15:09:34 masterbedroom_ap watchfrr[1933]: [QDG3Y-BY5TN] mgmtd state -> up : connect succeeded                                                                               
Apr 16 15:09:34 masterbedroom_ap watchfrr[1933]: [KWE5Q-QNGFC] all daemons up, doing startup-complete notify                                                                       
Apr 16 15:09:34 masterbedroom_ap procd: /etc/rc.d/S95frr: Started watchfrr
Apr 16 15:09:35 masterbedroom_ap watchfrr.sh: Failed to start mgmtd!
Apr 16 15:09:36 masterbedroom_ap watchfrr.sh: Failed to start ospfd!
Apr 16 15:09:36 masterbedroom_ap watchfrr[1933]: [RKHTV-CNGEG] Daemon: zebra: is in Up state but expected it to be in DAEMON_DOWN state                                            
Apr 16 15:09:36 masterbedroom_ap watchfrr[1933]: [RKHTV-CNGEG] Daemon: mgmtd: is in Up state but expected it to be in DAEMON_DOWN state                                            
Apr 16 15:09:36 masterbedroom_ap watchfrr[1933]: [RKHTV-CNGEG] Daemon: ospfd: is in Up state but expected it to be in DAEMON_DOWN state                                            
```

As you can see, even in addition to the issue being reported here other things are going wrong:

```
Apr 16 15:09:27 masterbedroom_ap procd: /etc/rc.d/S95frr: Cannot stop ospfd: pid file not found                                                                                    
Apr 16 15:09:27 masterbedroom_ap procd: /etc/rc.d/S95frr: Cannot stop mgmtd: pid file not found                                                                                    
Apr 16 15:09:27 masterbedroom_ap watchfrr.sh: Cannot stop mgmtd: pid file not found
Apr 16 15:09:27 masterbedroom_ap watchfrr.sh: Cannot stop ospfd: pid file not found
Apr 16 15:09:29 masterbedroom_ap procd: /etc/rc.d/S95frr: 2025/04/16 15:09:29 MGMTD: [QYS7B-8K2EV][EC 100663303] failed to mkdir "/var/lib/frr": No such file or directory         
Apr 16 15:09:29 masterbedroom_ap procd: /etc/rc.d/S95frr: 2025/04/16 15:09:29 OSPF: [QYS7B-8K2EV][EC 100663303] failed to mkdir "/var/lib/frr": No such file or directory          
Apr 16 15:09:31 masterbedroom_ap procd: /etc/rc.d/S95frr: 2025/04/16 15:09:31 MGMTD: [QYS7B-8K2EV][EC 100663303] failed to mkdir "/var/lib/frr": No such file or directory         
```

My `/etc/frr/frr.conf`:

```
hostname masterbedroom_ap
password [redacted]
enable password [redacted]
!
router ospf
    ospf router-id 10.75.22.195
    network 10.75.22.0/24 area 0

    redistribute connected
    redistribute kernel
    redistribute static

!log file /tmp/ospfd-debug.txt

!debug ospf packet all
!debug ospf ism
!debug ospf nsm
!debug ospf event
!debug ospf nssa
!debug ospf lsa
!!debug ospf te
!debug ospf zebra

log syslog
!
access-list vty permit 127.0.0.0/8
access-list vty deny any
!
line vty
 access-class vty
```