openwrt/packages #25910: openwrt 24.10.0: Unable to install https-dns-proxy

docplastic, 2025-02-06:
Doing:
 opkg update
 opkg install https-dns-proxy

shows:
Unknown package 'https-dns-proxy'.
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency resolveip for https-dns-proxy
 * pkg_hash_fetch_best_installation_candidate: Packages for https-dns-proxy found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package https-dns-proxy.

**router info:**
 ubus call system board

shows:
{
	"kernel": "6.6.73",
	"hostname": "OpenWrt",
	"system": "ARMv8 Processor rev 4",
	"model": "Cudy WR3000S v1",
	"board_name": "cudy,wr3000s-v1",
	"rootfs_type": "squashfs",
	"release": {
		"distribution": "OpenWrt",
		"version": "24.10.0",
		"revision": "r28427-6df0e3d02a",
		"target": "mediatek/filogic",
		"description": "OpenWrt 24.10.0 r28427-6df0e3d02a",
		"builddate": "1738624177"
	}
}

p0k33m0n, 2025-02-07:
Due to an unprecedented number of build requests, the ASU server ran out of storage for all the requested images. This caused spurious build errors at first, then totally crashed the system, so if you had an issue with your build and others had success on the same target/profile, you might want to try again, maybe tomorrow when things have settled down a bit...

docplastic, 2025-02-07:
That seems to have happened, it now works as expected. Thank you.

closed 2025-02-07 by docplastic