openwrt/packages #25912: OpenWRT 24.10 Perl 5.40.0 debugger not finding symbols

lsoltero, 2025-02-06:
the perl debugger is not finding symbols

root@PW-Hub:/tmp# perl -v

This is perl 5, version 40, subversion 0 (v5.40.0) built for mipsel-linux-musl

root@PW-Hub:/tmp# cat test.pl 
#!/usr/bin/perl

sub a() {
    print "a\n";
}

sub b() {
    print "b\n";
}

a();
b();

root@PW-Hub:/tmp# perl -d test.pl 

Loading DB routines from perl5db.pl version 1.80
Editor support available.

Enter h or 'h h' for help, or 'man perldebug' for more help.

main::(test.pl:11):	a();
  DB<1> b a
Subroutine main::a not found.
  DB<2> b main::a
Subroutine main::a not found.
  DB<3> l 1
1 	#!/usr/bin/perl
  DB<4> l
2 	
3 	sub a() {
4:	    print "a\n";
5 	}
6 	
7 	sub b() {
8:	    print "b\n";
9 	}
10 	
11==>	a();
  DB<4> b 4
  DB<5> S
Carp::BEGIN
Carp::__ANON__[(eval 3)[/usr/lib/perl5/5.40/Carp.pm:%]:%]
Carp::__ANON__[(eval 4)[/usr/lib/perl5/5.40/Carp.pm:%]:%]

...

feature::unknown_feature_bundle
main::BEGIN
main::a
main::b
overloading::BEGIN
overloading::_ops_to_nums

...

 DB<5> l a
Invalid line specification '%'.
  DB<6>

lsoltero, 2025-02-06:
i should add that this is on an mt7621 mipsel SOC.  OpenWRT 20.10 release

root@PW-Hub:/usr/local/bin# uname -a
Linux PW-Hub 6.6.73 #0 SMP Mon Feb  3 23:09:37 2025 mips GNU/Linux

lsoltero, 2025-02-06:
additionally compile is not printing line numbers on error.  It prints % instead of the line number.

root@PW-Hub:/tmp# cat test.pl
#!/usr/bin/perl

sub a() {
    print "a\n";
}

sub b() {
    print "b\n";
}

a()
b();

root@PW-Hub:/tmp# perl test.pl
Semicolon seems to be missing at test.pl line %.
syntax error at test.pl line %, near ")
b"
Execution of test.pl aborted due to compilation errors.

predators46, 2025-02-09:
is it like this?

```
root@OpenWrt:~# perl -Mwarnings -E 'say $x'
Name "main::x" used only once: possible typo at -e line 1.
Use of uninitialized value $x in say at -e line 1.
root@OpenWrt:~#
```

lsoltero, 2025-02-09:
here is what I get when I execute this on mt7621 router running the release version of OpenWRT 24.10

root@PW-Hub:~# perl -Mwarnings -E 'say $x'
Name "main::x" used only once: possible typo at -e line %.
Use of uninitialized value $x in say at -e line %.

Any idea why this works for you and not me?  I am compiling the perl package from sources and have made no changes or added any patches to perl.

here is the last entry in git log for the packages.

commit 201fd099b80a2931b7326ce20b0cbb824296c99f (HEAD -> 201fd099b80a2931b7326ce20b0cbb824296c99f)
Author: Lucian CRISTIAN <lucian.cristian@gmail.com>
Date:   Sun Feb 2 13:35:32 2025 +0200

    frr: 10.2.1 fixes
    
    add mgmtd and pim6d
    
    
    Signed-off-by: Lucian CRISTIAN <lucian.cristian@gmail.com>

predators46, 2025-02-13:
```
root@OpenWrt:~# perl -v
This is perl 5, version 40, subversion 1 (v5.40.1) built for aarch64-linux-musl
Copyright 1987-2025, Larry Wall
Perl may be copied only under the terms of either the Artistic License or the
GNU General Public License, which may be found in the Perl 5 source kit.
Complete documentation for Perl, including FAQ lists, should be found on
this system using "man perl" or "perldoc perl".  If you have access to the
Internet, point your browser at https://www.perl.org/, the Perl Home Page.
root@OpenWrt:~# perl -Mwarnings -E 'say $x'
Name "main::x" used only once: possible typo at -e line 1.
Use of uninitialized value $x in say at -e line 1.
root@OpenWrt:~#
```

lsoltero, 2025-02-14:
what platform are you running on?  All I did was run menu config and select perl and then make and sys install the package.  when I do this with the stock perl 5.40.0 package I get the error above. If if I checkout the version in branch 23.05 it works fine.  No patches or changes are being made to the perl code or the build system.  This is a basic openwrt 24.10 release install compiled from sources.  used scripts/feeds update ; scripts/feeds install -a  to compile...

I see you are running 5.40.1.  the version that is installed for me is 5.40.0. was there a fix in 5.40.1 to address this?

lsoltero, 2025-02-14:
the latest version of perl here https://github.com/openwrt/packages  is 5.40.0

lsoltero@ubuntu22:~/openwrt/packages/lang/perl$ git pull
Already up to date.
lsoltero@ubuntu22:~/openwrt/packages/lang/perl$ git branch -a
* master
  remotes/origin/HEAD -> origin/master
  remotes/origin/for-14.07
  remotes/origin/for-15.05
  remotes/origin/lede-17.01
  remotes/origin/master
  remotes/origin/openwrt-18.06
  remotes/origin/openwrt-19.07
  remotes/origin/openwrt-21.02
  remotes/origin/openwrt-22.03
  remotes/origin/openwrt-23.05
  remotes/origin/openwrt-24.10
lsoltero@ubuntu22:~/openwrt/packages/lang/perl$ cat perlver.mk 
PERL_VERSION:=5.40.0

PERL_EXPLODE:=$(subst ., ,$(PERL_VERSION))

PERL_MAJOR:=$(word 1,$(PERL_EXPLODE))
PERL_MINOR:=$(word 2,$(PERL_EXPLODE))
PERL_REL:=$(word 3,$(PERL_EXPLODE))

PERL_VERSION3:=$(PERL_VERSION)
PERL_VERSION2:=$(PERL_MAJOR).$(PERL_MINOR)

predators46, 2025-02-26:
you have to fix it yourself or use version 5.28

The-BB, 2025-04-10:
try it
```
diff --git a/lang/perl/files/base.config b/lang/perl/files/base.config
index 67232b6d8..66a5d35c5 100644
--- a/lang/perl/files/base.config
+++ b/lang/perl/files/base.config
@@ -656,6 +656,7 @@ html1direxp=''
 html3dir=' '
 html3direxp=''
 i16size='2'
+i32dformat='"d"'
 i32size='4'
 i64size='8'
 i8size='1'
@@ -976,6 +977,10 @@ trnl='\n'
 troff=''
 u16size='2'
 u16type='unsigned short'
+u32XUformat='"X"'
+u32oformat='"o"'
+u32uformat='"u"'
+u32xformat='"x"'
 u32size='4'
 u64size='8'
 u8size='1'
```

jw2013, 2026-01-03:
I've tested with the updated `architecture.config` (renamed to .txt for github)
[architecture.config.txt](https://github.com/user-attachments/files/24416862/architecture.config.txt)

Compiled package/perl via `openwrt-sdk-24.10.1-bcm27xx-bcm2710_gcc-13.3.0_musl.Linux-x86_64`, and installed the package on a RPi3 with OpenWrt 24.10. Tested, works:

```
root@OpenWrt:~# opkg install /tmp/perl_5.40.0-r4_aarch64_cortex-a53.ipk 
Upgrading perl on root from 5.40.0-r2 to 5.40.0-r4...
Configuring perl.
root@OpenWrt:~# perl
print "TesT\n";
print Test
;
use Lasdas;
Can't locate Lasdas.pm in @INC (you may need to install the Lasdas module) (@INC entries checked: /usr/lib/perl5/5.40) at - line 4.
BEGIN failed--compilation aborted at - line 4.

```

@pprindeville , shall I amend the perlbase PR that I opened yesterday, and add the fix to it?