openwrt/packages #25921: OpenWrt 24.10.0:  Curl and Wget fail on some https sites, openssl does not

Jeronimo17, 2025-02-08:
Back to the error I was supposed to get 😞  
https://github.com/openwrt/packages/issues/16652

Environment:  MediaTek MT7621 ver:1 eco:3 OpenWrt 24.10.0 r28427-6df0e3d02a / LuCI openwrt-24.10 branch 25.037.68331~42f464c
```

root@OpenWrt:/tmp/t# curl https://www.boe.es -v
* ssl_handshake returned: (-0x7080) SSL - The requested feature is not available
curl: (35) ssl_handshake returned: (-0x7080) SSL - The requested feature is not available
root@OpenWrt:/tmp/t# wget https://www.boe.es
Downloading 'https://www.boe.es'
Connecting to 81.89.32.200:443
SSL error: SSL - The requested feature is not available
Connection error: Connection failed
root@OpenWrt:/tmp/t# openssl s_client -connect boe.es:443
CONNECTED(00000003)
depth=2 C = US, ST = New Jersey, L = Jersey City, O = The USERTRUST Network, CN = USERTrust RSA Certification Authority
verify return:1
depth=1 C = NL, O = GEANT Vereniging, CN = GEANT OV RSA CA 4
verify return:1
depth=0 C = ES, ST = Madrid, O = Bolet\C3\ADn Oficial del Estado, CN = www.boe.es
verify return:1
---
Certificate chain
 0 s:C = ES, ST = Madrid, O = Bolet\C3\ADn Oficial del Estado, CN = www.boe.es
   i:C = NL, O = GEANT Vereniging, CN = GEANT OV RSA CA 4
   a:PKEY: rsaEncryption, 2048 (bit); sigalg: RSA-SHA384
   v:NotBefore: Apr 25 00:00:00 2024 GMT; NotAfter: Apr 25 23:59:59 2025 GMT
 1 s:C = NL, O = GEANT Vereniging, CN = GEANT OV RSA CA 4
   i:C = US, ST = New Jersey, L = Jersey City, O = The USERTRUST Network, CN = USERTrust RSA Certification Authority
   a:PKEY: rsaEncryption, 4096 (bit); sigalg: RSA-SHA384
   v:NotBefore: Feb 18 00:00:00 2020 GMT; NotAfter: May  1 23:59:59 2033 GMT
---
Server certificate
-----BEGIN CERTIFICATE-----
MIIHUTCCBTmgAwIBAgIRALLJRdkMMwnUA0GjGzMJd/wwDQYJKoZIhvcNAQEMBQAw
RDELMAkGA1UEBhMCTkwxGTAXBgNVBAoTEEdFQU5UIFZlcmVuaWdpbmcxGjAYBgNV
BAMTEUdFQU5UIE9WIFJTQSBDQSA0MB4XDTI0MDQyNTAwMDAwMFoXDTI1MDQyNTIz
NTk1OVowWTELMAkGA1UEBhMCRVMxDzANBgNVBAgTBk1hZHJpZDEkMCIGA1UECgwb
Qm9sZXTDrW4gT2ZpY2lhbCBkZWwgRXN0YWRvMRMwEQYDVQQDEwp3d3cuYm9lLmVz
MIIBIjANBgkqhkiG9w0BAQEFAAOCAQ8AMIIBCgKCAQEAnHlcLeNXrHYRelqZ3YNP
RV7aGCkgVj0QgKDRumtGQppCVj/A5ycOh2HG270bWqkX3af76hoDUJ0ePSvAPXC9
+jlgduKipjNT5eWLjudHsy4wJuhLWjLDgOREm7jWQXmE2NJLd23AUgnEzgzmuHbe
M+i9KiW44w88x1ENP898rAXf77NF7LpufBeylbWTVzBkUOKGn9tpDaDiX0FvSf3C
+wmagVaYZxd946GJEECk9VZzVrixjiIGTFDC2RXKB9H0HK/Rwnfz+iLAo9l+fBJp
DqbxOUjJ5MUulu6TDNrAr790RjQoj8p2EU7LCJeRZZsakUNU+6wexvEsh41w/i04
PwIDAQABo4IDJzCCAyMwHwYDVR0jBBgwFoAUbx01SRBsMvpZoJ68iugflb5xegww
HQYDVR0OBBYEFH9XjxlrwzfL0SHRCS+KS0UCUrCPMA4GA1UdDwEB/wQEAwIFoDAM
BgNVHRMBAf8EAjAAMB0GA1UdJQQWMBQGCCsGAQUFBwMBBggrBgEFBQcDAjBJBgNV
HSAEQjBAMDQGCysGAQQBsjEBAgJPMCUwIwYIKwYBBQUHAgEWF2h0dHBzOi8vc2Vj
dGlnby5jb20vQ1BTMAgGBmeBDAECAjA/BgNVHR8EODA2MDSgMqAwhi5odHRwOi8v
R0VBTlQuY3JsLnNlY3RpZ28uY29tL0dFQU5UT1ZSU0FDQTQuY3JsMHUGCCsGAQUF
BwEBBGkwZzA6BggrBgEFBQcwAoYuaHR0cDovL0dFQU5ULmNydC5zZWN0aWdvLmNv
bS9HRUFOVE9WUlNBQ0E0LmNydDApBggrBgEFBQcwAYYdaHR0cDovL0dFQU5ULm9j
c3Auc2VjdGlnby5jb20wHQYDVR0RBBYwFIIKd3d3LmJvZS5lc4IGYm9lLmVzMIIB
gAYKKwYBBAHWeQIEAgSCAXAEggFsAWoAdgDPEVbu1S58r/OHW9lpLpvpGnFnSrAX
7KwB0lt3zsw7CAAAAY8UR+RtAAAEAwBHMEUCIQCfF/j27O/3NFuRSWroqtNFHbsU
TOb2XZQy5kx3KAISAQIgFbi6xr5LFw1bpY4+4rPmWR1C+21tNj3uPjqEp/7LeN0A
dwCi4wrkRe+9rZt+OO1HZ3dT14JbhJTXK14bLMS5UKRH5wAAAY8UR+QXAAAEAwBI
MEYCIQCskvfK7KST9DWTM4+IH0Hal7v2O/b1zNV5vWjF6Cf5MgIhANe7B4XqJQRk
JBsTqozSgo9YH1Jfx7A+qxUqsndu50j3AHcATnWjJ1yaEMM4W2zU3z9S6x3w4I4b
jWnAsfpksWKaOd8AAAGPFEfkFQAABAMASDBGAiEA6FQqb8qBLIp0jhF0ucnDPbfj
rGoYgrxwT74CAzoJcngCIQD456Xu03gU1+b2QnZZ41ehCO0ZvfNn5cduXmISXNRa
NTANBgkqhkiG9w0BAQwFAAOCAgEAfT8MeiLXP4PzqsbxofBpKRMg1aYWmWry+Udo
YGRBzP57Rd5EknyW5LPqXvN5illSGCQdGs32Qht2F95Aa5Q3iK31FBAQaTk34bxE
kuR0QCT0G6p2psHU2L05FBufeo8aslPYvBwRJIQeziAgc7/AfV3C4tp7+3Jd72Ow
yceWnarohMum2wtQMo/Q3cJEJko+Uc8sfTsvT49DxvXJZbPET2Eux3V1SBICK49X
9Cr1mDZdYL8ism9jbVs/UNs6TigVWt02BzCYNuElrgU7FprHDPFTzLxXcBpUr4mx
/6JHgTbW65cR25mwN8t3brzvNxYKAXz9Fuk3uWg/LrVymwdhEzm4it9HjAPoscPH
Gi8rT2ajjW4EdquAMMno5V51JzacfNWMAPA5ylN7EuH1TY4CAAYTtsz5pun6CNW2
serUNXpVz9GB0DabKx5MI+4KfgcNISWcYfciRQ0RJGFWUWkz7ixydNmxlBzBqXGo
DJUG3pyvktUQ6p2nuV1ip8tdpI5sVL2qbHctC/h13NKbhf9Slq0Vl+jCcGXSvtRI
Lyh18o98tBnN/ZwNkvHYqF+rzcwzx5PIXH0qdNStIMTes3cEAvxMFYpRQgKGdU1R
CgGk2myWudu4XCrVcc90Bi8tbN21O9U3DrcFweTMXfa3nc5mxLu8rwmcRe2OdSzi
22rUEYs=
-----END CERTIFICATE-----
subject=C = ES, ST = Madrid, O = Bolet\C3\ADn Oficial del Estado, CN = www.boe.es
issuer=C = NL, O = GEANT Vereniging, CN = GEANT OV RSA CA 4
---
No client certificate CA names sent
Peer signing digest: SHA256
Peer signature type: RSA-PSS
Server Temp Key: X25519, 253 bits
---
SSL handshake has read 4311 bytes and written 392 bytes
Verification: OK
---
New, TLSv1.3, Cipher is TLS_AES_256_GCM_SHA384
Server public key is 2048 bit
Secure Renegotiation IS NOT supported
No ALPN negotiated
Early data was not sent
Verify return code: 0 (ok)
---
---
Post-Handshake New Session Ticket arrived:
SSL-Session:
    Protocol  : TLSv1.3
    Cipher    : TLS_AES_256_GCM_SHA384
    Session-ID: A712C623732602BD0C1F21F5A1779B662D262D79856816ECF9A802B968C8FAB1
    Session-ID-ctx: 
    Resumption PSK: 6998186A775DC38E7F36A7DB1B117774C9EB826A34321454D1DEA79E3716D250990190E297C69FD2630B9C8FE50AF0CF
    PSK identity: None
    PSK identity hint: None
    SRP username: None
    TLS session ticket lifetime hint: 7200 (seconds)
    TLS session ticket:
    0000 - 3b 72 4c 2e 2f 88 f7 d2-b8 c2 43 77 ff 9a 9e b4   ;rL./.....Cw....
    0010 - d6 3e 94 eb 2c 9d 39 41-af 70 65 c9 8b ba 2d 74   .>..,.9A.pe...-t

    Start Time: 1739024064
    Timeout   : 7200 (sec)
    Verify return code: 0 (ok)
    Extended master secret: no
    Max Early Data: 0
---
read R BLOCK
GET / HTTP/1.0
Host:boe.es

HTTP/1.1 200 OK
Server: nginx
Date: Sat, 08 Feb 2025 14:13:36 GMT
Content-Type: text/html; charset=UTF-8
Content-Security-Policy: default-src afirma://* ; script-src 'unsafe-inline' 'unsafe-eval' 'self' https://www.google.com/recaptcha/ https://www.gstatic.com/recaptcha/ ; frame-src 'self' https://www.google.com/recaptcha/ https://recaptcha.google.com/recaptcha/  afirma://* ; connect-src 'self' ; img-src 'self' eur-lex.europa.eu data: ; style-src 'unsafe-inline' 'self' *.boe.es ; font-src 'self' ; child-src 'self'  www.youtube.com afirma://* ; object-src 'self'; media-src 'self'
X-Varnish: 710219320 706352744
Age: 67
Via: 1.1 cachem1.boe.es (Varnish/7.2)
Accept-Ranges: bytes
Content-Length: 22646
Connection: close

....
....
....
        </div> <!-- .franjaMenu -->
      </div> <!-- #menuPie -->
      <div class="franja-pie">
        <p class="nombre-organismo">Agencia Estatal Bolet&iacute;n Oficial del Estado</p>
        <p class="direccion-organismo"><abbr title="Avenida">Avda.</abbr> de Manoteras, 54 - 28050 Madrid</p>
      </div>
    </div> <!-- #pie -->
    <script src="/js/desplegable.js"></script>
  </body>
</html>
84EDE877:error:0A000126:SSL routines:ssl3_read_n:unexpected eof while reading:ssl/record/rec_layer_s3.c:322:
```

Up to version 23.05.5 it did not fail.
Thank you

Jeronimo17, 2025-02-08:
Provisional working installing the curl and libraries of version **23.05.5**

rscarvalho90, 2025-02-10:
The same occurs for Instagram when acessing via IPv6 (Android App). Via IPv4 the connection occurs normally.
As bellow, connection to Google via HTTPS using IPv6 occurs normally.

```
root@OpenWrt:~# uclient-fetch https://www.instagram.com
Downloading 'https://www.instagram.com'
Connecting to 2a03:2880:f205:e5:face:b00c:0:4420:443
SSL error: SSL - The requested feature is not available
Connection error: Connection failed
root@OpenWrt:~# uclient-fetch https://www.facebook.com
Downloading 'https://www.facebook.com'
Connecting to 2a03:2880:f148:181:face:b00c:0:25de:443
SSL error: SSL - The requested feature is not available
Connection error: Connection failed
root@OpenWrt:~# uclient-fetch https://www.google.com
Downloading 'https://www.google.com'
Connecting to 2800:3f0:4001:83c::2004:443
Writing to 'index.html'

Download completed (19304 bytes)
```

Catfriend1, 2025-06-24:
@[Catfriend1](https://github.com/Catfriend1)
Catfriend1
[last month](https://github.com/openwrt/packages/commit/43b9a37a6ef0c69e834cf2967a3796804b7e3a48#commitcomment-158054568)
Something is broken since then, digest auth is not taking place or runs into a problem. See https://forum.openwrt.org/t/curl-digest-auth-broken/230370

Catfriend1 commented last month
@[Catfriend1](https://github.com/Catfriend1)
Catfriend1
[last month](https://github.com/openwrt/packages/commit/43b9a37a6ef0c69e834cf2967a3796804b7e3a48#commitcomment-158161590)
@feckert Workaround: Take the file "libcurl.so.4.8.0" from "OpenWrt r28926-9a7192c08e" and copy it into "/usr/lib" of a recent snapshot build. curl digest auth works again :-)