openwrt/packages #25926: Firmware selector custom build fails on installing packages (checksum mismatch)

Belkaar, 2025-02-09:
This is the error log:

Generate local signing keys...
Generate local certificate...
Package list missing or not up-to-date, generating it.

```
Building package index...
Downloading http://downloads.openwrt.org/releases/24.10.0/targets/rockchip/armv8/packages/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_generic_musl/root-rockchip/../../../../builder/dl/openwrt_core
Downloading http://downloads.openwrt.org/releases/24.10.0/targets/rockchip/armv8/packages/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/aarch64_generic/base/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_generic_musl/root-rockchip/../../../../builder/dl/openwrt_base
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/aarch64_generic/base/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/24.10.0/targets/rockchip/armv8/kmods/6.6.73-1-f35e93bc2c89b98d107e57cdea041972/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_generic_musl/root-rockchip/../../../../builder/dl/openwrt_kmods
Downloading http://downloads.openwrt.org/releases/24.10.0/targets/rockchip/armv8/kmods/6.6.73-1-f35e93bc2c89b98d107e57cdea041972/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/aarch64_generic/luci/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_generic_musl/root-rockchip/../../../../builder/dl/openwrt_luci
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/aarch64_generic/luci/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/aarch64_generic/packages/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_generic_musl/root-rockchip/../../../../builder/dl/openwrt_packages
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/aarch64_generic/packages/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/aarch64_generic/routing/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_generic_musl/root-rockchip/../../../../builder/dl/openwrt_routing
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/aarch64_generic/routing/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/aarch64_generic/telephony/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_generic_musl/root-rockchip/../../../../builder/dl/openwrt_telephony
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/aarch64_generic/telephony/Packages.sig
Signature check passed.
Downloading file:packages/Packages
Updated list of available packages in /builder/build_dir/target-aarch64_generic_musl/root-rockchip/../../../../builder/dl/imagebuilder
Downloading file:packages/Packages.sig
Signature check passed.
Collected errors:
 * opkg_install_pkg: Checksum or size mismatch for package busybox. Either the opkg or the package index are corrupt. Try 'opkg update'.
 * opkg_install_cmd: Cannot install package base-files.
 * opkg_install_pkg: Checksum or size mismatch for package cgi-io. Either the opkg or the package index are corrupt. Try 'opkg update'.
 * opkg_install_cmd: Cannot install package luci.
 * opkg_install_pkg: Checksum or size mismatch for package parted. Either the opkg or the package index are corrupt. Try 'opkg update'.
 * opkg_install_cmd: Cannot install package parted.
make[2]: *** [Makefile:234: package_install] Error 255
make[1]: *** [Makefile:171: _call_manifest] Error 2
make: *** [Makefile:349: manifest] Error 2
```

cmhrky, 2025-02-10:
Model  Xiaomi Mi Router 4A Gigabit Edition
Platform  ramips/mt7621
Version  24.10.0 (r28427-6df0e3d02a)

  
```
Error: Impossible package selection
STDERR
Generate local signing keys...
Generate local certificate...
Package list missing or not up-to-date, generating it.

Building package index...
Downloading http://downloads.openwrt.org/releases/24.10.0/targets/ramips/mt7621/packages/Packages.gz
Updated list of available packages in /builder/build_dir/target-mipsel_24kc_musl/root-ramips/../../../../builder/dl/openwrt_core
Downloading http://downloads.openwrt.org/releases/24.10.0/targets/ramips/mt7621/packages/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/mipsel_24kc/base/Packages.gz
Updated list of available packages in /builder/build_dir/target-mipsel_24kc_musl/root-ramips/../../../../builder/dl/openwrt_base
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/mipsel_24kc/base/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/24.10.0/targets/ramips/mt7621/kmods/6.6.73-1-3abe85def815b59c6c75ac1f92135cb6/Packages.gz
Updated list of available packages in /builder/build_dir/target-mipsel_24kc_musl/root-ramips/../../../../builder/dl/openwrt_kmods
Downloading http://downloads.openwrt.org/releases/24.10.0/targets/ramips/mt7621/kmods/6.6.73-1-3abe85def815b59c6c75ac1f92135cb6/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/mipsel_24kc/luci/Packages.gz
Updated list of available packages in /builder/build_dir/target-mipsel_24kc_musl/root-ramips/../../../../builder/dl/openwrt_luci
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/mipsel_24kc/luci/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/mipsel_24kc/packages/Packages.gz
Updated list of available packages in /builder/build_dir/target-mipsel_24kc_musl/root-ramips/../../../../builder/dl/openwrt_packages
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/mipsel_24kc/packages/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/mipsel_24kc/routing/Packages.gz
Updated list of available packages in /builder/build_dir/target-mipsel_24kc_musl/root-ramips/../../../../builder/dl/openwrt_routing
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/mipsel_24kc/routing/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/mipsel_24kc/telephony/Packages.gz
Updated list of available packages in /builder/build_dir/target-mipsel_24kc_musl/root-ramips/../../../../builder/dl/openwrt_telephony
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/mipsel_24kc/telephony/Packages.sig
Signature check passed.
Downloading file:packages/Packages
Updated list of available packages in /builder/build_dir/target-mipsel_24kc_musl/root-ramips/../../../../builder/dl/imagebuilder
Downloading file:packages/Packages.sig
Signature check passed.
Collected errors:
 * opkg_install_pkg: Checksum or size mismatch for package umdns. Either the opkg or the package index are corrupt. Try 'opkg update'.
 * opkg_install_cmd: Cannot install package dawn.
make[2]: *** [Makefile:234: package_install] Error 255
make[1]: *** [Makefile:171: _call_manifest] Error 2
make: *** [Makefile:349: manifest] Error 2
```

ly4096x, 2025-02-13:
+1 here. Trying to build 24.10.0.

Target:

```
    "profile": "bananapi,bpi-r4",
    "target": "mediatek/filogic",
    "diff_packages": true,
    "filesystem": "squashfs"
```

Error:

```
Generate local signing keys...
Generate local certificate...
Package list missing or not up-to-date, generating it.

Building package index...
Downloading http://downloads.openwrt.org/releases/24.10.0/targets/mediatek/filogic/packages/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../builder/dl/openwrt_core
Downloading http://downloads.openwrt.org/releases/24.10.0/targets/mediatek/filogic/packages/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/aarch64_cortex-a53/base/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../builder/dl/openwrt_base
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/aarch64_cortex-a53/base/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/24.10.0/targets/mediatek/filogic/kmods/6.6.73-1-d649d775435da5a8637f7a955a80d331/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../builder/dl/openwrt_kmods
Downloading http://downloads.openwrt.org/releases/24.10.0/targets/mediatek/filogic/kmods/6.6.73-1-d649d775435da5a8637f7a955a80d331/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/aarch64_cortex-a53/luci/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../builder/dl/openwrt_luci
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/aarch64_cortex-a53/luci/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/aarch64_cortex-a53/packages/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../builder/dl/openwrt_packages
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/aarch64_cortex-a53/packages/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/aarch64_cortex-a53/routing/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../builder/dl/openwrt_routing
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/aarch64_cortex-a53/routing/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/aarch64_cortex-a53/telephony/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../builder/dl/openwrt_telephony
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/aarch64_cortex-a53/telephony/Packages.sig
Signature check passed.
Downloading file:packages/Packages
Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../builder/dl/imagebuilder
Downloading file:packages/Packages.sig
Signature check passed.
Collected errors:
 * opkg_install_pkg: Checksum or size mismatch for package wpad-openssl. Either the opkg or the package index are corrupt. Try 'opkg update'.
 * opkg_install_cmd: Cannot install package wpad-openssl.
make[2]: *** [Makefile:234: package_install] Error 255
make[1]: *** [Makefile:171: _call_manifest] Error 2
make: *** [Makefile:349: manifest] Error 2
```

If I replace `wpad-openssl` with `wpad` it will work, but anything else like `wpad-wolfssl` won't work.

kristof-dragon, 2025-02-13:
I noticed the same thing while trying to install packages from the cli using opkg on a freshly downloaded stock image for Radxa 3B board.
The same errors appear in the firmware selector as well.

root@OpenWrt:~# opkg update
Downloading https://downloads.openwrt.org/releases/24.10.0/targets/rockchip/armv8/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_core
Downloading https://downloads.openwrt.org/releases/24.10.0/targets/rockchip/armv8/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/24.10.0/packages/aarch64_generic/base/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_base
Downloading https://downloads.openwrt.org/releases/24.10.0/packages/aarch64_generic/base/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/24.10.0/targets/rockchip/armv8/kmods/6.6.73-1-f35e93bc2c89b98d107e57cdea041972/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_kmods
Downloading https://downloads.openwrt.org/releases/24.10.0/targets/rockchip/armv8/kmods/6.6.73-1-f35e93bc2c89b98d107e57cdea041972/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/24.10.0/packages/aarch64_generic/luci/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_luci
Downloading https://downloads.openwrt.org/releases/24.10.0/packages/aarch64_generic/luci/Packages.sig
Signature check failed.
Remove wrong Signature file.
Downloading https://downloads.openwrt.org/releases/24.10.0/packages/aarch64_generic/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_packages
Downloading https://downloads.openwrt.org/releases/24.10.0/packages/aarch64_generic/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/24.10.0/packages/aarch64_generic/routing/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_routing
Downloading https://downloads.openwrt.org/releases/24.10.0/packages/aarch64_generic/routing/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/24.10.0/packages/aarch64_generic/telephony/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_telephony
Downloading https://downloads.openwrt.org/releases/24.10.0/packages/aarch64_generic/telephony/Packages.sig
Signature check failed.
Remove wrong Signature file.

and this is in formware-selector:

Generate local signing keys...
Generate local certificate...
Package list missing or not up-to-date, generating it.

Building package index...
Downloading http://downloads.openwrt.org/releases/24.10.0/targets/rockchip/armv8/packages/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_generic_musl/root-rockchip/../../../../builder/dl/openwrt_core
Downloading http://downloads.openwrt.org/releases/24.10.0/targets/rockchip/armv8/packages/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/aarch64_generic/base/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_generic_musl/root-rockchip/../../../../builder/dl/openwrt_base
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/aarch64_generic/base/Packages.sig
Signature check failed.
Remove wrong Signature file.
Downloading http://downloads.openwrt.org/releases/24.10.0/targets/rockchip/armv8/kmods/6.6.73-1-f35e93bc2c89b98d107e57cdea041972/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_generic_musl/root-rockchip/../../../../builder/dl/openwrt_kmods
Downloading http://downloads.openwrt.org/releases/24.10.0/targets/rockchip/armv8/kmods/6.6.73-1-f35e93bc2c89b98d107e57cdea041972/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/aarch64_generic/luci/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_generic_musl/root-rockchip/../../../../builder/dl/openwrt_luci
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/aarch64_generic/luci/Packages.sig
Signature check failed.
Remove wrong Signature file.
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/aarch64_generic/packages/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_generic_musl/root-rockchip/../../../../builder/dl/openwrt_packages
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/aarch64_generic/packages/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/aarch64_generic/routing/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_generic_musl/root-rockchip/../../../../builder/dl/openwrt_routing
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/aarch64_generic/routing/Packages.sig
Signature check failed.
Remove wrong Signature file.
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/aarch64_generic/telephony/Packages.gz
[... 79 lines cut ...]
 * opkg_install_cmd: Cannot install package dnsmasq.
 * opkg_install_cmd: Cannot install package e2fsprogs.
 * opkg_install_cmd: Cannot install package firewall4.
 * opkg_install_cmd: Cannot install package libustream-mbedtls.
 * opkg_install_cmd: Cannot install package logd.
 * opkg_install_cmd: Cannot install package mkf2fs.
[... cut ...]
 * opkg_install_cmd: Cannot install package ppp.
 * opkg_install_cmd: Cannot install package ppp-mod-pppoe.
 * opkg_install_cmd: Cannot install package procd-ujail.
 * opkg_install_cmd: Cannot install package uci.
 * opkg_install_cmd: Cannot install package uclient-fetch.
 * opkg_install_cmd: Cannot install package urandom-seed.
 * opkg_install_cmd: Cannot install package urngd.
make[2]: *** [Makefile:234: package_install] Error 255
make[1]: *** [Makefile:171: _call_manifest] Error 2
make: *** [Makefile:349: manifest] Error 2

kristof-dragon, 2025-02-13:
This has already hapened a couple of days ago for me, and then it resolved on it's own.

kristof-dragon, 2025-02-13:
getting worst on RC7.  --->  opkg_install_cmd: Cannot install package luci.

Model: Radxa ROCK 3B
Platform: rockchip/armv8
Version: 24.10.0-rc7 (r28417-daef29c75d)

Error: Impossible package selection
STDERR

Generate local signing keys...
Generate local certificate...
Package list missing or not up-to-date, generating it.

Building package index...
Downloading http://downloads.openwrt.org/releases/24.10.0-rc7/targets/rockchip/armv8/packages/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_generic_musl/root-rockchip/../../../../builder/dl/openwrt_core
Downloading http://downloads.openwrt.org/releases/24.10.0-rc7/targets/rockchip/armv8/packages/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/24.10.0-rc7/packages/aarch64_generic/base/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_generic_musl/root-rockchip/../../../../builder/dl/openwrt_base
Downloading http://downloads.openwrt.org/releases/24.10.0-rc7/packages/aarch64_generic/base/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/24.10.0-rc7/targets/rockchip/armv8/kmods/6.6.73-1-f35e93bc2c89b98d107e57cdea041972/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_generic_musl/root-rockchip/../../../../builder/dl/openwrt_kmods
Downloading http://downloads.openwrt.org/releases/24.10.0-rc7/targets/rockchip/armv8/kmods/6.6.73-1-f35e93bc2c89b98d107e57cdea041972/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/24.10.0-rc7/packages/aarch64_generic/luci/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_generic_musl/root-rockchip/../../../../builder/dl/openwrt_luci
Downloading http://downloads.openwrt.org/releases/24.10.0-rc7/packages/aarch64_generic/luci/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/24.10.0-rc7/packages/aarch64_generic/packages/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_generic_musl/root-rockchip/../../../../builder/dl/openwrt_packages
Downloading http://downloads.openwrt.org/releases/24.10.0-rc7/packages/aarch64_generic/packages/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/24.10.0-rc7/packages/aarch64_generic/routing/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_generic_musl/root-rockchip/../../../../builder/dl/openwrt_routing
Downloading http://downloads.openwrt.org/releases/24.10.0-rc7/packages/aarch64_generic/routing/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/24.10.0-rc7/packages/aarch64_generic/telephony/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_generic_musl/root-rockchip/../../../../builder/dl/openwrt_telephony
Downloading http://downloads.openwrt.org/releases/24.10.0-rc7/packages/aarch64_generic/telephony/Packages.sig
Signature check passed.
Downloading file:packages/Packages
Updated list of available packages in /builder/build_dir/target-aarch64_generic_musl/root-rockchip/../../../../builder/dl/imagebuilder
Downloading file:packages/Packages.sig
Signature check passed.
Collected errors:
 * opkg_install_pkg: Checksum or size mismatch for package ucode-mod-html. Either the opkg or the package index are corrupt. Try 'opkg update'.
 * opkg_install_cmd: Cannot install package luci.
make[2]: *** [Makefile:234: package_install] Error 255
make[1]: *** [Makefile:171: _call_manifest] Error 2
make: *** [Makefile:349: manifest] Error 2

kristof-dragon, 2025-02-13:
and now it is working on the system that I was working on. no restart, not change, opkg started working again.

ideal-wheelMN, 2025-03-08:
Model: ASUS RT-AC58U 

Firmware: OpenWrt 24.10.0 r28427-6df0e3d02a / LuCI openwrt-24.10 branch 25.014.55016~7046a1c

Running into similar issue as above with failing signature check for Packages file on a fresh install of OpenWRT
for this target.

The checksums for the extracted  **Packages.gz**  and the  **Packages**  file differ. Using usign manually to check the
signature works for the plain Packages file . Diff output for the files is attached. It looks like some of the packages have different checksums in each file.

https://downloads.openwrt.org/releases/24.10.0/packages/arm_cortex-a7_neon-vfpv4/base/Packages.gz

https://downloads.openwrt.org/releases/24.10.0/packages/arm_cortex-a7_neon-vfpv4/base/Packages

[packages_diff_context.txt](https://github.com/user-attachments/files/19144646/packages_diff_context.txt)

[packages_diff.txt](https://github.com/user-attachments/files/19144647/packages_diff.txt)

jclab-joseph, 2025-03-15:
Same issue on BPI-R4.

```
# cat /etc/os-release
NAME="OpenWrt"
VERSION="24.10.0"
ID="openwrt"
ID_LIKE="lede openwrt"
PRETTY_NAME="OpenWrt 24.10.0"
VERSION_ID="24.10.0"
HOME_URL="https://openwrt.org/"
BUG_URL="https://bugs.openwrt.org/"
SUPPORT_URL="https://forum.openwrt.org/"
BUILD_ID="r28427-6df0e3d02a"
OPENWRT_BOARD="mediatek/filogic"
OPENWRT_ARCH="aarch64_cortex-a53"
OPENWRT_TAINTS=""
OPENWRT_DEVICE_MANUFACTURER="OpenWrt"
OPENWRT_DEVICE_MANUFACTURER_URL="https://openwrt.org/"
OPENWRT_DEVICE_PRODUCT="Generic"
OPENWRT_DEVICE_REVISION="v0"
OPENWRT_RELEASE="OpenWrt 24.10.0 r28427-6df0e3d02a"
OPENWRT_BUILD_DATE="1738624177"
root@OpenWrt:/etc/opkg/keys# ls
d310c6f2833e97f7
root@OpenWrt:/etc/opkg/keys# opkg update
Downloading https://downloads.openwrt.org/releases/24.10.0/targets/mediatek/filogic/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_core
Downloading https://downloads.openwrt.org/releases/24.10.0/targets/mediatek/filogic/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/24.10.0/packages/aarch64_cortex-a53/base/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_base
Downloading https://downloads.openwrt.org/releases/24.10.0/packages/aarch64_cortex-a53/base/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/24.10.0/targets/mediatek/filogic/kmods/6.6.73-1-d649d775435da5a8637f7a955a80d331/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_kmods
Downloading https://downloads.openwrt.org/releases/24.10.0/targets/mediatek/filogic/kmods/6.6.73-1-d649d775435da5a8637f7a955a80d331/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/24.10.0/packages/aarch64_cortex-a53/luci/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_luci
Downloading https://downloads.openwrt.org/releases/24.10.0/packages/aarch64_cortex-a53/luci/Packages.sig
Signature check failed.
Remove wrong Signature file.
Downloading https://downloads.openwrt.org/releases/24.10.0/packages/aarch64_cortex-a53/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_packages
Downloading https://downloads.openwrt.org/releases/24.10.0/packages/aarch64_cortex-a53/packages/Packages.sig
Signature check failed.
Remove wrong Signature file.
Downloading https://downloads.openwrt.org/releases/24.10.0/packages/aarch64_cortex-a53/routing/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_routing
Downloading https://downloads.openwrt.org/releases/24.10.0/packages/aarch64_cortex-a53/routing/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/24.10.0/packages/aarch64_cortex-a53/telephony/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_telephony
Downloading https://downloads.openwrt.org/releases/24.10.0/packages/aarch64_cortex-a53/telephony/Packages.sig
Signature check failed.
Remove wrong Signature file.
```

kristof-dragon, 2025-03-18:
I have encountered the same issue again, but this time it was trying to install a few packages via CLI. Around 5-6pm UK time there were problems with the signatures of packages.... digging into the issue, it seems that this is related to mirrors not syncing and causing this issue with the repository in general.. I did not investigate any further than that, and it resolved on it's own after a couple of hours. (Exactly the same way as it did before)

pawansingh126, 2025-05-01:
Was facing the same issue and cleaning up /tmp/opkg-lists fixed it for me!