openwrt/packages #25931: wifischedule - problem with 24.10

p0k33m0n, 2025-02-10:
After updating to firmware 24.10 wifischedule repeatedly starts with WiFi disabled. This only occurs on cold boot, not reboot.

```
Model | Netgear WNDR4300
Architecture | Atheros AR9344 rev 2
Target Platform | ath79/nand
Firmware Version | OpenWrt 24.10.0 r28427-6df0e3d02a / LuCI openwrt-24.10 branch 25.037.68331~42f464c
Kernel Version | 6.6.73
Local Time | 2025-02-10 09:55:34
Uptime | 0h 6m 41s
Load Average | 0.05, 0.52, 0.35
```

p0k33m0n, 2025-02-19:
> I can confirm that on latest snapshot wifischedule doesn't work at all... script doesn't start... so something is wrong...

The script starts, only immediately turns off WiFi. I did it manually through Cron eg:

Scheduled Tasks ->
`00 22 * * Mon,Tue,Wed,Thu,Sun wifi down`

Mile-Lile, 2025-02-26:
> I can confirm that on latest snapshot wifischedule doesn't work at all... script doesn't start... so something is wrong...

problem was I am using luci-app-https-dns-proxy and in ntp section I had hostnames not IPs (default openwrt setting), so hostnames of ntp servers didn't get resolved and my router clock had wrong time.... so script didn't started in correct time of the day...

@p0k33m0n 
your problem is probably is probably related to clock and time is not corret so when you turn on router, script starts immediatelly and shut down wifi...

p0k33m0n, 2025-02-26:
> your problem is probably is probably related to clock and time is not corret so when you turn on router, script starts immediatelly and shut down wifi...

My time is set correctly, I assure you, otherwise Cron would not work correctly, but it works. But wifischedule is no matter for me, my solutions work correctly. By the way https-dns-proxy-this is an old code, with a lot of error. More problems than benefit. After switching to 24.10, it completely spilled - I know, because I used myself. I switched to SmartDNS, much newer code, greater possibilities.

ratzrattillo, 2025-04-20:
See also: https://github.com/openwrt/packages/issues/26355