openwrt/packages #25933: acme dns_duckdb Error extracting the domain

artemklevtsov, 2025-02-10:
Hi,

Faced with issue.

To reproduce:

```bash
# export DuckDNS_Token="184b69af-5862-410f-be2b-883f0743b8f9"
# /usr/lib/acme/client/acme.sh --test --renew --home /etc/acme --domain test.duckdns.org
[Mon Feb 10 19:03:02 +07 2025] Using ACME_DIRECTORY: https://acme-staging-v02.api.letsencrypt.org/directory
[Mon Feb 10 19:03:02 +07 2025] Renew: 'test.duckdns.org'
[Mon Feb 10 19:03:02 +07 2025] Renew to Le_API=https://acme-v02.api.letsencrypt.org/directory
[Mon Feb 10 19:03:03 +07 2025] Using CA: https://acme-v02.api.letsencrypt.org/directory
[Mon Feb 10 19:03:04 +07 2025] Single domain='test.duckdns.org'
[Mon Feb 10 19:03:04 +07 2025] Getting domain auth token for each domain
[Mon Feb 10 19:03:07 +07 2025] Getting webroot for domain='test.duckdns.org'
[Mon Feb 10 19:03:07 +07 2025] Adding txt value: EQbV2_zKJf9C0oiqnUcdSqUOLLhjjoRRzs0ygLaVPFQ for domain:  _acme-challenge.test.duckdns.org
[Mon Feb 10 19:03:07 +07 2025] Error extracting the domain.
[Mon Feb 10 19:03:07 +07 2025] Error add txt for domain:_acme-challenge.test.duckdns.org
[Mon Feb 10 19:03:07 +07 2025] Please check log file for more details: /tmp/acme.log
```

Error raised from this: `/usr/lib/acme/client/dnsapi/dns_duckdns.sh` 
dns_duckdns_add => _duckdns_get_domain => _egrep_o

`_egrep_o` seems uses `sed` which works incorrect with this regular expression. As result `_duckdns_domain` is always empty.

Source code links:
- https://github.com/acmesh-official/acme.sh/blob/3.0.7/dnsapi/dns_duckdns.sh#L102
- https://github.com/acmesh-official/acme.sh/blob/3.0.7/acme.sh#L929
- https://github.com/acmesh-official/acme.sh/blob/3.0.7/acme.sh#L936

OS info:

```bash
# cat /etc/os-release 
NAME="OpenWrt"
VERSION="24.10.0"
ID="openwrt"
ID_LIKE="lede openwrt"
PRETTY_NAME="OpenWrt 24.10.0"
VERSION_ID="24.10.0"
HOME_URL="https://openwrt.org/"
BUG_URL="https://bugs.openwrt.org/"
SUPPORT_URL="https://forum.openwrt.org/"
BUILD_ID="r28427-6df0e3d02a"
OPENWRT_BOARD="mediatek/filogic"
OPENWRT_ARCH="aarch64_cortex-a53"
OPENWRT_TAINTS=""
OPENWRT_DEVICE_MANUFACTURER="OpenWrt"
OPENWRT_DEVICE_MANUFACTURER_URL="https://openwrt.org/"
OPENWRT_DEVICE_PRODUCT="Generic"
OPENWRT_DEVICE_REVISION="v0"
OPENWRT_RELEASE="OpenWrt 24.10.0 r28427-6df0e3d02a"
OPENWRT_BUILD_DATE="1738624177"
```

Also tried with this code:

```bash
if [ "$(echo abc | egrep -o b 2>/dev/null)" = "b" ]; then
  __USE_EGREP=1
else
  __USE_EGREP=""
fi

_egrep_o() {
  if [ "$__USE_EGREP" ]; then
    egrep -o -- "$1" 2>/dev/null
  else
    sed -n 's/.*\('"$1"'\).*/\1/p'
  fi
}

_lower_case() {
  # shellcheck disable=SC2018,SC2019
  tr '[A-Z]' '[a-z]'
}

fulldomain="_acme-challenge.test.duckdns.org"

_duckdns_domain="$(printf "%s" "$fulldomain" | _lower_case | _egrep_o '^(_acme-challenge\.)?([a-z0-9-]+\.)+duckdns\.org' | sed -n 's/^\([^.]\{1,\}\.\)*\([a-z0-9-]\{1,\}\)\.duckdns\.org$/\2/p;')"

echo "$__USE_EGREP"
echo "$fulldomain"
echo "$_duckdns_domain"
```

`__USE_EGREP` and `_duckdns_domain` is empty

artemklevtsov, 2025-02-10:
After `alias egrep='grep -E'` `_egrep_o` works correctly.

artemklevtsov, 2025-02-15:
> can you update to the latest acme v3.1.0? Maybe something was changed in the egrep [#24262](https://github.com/openwrt/packages/issues/24262) may be related

How can I upgrade the acme-acmesh-dnsapi? Should I build it from the source?

artemklevtsov, 2025-02-15:
> Anyway, we can check code in the v3.1.0 was added redirecting errors to /dev/null so they won't be shown `egrep -o -- "$1" 2>/dev/null`
> 
> https://github.com/acmesh-official/acme.sh/blob/3.1.0/acme.sh#L939
> 
> I wondered why the sed was used. The `__USE_EGREP` var should be 1.

egrep not available on the openwrt if `grep` package installed.

```bash
# egrep 
BusyBox v1.36.1 (2025-02-03 00:55:01 UTC) multi-call binary.

No help available
# opkg install grep
Installing grep (3.11-r1) to root...
Downloading https://downloads.openwrt.org/releases/24.10.0/packages/mips_24kc/packages/grep_3.11-r1_mips_24kc.ipk
Installing libpcre2 (10.42-r1) to root...
Downloading https://downloads.openwrt.org/releases/24.10.0/packages/mips_24kc/base/libpcre2_10.42-r1_mips_24kc.ipk
Configuring libpcre2.
Configuring grep.
# egrep
-ash: egrep: not found
```

artemklevtsov, 2025-02-15:
Remove grep package and acme worked successfully.

It is strange behavior with grep package:

```bash
# egrep
BusyBox v1.36.1 (2025-02-04 19:21:11 UTC) multi-call binary.

No help available
# opkg install grep
Installing grep (3.11-r1) to root...
Downloading https://downloads.openwrt.org/releases/24.10.0/packages/aarch64_cortex-a53/packages/grep_3.11-r1_aarch64_cortex-a53.ipk
Configuring grep.
# egrep
-ash: egrep: not found
# which egrep
/bin/egrep
# ls -l /bin/egrep
lrwxrwxrwx    1 root     root            22 Feb 15 13:43 /bin/egrep -> /usr/libexec/egrep-gnu
# /usr/libexec/egrep-gnu
-ash: /usr/libexec/egrep-gnu: not found
root@UNIKUM-ROUTER:~# cat /usr/libexec/egrep-gnu
#!/bin/bash
cmd=${0##*/}
echo "$cmd: warning: $cmd is obsolescent; using grep -E" >&2
exec grep -E "$@"
# sh /usr/libexec/egrep-gnu
egrep-gnu: warning: egrep-gnu is obsolescent; using grep -E
Usage: grep [OPTION]... PATTERNS [FILE]...
Try 'grep --help' for more information.
```

artemklevtsov, 2025-02-15:
Replace `#!/bin/bash` with `#!/bin/ash`
```bash
# cp /usr/libexec/egrep-gnu /tmp/egrep1
# sed 's|/bin/bash|/bin/ash|' /tmp/egrep1 > /tmp/egrep2
# chmod +x /tmp/egrep2
# /tmp/egrep1
-ash: /tmp/egrep1: not found
# /tmp/egrep2
egrep2: warning: egrep2 is obsolescent; using grep -E
Usage: grep [OPTION]... PATTERNS [FILE]...
Try 'grep --help' for more information.
```

Wiedzmin89, 2025-05-06:
`opk install bash` - worked for me