openwrt/packages #25934: OpenWrt 24.10.0 dropbear refusing pubkey authentication.

Spamfast, 2025-02-10:
I've just upgraded my GL-MT300N-V2 (GL.iNet Mango) from 23.05.5 to 24.10.0. Everything went fine except that now it is refusing ED25519 and other types of public key authentication.

I've tried reinstalling the public keys via LuCI to no avail. Here's the listing for dropbear on the failing installation.

```
root@mango:/etc/dropbear# ls -al /etc/dropbear
drwxr-xr-x    1 root     root             0 Jan  1  1970 .
drwxr-xr-x    1 root     root             0 Feb  7 11:43 ..
-rw-------    1 root     root           485 Feb  7 11:58 authorized_keys
-rw-------    1 root     root            83 Oct  7  2022 dropbear_ed25519_host_key
-rw-------    1 root     root           805 Oct  7  2022 dropbear_rsa_host_key
```
```
root@mango:/etc/dropbear# ps w | grep dropbear
 1322 root      1288 S    /usr/sbin/dropbear -F -P /var/run/dropbear.1.pid -p 22 -K 300 -T 3 -W 262144
 1323 root      1288 S    /usr/sbin/dropbear -F -P /var/run/dropbear.2.pid -p 7003 -s -g -K 300 -T 3 -W 262144
 8950 root      1308 S    /usr/sbin/dropbear -F -P /var/run/dropbear.1.pid -p 22 -K 300 -T 3 -W 262144 -2 9
 9024 root      1396 S    grep dropbear
```
```
root@mango:/etc/dropbear# cat /etc/dropbear/authorized_keys
ssh-ed25519 AAAAC3NzaC1lZDI1NTE5AAAAIP8Qw2XLi3+KxWI5s1TQtzZfbnvzPSz+TGpF5JhQpbPf samedge@gwaihir
ssh-ed25519 AAAAC3NzaC1lZDI1NTE5AAAAIKD1yVavXULlS9w69VtVwKiXnDRlzrpWXQ+7UI6BCZ/5 samedge@ceres
ssh-ed25519 AAAAC3NzaC1lZDI1NTE5AAAAIItom1vpCJjwJtuJGnH9JyGuBgYP69XNUEZW96qHL7jV root@lenny.juice
ssh-ed25519 AAAAC3NzaC1lZDI1NTE5AAAAIES36Z4WbpNoGSaIwxC3JM9byQF2wFUzjL/DI9yWffo5 samedge@ceres
ssh-ed25519 AAAAC3NzaC1lZDI1NTE5AAAAIMwd4i7ZWB/+n8LIBFppzpDOa0gyWewdRgz30aAYPG5H samedge@hephaestus
```

On a different device, still running 23.05.5 and accepting public key auth, I have.
```
root@scylla:~# ls -la /etc/dropbear/
drwxr-xr-x    1 root     root             0 Jan  1  1970 .
drwxr-xr-x    1 root     root             0 Feb  6 18:59 ..
-rw-------    1 root     root          1055 Aug  1  2023 authorized_keys
-rw-------    1 root     root            83 Sep 25  2021 dropbear_ed25519_host_key
-rw-r--r--    1 root     root             0 Apr  9  2022 dropbear_ed25519_host_key-opkg
-rw-------    1 root     root           805 Sep 25  2021 dropbear_rsa_host_key
-rw-r--r--    1 root     root             0 Apr  9  2022 dropbear_rsa_host_key-opkg
```
```
root@scylla:~# ps w | grep dropbear
 1272 root      1268 S    /usr/sbin/dropbear -F -P /var/run/dropbear.1.pid -p 22 -K 300 -T 3 -W 262144
 1273 root      1268 S    /usr/sbin/dropbear -F -P /var/run/dropbear.2.pid -s -g -p 7002 -K 300 -T 3 -W 262144
10067 root      1288 S    /usr/sbin/dropbear -F -P /var/run/dropbear.1.pid -p 22 -K 300 -T 3 -W 262144 -2 9
10133 root      1372 S    grep dropbear
```
```
root@scylla:~# cat /etc/dropbear/authorized_keys
ecdsa-sha2-nistp256 AAAAE2VjZHNhLXNoYTItbmlzdHAyNTYAAAAIbmlzdHAyNTYAAABBBNFmfw80VbJGq1SSRp8dFAkiAeI685p1pnzW9/7ate119Znd/lW/LJJCsCPssdr8MYCNvMpmI0+Y5J+Ox4sDXeM= SamEdge@Ceres
ssh-ed25519 AAAAC3NzaC1lZDI1NTE5AAAAIPfOt+nHcSlzK/FCTZZzEbIgF7ONPHYG60HWd9r+oadW root@bolg5
ssh-rsa AAAAB3NzaC1yc2EAAAADAQABAAABAQDyg2wvRO0sOYNWMUTx2/eWAadfHsCnfAWHHC9vzyedrV5c2w52I+T6NT3LB5hBVCj3P9vCBnV/F8QiOQLGjjy2EH92tqB7gK6l6JrEsXNY1x3ph9saWpe3+/Z4/1NPaHfEPapG55WglAvqF+yA6yFfwfMes027rDYp9Gl1QXk4RifSGqUgDzx2ptknBZDfE4m7WSoobYtzLFTOfkdT9RpvWWwJgN/2E4lJMYDeTqNtq9QotOq5aSOFkfYdP/lQd+34wAzv3SAJhHleOP9/YNwU3UsTb4/ZMiFtCj6AdHErMChUAQ5N8k5sFbhmq+pJiYyCp3EG7uePHXQTwZ+H5CeB samedge@bapple
ssh-ed25519 AAAAC3NzaC1lZDI1NTE5AAAAIMwd4i7ZWB/+n8LIBFppzpDOa0gyWewdRgz30aAYPG5H samedge@hephaestus
ssh-ed25519 AAAAC3NzaC1lZDI1NTE5AAAAIP8Qw2XLi3+KxWI5s1TQtzZfbnvzPSz+TGpF5JhQpbPf samedge@gwaihir
ssh-ed25519 AAAAC3NzaC1lZDI1NTE5AAAAIPJF1TULDBgsfYas13YdPN7/5A/SIE3Po4ltMrF8SAXI samedge@penguin
ssh-ed25519 AAAAC3NzaC1lZDI1NTE5AAAAIItom1vpCJjwJtuJGnH9JyGuBgYP69XNUEZW96qHL7jV root@lenny.juice
```

The authorized key for `gwaihir` for example is identical on both but not being accepted by 24.10.0 whereas no problem with 23.05.5.

Spamfast, 2025-02-10:
The configuration on both defices is as follows.
```
root@scylla:~# cat /etc/config/dropbear

config dropbear
        option PasswordAuth 'on'
        option Port '22'

config dropbear
        option Port '7003'
        option PasswordAuth 'off'
        option RootPasswordAuth 'off'
```

Logging in via either the local instance on port 22 or the world-access instance on 7003 doesn't allow pubkey and even more worryingly DOES allow password authentication.

Spamfast, 2025-02-10:
Confirmed - if I upload an RSA public key via LuCI, pubkey works again.

brada4, 2025-02-11:
It is intentional
https://github.com/openwrt/openwrt/blob/04570f5ee2dfe6609c22d2d21678b14cd3314309/package/network/services/dropbear/Config.in#L52

Spamfast, 2025-02-11:
> It is intentional https://github.com/openwrt/openwrt/blob/04570f5ee2dfe6609c22d2d21678b14cd3314309/package/network/services/dropbear/Config.in#L52

Ah. Thanks for the info. Could be a bit of a problem if installing remotely in the event that, as I do, password auth is disabled except for local access.

Since this conditional Kconfig seems to to have been introducted a while back, presumably SMALL_FLASH is set for the Mango in 24.10 but wasn't in 23.05? The implications of that change might have been advertised a bit better in the release notes - unless it was and I missed it?

Spamfast, 2025-02-11:
Anyway, as it's a feature not a bug, I'll close this.

closed 2025-02-11 by Spamfast