openwrt/packages #25959: OpenWRT 24 is missing files for Hostapd/WPA_SUPPLICANT

Gamerayers, 2025-02-12:
When installing Hostapd-common and hostapd, files are not available which should be in Openwrt. This was identified in a snapshot build prior to release of the official and again after.

This is the error received.
wpa_supplicant[3478]: Error loading ucode: Syntax error: Unable to resolve path for module 'common' In line 3, byte 116:   `import { wdev_create, wdev_set_mesh_params, wdev_remove, is_equal, wdev_set_up, vlist_new, phy_open } from "common";`   Near here ---------------------------------------------------------------------------------------------------------^

Here's the reddit discussion: https://www.reddit.com/r/openwrt/comments/1ibf1tz/moving_to_24_have_an_issue_with_wpa_supplicant/

It appears that /usr/share/hostap/common.uc does not exist. Once the file is transfered from a Openwrt 23 build, it then still needs hostapd.uc. After putting those in, software on the router can no longer be updated since the apk installer says the database is now corrupt.

matjon, 2025-02-21:
Hello,

I also had a similar problem after upgrading to OpenWRT 24.10 on my Asus RT-AX53U. Ultimately, I have determined that I had WiFi drivers missing, probably because of a mistake during compilation (I use a self-compiled OpenWRT image with an unrelated patch). I fixed this by executing:

    opkg install kmod-mt7915-firmware kmod-usb3 kmod-usb-ledtrig-usbport

(your drivers will probably be different, IIRC this command installed many kernel modules in addition to the WiFi firmware).

So, I would like to ask: what is your device model? Did you compile your firmware by hand?

matjon, 2025-02-21:
See also: https://github.com/openwrt/openwrt/issues/14445#issuecomment-2316386219

> Wifi support is absent from x86 images, probably somebody should write a guide how to "install wifi" like wpad and identify driver package(s) from lspci/lsusb.