openwrt/packages #25963: 24.10.0 unbound issues

K2IE, 2025-02-13:
There are reports of unbound not appearing in the Luci Startup listing in the Forums in this thread:

https://forum.openwrt.org/t/24-10-0-unbound-issues-luci-and-ipv4/224578/1

A possible cause for unbound not appearing in the Startup listing page relates to ubus and the following command:

ubus call rc list

shows that unbound is not in the returned list.

In my installation, which was an upgrade from 23.05.05, the openwrt host could resolve both IPv4 and IPv6 addresses but IPv4 clients could not resolve anything via unbound.  An analysis with tcpdump showed that after some seconds, openwrt would respond with an NXDOMAIN, even when the address was in the unbound cache already.

I have reverted to 23.05.05 but please let me know if any additional data is required.

dave14305, 2025-02-19:
> So probably a bug in ubus.

This is a change from procd by @Ansuel 

https://github.com/openwrt/rpcd/commit/4de3f02e3124994ad30600a69711150f1e9c9956

Ansuel, 2025-02-19:
@friendly-bits yes it's probably the 10 line limit as documented in the commit... I'm suprised to check the makefile of unbound tho...

friendly-bits, 2025-02-19:
@Ansuel the 10-line limit is not very intuitive and probably not documented. Do you think it could be removed? (this issue affects adblock-lean which is why I care)

Ansuel, 2025-02-19:
problem is that for massive init script it cause very long time for parsing...

friendly-bits, 2025-02-19:
> long time for parsing

Maybe there is a way to leave the limit in place (maybe slightly increase it) but have it ignore empty lines and lines which only contain comments without significantly increasing the parsing time?

Ansuel, 2025-02-19:
yes that was my idea... skip empty line and comments

friendly-bits, 2025-02-19:
This limitation should also be mentioned in the developer's guide. Otherwise it is basically impossible to guess.

devianceluka, 2025-03-02:
How to fix ipv4 queries to unbound in 24.10?

devianceluka, 2025-03-03:
I've been testing some more with a x86 virtualized **barebone stock** openwrt 23.05 and 24.10, with **disabled ipv6** on all interfaces through `/etc/config/network`, installed `unbound-daemon` and recursive unbound **works.**

When I port **the same `unbound.conf`** to a production Redmi AX6000 + v24.10, it doesnt work anymore.

For the moment I found a clue to make it work: `service dnsmasq stop`. I tried `service dnsmasq restart` but it doesnt make any difference.

The **only** real difference between Redmi and VM should be VLANs, since I have a few. It will take some time to virtualize VLANs and test more.

Anyone got a clue why a running `dnsmasq` (or VLANs) would break `unbound`?

devianceluka, 2025-03-04:
I took out from production this Redmi AX6000 and flashed it to **bone-stock** v24.10 with **only** ipv6 disabled (through `/etc/config/network`) and can confirm it still doesnt work. It only works when I stop `dnsmasq` and as soon as I start `dnsmasq` it doesnt work anymore. So the problem is not with VLANs, but with some difference between x86/aarch64 and `dnsmasq` itself.

BKPepe, 2025-09-24:
Closing, fix was backported. :) You can see the reference above.