openwrt/packages #25973: adblock-fast: block list processing drops subdomains starting with alpha characters

timfeierabend, 2025-02-13:
@stangri 

### Issue
adblock-fast is not loading all domains from HaGeZi's Threat Intelligence Feeds into dnsmasq.servers file. It seems to be dropping all domains that start with alpha characters and loading only numeric ones.

### To Re-Create
Configure adblock-fast with default settings and only 1 block list - HaGeZi's Threat Intelligence Feeds wildcard domain (https://raw.githubusercontent.com/hagezi/dns-blocklists/main/wildcard/tif-onlydomains.txt). Files uploaded for research.

[tif-onlydomains.txt](https://github.com/user-attachments/files/18785535/tif-onlydomains.txt)
[dnsmasq.servers.txt](https://github.com/user-attachments/files/18785550/dnsmasq.servers.txt)
[adblock-fast.txt](https://github.com/user-attachments/files/18785559/adblock-fast.txt)

### Expected Result
There are 753874 domains blocked.

### Actual result:
There are 34750 domains blocked.

### Context
I am using adblock-fast as a malware domain-blocker on my OpenWrt router with 1 block list configured - HaGeZi's Threat Intelligence Feeds Full. I don't mind seeing ads, but I do want to protect my network from malware. I've attached my adblock-fast UCI config file, the dnsmasq.servers file produced by the current adblock-fast version, and the current TIF wildcard domain file.

OpenWRT Version: 23.05.5
Hardware: FriendlyElec NanoPi R4S
Adblock-Fast version: 1.1.2-r20

stangri, 2025-02-13:
@timfeierabend that's a serious issue, give me some time to reproduce and come up with the fix.

timfeierabend, 2025-02-20:
Hi @stangri 

I appreciate the insight you've provided into the optimization process.

I did some further analysis on the attached tif-onlydomains.txt block list. There are no instances of both _n_ level and _n-1_ level domains in this file.

I ran `grep "shop" * > grep-shop.txt` in a directory containing only `dnsmasq.servers.txt` and `tif-onlydomains.txt` I attached above. You can see in this file there are many more entries with ".shop" in the tif-onlydomains.txt versus the dnsmasq.servers.txt.

[grep-shop.txt](https://github.com/user-attachments/files/18894516/grep-shop.txt)

stangri, 2025-02-21:
@timfeierabend I see what you mean, I'll test this in a vm to try to see what's going on.

timfeierabend, 2025-02-21:
@stangri 

I found the culprit. In `detect_file_type()`, your regex for identifying a **hosts** file isn't checking for white space after the IP address. Changing the grep expressions to `^0\.0\.0\.0\s` and `^127\.0\.0\.1\s` resolved the issue. There are some malicious domains in the threat intelligence feeds that start with "0.0.0.0" that cause this false positive.

stangri, 2025-02-21:
Wow, this is amazing! Fixed in adblock-fast 1.1.3-r5 available in both OPKG and APK of my repositories and this issue will be tagged when I create the PR for packages repo.

closed 2025-03-01 by stangri