openwrt/packages #25974: Wireguard AllowedIPs config option removed for first peer when adding more than one peer

hydro-b, 2025-02-14:
When wireguard is configured with one peer, all works as expected. However, when a second peer is added the `AllowedIPs` config option of the first peer is removed, rendering the peer unable to send/receive traffic.

The config is correctly displayed in the LuCI GUI. Also, `uci show network` shows the right configuration options (including the AllowedIPs option). However, a `wg showconf wireguard` shows that the `AllowedIPs` config option of the first peer is not present.

Restarting the wireguard interface, or the whole router for that matter, does not make a difference.

```
OpenWrt 24.10.0 r28427-6df0e3d02a / LuCI openwrt-24.10 branch 25.014.55016~7046a1c

kmod-wireguard - 6.6.73-r1
luci-proto-wireguard - 25.037.68331~42f464c
wireguard-tools - 1.0.20210914-r4
Linux OpenWrt 6.6.73 #0 SMP Mon Feb  3 23:09:37 2025 aarch64 GNU/Linux
```

egc112, 2025-03-12:
Maybe not applicable in your case but allowed IPs of peers cannot overlap as a rule of thumb, if they do it can result in only the last one is working.

You can always make new thread in the support forum posting your config if in doubt

hydro-b, 2025-03-12:
AllowedIPs is scoped per /32 IPv4. I have another wireguard server (non-OpenWRT) that is configured in a similar way that does allow that (/32 per peer). So I do think this is a bug.

egc112, 2025-03-12:
If you use only different Allowed IPs with /32 mask then that should not be a problem.
But why are you the only one having this while this is used by many users?
I have 9 peers which are working without a problem

hydro-b, 2025-03-12:
> If you use only different Allowed IPs with /32 mask then that should not be a problem. But why are you the only one having this while this is used by many users? I have 9 peers which are working without a problem

That's a good question. I don't know. Is the wireguard configuration stored in a file somewhere? I might post it here (I will redact private keys, etc.).