openwrt/packages #25976: adblock 4.2.4-r1: adding hostnames or domains with a custom tld (.somedomain) that is whitelisted and forwarded via VPN in dnsmasq to the adblock whitelist results in NX

MaxRower, 2025-02-14:
When pausing adblock, DNS resolution works fine.
All those whitelisted entries are listed in /tmp/dnsmasq.d/adb_list.overall as 
local=/somedomain/#
local=/somehost.somedomain/#
which seems to prevent forwarding the request to the other DNS via VPN.

dibdot, 2025-02-14:
I don't get it.
Please make a reproducible example, provide your full dnsmasq/adblock config ... and explain "which seems to prevent forwarding the request to the other DNS via VPN" - what other DNS!? 
Thanks!

MaxRower, 2025-02-14:
When I lookup those whitelistes hostnamed, it always returns NX, on all clients in the network and on the router itself.

MaxRower, 2025-02-14:
relevant entries in the config:

/etc/config/adblock

config adblock 'global'
	option adb_enabled '1'
	option adb_debug '0'
	option adb_forcedns '0'
	option adb_safesearch '0'
	option adb_dnsfilereset '0'
	option adb_mail '0'
	option adb_backup '1'
	option adb_maxqueue '4'
	list adb_sources 'adaway'
	list adb_sources 'adguard'
	list adb_sources 'disconnect'
	list adb_sources 'yoyo'
	option adb_dns 'dnsmasq'
	option adb_fetchutil 'uclient-fetch'
	option adb_report '1'
	option adb_repiface 'br-lan'
	option adb_trigger 'wan'
	option adb_represolve '1'

in /etc/adblock/adblock.whitelist
somedomain
somehost.somedomain

in /etc/config/dnsmasq
config dnsmasq
    list rebind_domain 'somedomain'
    list server '/somedomain/192.168.23.1'
    list server '/23.168.192.in-addr.arpa/192.168.23.1'

MaxRower, 2025-02-14:
I have multiple locations interconnected via wireguard  VPN, with unique local domain names, and DNS lookups via VPN are a requirement. To prevent accidental errors from externals block lists matching my internal domains, I added those to the whitelist.

dibdot, 2025-02-14:
> When I lookup those whitelisted hostnames, it always returns NX
Than your local DNS  can't resolve those (sub-)domains.  The blocklist syntax used for whitelisting should do the following (from the dnsmasq man page):

```
The special server address '#' means, "use the standard servers", so --server=/google.com/1.2.3.4 --server=/[www.google.com](http://www.google.com/)/# will send queries for google.com and its subdomains to 1.2.3.4, except [www.google.com](http://www.google.com/) (and its subdomains) which will be forwarded as usual.
[...]
Also permitted is a -S flag which gives a domain but no IP address; this tells dnsmasq that a domain is local and it may answer queries from /etc/hosts or DHCP but should never forward queries on that domain to any upstream servers. --local is a synonym for --server to make configuration files clearer in this case.
```
 Maybe there is a behavior change in a newer dnsmasq version/patch level in 24.x!?  I doubt that this is solvable in adblock (which only generates such lists and the list syntax hasn't changed for a long time). When I test the whitelisting with usual (sub-)domains in the wild it works for me.

dibdot, 2025-10-19:
nothing todo - close this outdated ticket.