openwrt/packages #25986: Openwrt 24.10 borgbackup is broken because of unsupported msgpack version

returntoreality, 2025-02-16:
Borgbackup has a version check for msgpack and will terminate if it is unsupported. The 24.10 repos have a version which is too new (1.0.7) for the borgbackup version in the repo (1.2.4) which only supports msgpack up to version 1.0.5. The newest 1.2 version of borgbackup (1.2.8) would increase the support to msgpack 1.0.8.
This is the error borgbackup reports:
```
You do not have a supported version of the msgpack python package installed. Terminating.
This should never happen as specific, supported versions are required by our setup.py.
Do not contact borgbackup support about this.
```
Probably updating the 24.10 repo to the latest borgbackup 1.2 version would be the best solution.

thatguyatgithub, 2025-02-18:
Just seconding this report as well. 

This was fixed last year at 1.4.0, here mentioned as well https://github.com/borgbackup/borg/issues/8144

returntoreality, 2025-02-22:
1.4.0 would also work, but would be a non-patch release update. I don't know how these are handled in openwrt stable releases.

projectgus, 2025-05-18:
This should be fixed now that #26497 has merged.

PR #26497 (2025-05-16): borgbackup: Update to 1.2.8 for openwrt-24.10
Maintainer: @julienmalik 
Compile tested: mediatek/filogic, TP-Link TL-XDR6088, openwrt-24.10 HEAD (52d4559fe)
Run tested: mediatek/filogic, TP-Link TL-XDR6088, OpenWRT v24.10.1 release. Ran `borg create` and `borg prune` and observed expected behaviour.

Description:

Update is necessary for compatibility with python3-msgpack v1.0.7. Borgbackup broke when this was updated in f8bfe730. Closes #25986.

Master branch borgbackup is already updated to newer v1.4.0 in c565bbad. *EDIT: Cherry-picking the 1.4.0 update here instead would require also updat [... cut]
closed 2025-05-20 by BKPepe