openwrt/packages #26011: proto-bonding LACP issue

trenchcoatjedi, 2025-02-18:
With the latest version of proto-bonding, I am seeing major issues with VLAN handling over interfaces with a bond as the underlying network device.  

Specifically, when using using a bond as the underlying device of the WAN, the downstream clients on VLAN7 are seeing speeds in the single digit megabit/second speed over a gigabit physical link. Here's the relevant config:

```
config device
	option type 'bonding'
	option name 'bond1'
	list ports 'eth0'
	list ports 'eth1'
	option bonding_policy '802.3ad'
	option all_slaves_active '0'
	option link_monitoring 'off'
	option min_links '0'
	option ad_actor_sys_prio '65535'
	option ad_select 'stable'
	option lacp_rate 'slow'
	option xmit_hash_policy 'layer2'
	option policy '802.3ad'
	option all_ports_active '1'

config interface 'wan'
	option device 'bond1'
	option proto 'dhcp'
	option hostname '*'
	option peerdns '0'

config device
	option type 'bridge'
	option name 'Switch'
	option bridge_empty '1'
	option mtu '1500'
	option ipv6 '0'
	option ageing_time '3'
	list ports 'eth2'
	list ports 'eth3'
	list ports 'eth4'
	list ports 'eth5'
	list ports 'eth8'

config bridge-vlan
	option device 'Switch'
	option vlan '7'
	list ports 'eth3:t'
	list ports 'eth4:t'
	list ports 'eth5:t'
	list ports 'eth8:t'

config interface 'Services'
	option proto 'static'
	option device 'Switch.7'
	option ipaddr '10.0.7.1'
	option netmask '255.255.255.0'
```

trenchcoatjedi, 2025-02-19:
> This seems more broken - https://openwrt.pkgs.org/23.05/openwrt-luci-x86_64/luci-proto-bonding_git-25.037.68351-2ca3f19_all.ipk.html this package is gone now? and there's no native support.

This package is depreciated in 24.10 since netifd was updated to include bonds as devices.

brada4, 2025-02-19:
All you need is `proto-bonding`

brada4, 2025-03-27:
You did not share device spec, but if dsa switch lacks support for lacp all traffic crosses soc cpu.

trenchcoatjedi, 2025-03-28:
This is an x86 PC using all Intel gigabit controllers. This is a regression as there was no issue prior to 24.10

trenchcoatjedi, 2025-03-29:
There is webui support without the linked package. The webui is not the issue. Are you saying that bonds won't work correctly without that package in general or on x86?

brada4, 2025-03-29:
```
cat /proc/net/nonding/*
```

BKPepe, 2025-05-20:
Details were requested by @brada4 and also, he provided some hints. Because of that, closing.

closed 2025-05-20 by BKPepe