openwrt/packages #26020: dns backend restart with adblock blocklist failed

lemonadelemon, 2025-02-20:
Hello Guys!
For a few days now, the Adblock package has not been working and an error is displayed in the panel. The log shows the message in the title. The error with the same message also occurs when all lists are disabled and the whitelist and blacklist are cleared. Updating the package with dependencies and reinstalling them did not help. What could be the cause?

Here are the log entries:
`Wed Feb 19 19:04:33 2025 user.info adblock-4.2.5-r1[2135]: adblock instance started ::: action: start, priority: 0, pid: 2135
Thu Feb 20 08:05:50 2025 user.err adblock-4.2.5-r1[2135]: dns backend restart with adblock blocklist failed
Thu Feb 20 08:17:09 2025 user.info adblock-4.2.5-r1[7020]: adblock instance started ::: action: reload, priority: 0, pid: 7020
Thu Feb 20 08:18:09 2025 user.err adblock-4.2.5-r1[7020]: dns backend restart with adblock blocklist failed
Thu Feb 20 10:18:09 2025 user.info adblock-4.2.5-r1[16130]: adblock instance started ::: action: reload, priority: 0, pid: 16130
Thu Feb 20 10:19:08 2025 user.err adblock-4.2.5-r1[16130]: dns backend restart with adblock blocklist failed`

And this is my machine:
`  OpenWrt 24.10-SNAPSHOT, r28432-7609571dcd                                 
 |                                                                          
  Build time: 2025-02-07 06:44 UTC                                          
  Cezary Jackiewicz, https://eko.one.pl                                     
                                                                            
 -----------------------------------------------------------------------------
  Machine: Netgear R6220                                                    
  Uptime: 0d, 02:21:39                                                      
  Load: 0.16 0.20 0.10                                                      
  Flash: total: 12.7MB, free: 11.8MB, used: 2%                              
  Memory: total: 116.6MB, free: 61.9MB, used: 46%                           
  Leases: 8                                                                
  lan: static, 192.168.1.1                                                  
`

lemonadelemon, 2025-02-20:
`config adblock 'global'
	option adb_enabled '1'
	option adb_debug '0'
	option adb_forcedns '0'
	option adb_safesearch '0'
	option adb_dnsfilereset '0'
	option adb_mail '0'
	option adb_report '0'
	option adb_backup '1'
	option adb_dns 'dnsmasq'
	option adb_fetchutil 'uclient-fetch'
	list adb_sources 'adaway'
`

dibdot, 2025-02-20:
Still no adblock debug output, just dnsmasq stuff.

lemonadelemon, 2025-02-20:
Sorry for my mistake. I hope this is what you meant:

> Thu Feb 20 18:55:31 2025 user.debug adblock-4.2.5-r1[13216]: f_dns    ::: dns: dnsmasq, dns_dir: /tmp/dnsmasq.d, dns_file: adb_list.overall, dns_user: dnsmasq, dns_instance: 0, backup_dir: /tmp/adblock-Backup, jail_dir: /tmp
> Thu Feb 20 18:55:31 2025 user.debug adblock-4.2.5-r1[13216]: f_fetch  ::: fetch_util: /bin/uclient-fetch, fetch_parm:  --timeout=20 -O
> Thu Feb 20 18:55:35 2025 user.debug adblock-4.2.5-r1[13216]: f_dnsup  ::: dns: dnsmasq, cache_cmd: -, lookup_cmd: /usr/bin/nslookup, lookup_domain: example.com, restart_rc: 0, dns_flush: 0, dns_timeout: 20, dns_pid: , dns_cnt: 0, in_rc: 4, out_rc: 4
> Thu Feb 20 18:55:35 2025 user.debug adblock-4.2.5-r1[13527]: f_dns    ::: dns: dnsmasq, dns_dir: /tmp/dnsmasq.d, dns_file: adb_list.overall, dns_user: dnsmasq, dns_instance: 0, backup_dir: /tmp/adblock-Backup, jail_dir: /tmp
> Thu Feb 20 18:55:35 2025 user.debug adblock-4.2.5-r1[13527]: f_fetch  ::: fetch_util: /bin/uclient-fetch, fetch_parm:  --timeout=20 -O
> Thu Feb 20 18:55:35 2025 user.info adblock-4.2.5-r1[13527]: adblock instance started ::: action: restart, priority: 0, pid: 13527
> Thu Feb 20 18:56:04 2025 user.debug adblock-4.2.5-r1[13527]: f_dnsup  ::: dns: dnsmasq, cache_cmd: -, lookup_cmd: /usr/bin/nslookup, lookup_domain: example.com, restart_rc: 0, dns_flush: 0, dns_timeout: 20, dns_pid: 13846, dns_cnt: 21, in_rc: 0, out_rc: 4
> Thu Feb 20 18:56:04 2025 user.debug adblock-4.2.5-r1[13527]: f_list   ::: name: blacklist, mode: blacklist, cnt: 0, in_rc: 0, out_rc: 0
> Thu Feb 20 18:56:04 2025 user.debug adblock-4.2.5-r1[13527]: f_list   ::: name: whitelist, mode: whitelist, cnt: 1, in_rc: 0, out_rc: 0
> Thu Feb 20 18:56:04 2025 user.debug adblock-4.2.5-r1[13527]: f_list   ::: name: iplist, mode: iplist, cnt: 0, in_rc: 0, out_rc: 0
> Thu Feb 20 18:56:04 2025 user.debug adblock-4.2.5-r1[13527]: f_list   ::: name: adaway, mode: restore, cnt: 6540, in_rc: 4, out_rc: 0
> Thu Feb 20 18:56:05 2025 user.debug adblock-4.2.5-r1[13527]: f_list   ::: name: -, mode: merge, cnt: 6540, in_rc: 4, out_rc: 0
> Thu Feb 20 18:56:05 2025 user.debug adblock-4.2.5-r1[13527]: f_tld    ::: name: -, cnt: 6540, cnt_tld: 4289, cnt_rem: 4289
> Thu Feb 20 18:56:05 2025 user.debug adblock-4.2.5-r1[13527]: f_list   ::: name: -, mode: final, cnt: 4289, in_rc: 4, out_rc: 0
> Thu Feb 20 18:56:32 2025 user.debug adblock-4.2.5-r1[13527]: f_dnsup  ::: dns: dnsmasq, cache_cmd: -, lookup_cmd: /usr/bin/nslookup, lookup_domain: example.com, restart_rc: 0, dns_flush: 0, dns_timeout: 20, dns_pid: 14409, dns_cnt: 21, in_rc: 0, out_rc: 4
> Thu Feb 20 18:56:32 2025 user.err adblock-4.2.5-r1[13527]: dns backend restart with adblock blocklist failed
> Thu Feb 20 18:56:37 2025 user.debug adblock-4.2.5-r1[13527]: f_dnsup  ::: dns: dnsmasq, cache_cmd: -, lookup_cmd: /usr/bin/nslookup, lookup_domain: example.com, restart_rc: 0, dns_flush: 0, dns_timeout: 20, dns_pid: 14409, dns_cnt: 0, in_rc: 4, out_rc: 4

dibdot, 2025-02-20:
OK, thanks. Please post the output of `nslookup example.com` (that's the default lookup/check domain). Most probably this lookup domain is blocked in your environment. Either you use a different domain or you set this check to “false” (adb_lookupdomain, see readme).

lemonadelemon, 2025-02-20:
```
root@OpenWrt:~# nslookup example.com
Server:         127.0.0.1
Address:        127.0.0.1:53

** server can't find example.com: SERVFAIL

** server can't find example.com: SERVFAIL
```

lemonadelemon, 2025-02-20:
Ok, changed it to "google.com", and now in log you can see:

```
Thu Feb 20 19:52:47 2025 user.debug adblock-4.2.5-r1[18968]: f_dns    ::: dns: dnsmasq, dns_dir: /tmp/dnsmasq.d, dns_file: adb_list.overall, dns_user: dnsmasq, dns_instance: 0, backup_dir: /tmp/adblock-Backup, jail_dir: /tmp
Thu Feb 20 19:52:47 2025 user.debug adblock-4.2.5-r1[18968]: f_fetch  ::: fetch_util: /bin/uclient-fetch, fetch_parm:  --timeout=20 -O
Thu Feb 20 19:52:51 2025 user.debug adblock-4.2.5-r1[18968]: f_dnsup  ::: dns: dnsmasq, cache_cmd: -, lookup_cmd: /usr/bin/nslookup, lookup_domain: google.com, restart_rc: 0, dns_flush: 0, dns_timeout: 20, dns_pid: , dns_cnt: 0, in_rc: 4, out_rc: 4
Thu Feb 20 19:52:51 2025 user.debug adblock-4.2.5-r1[19279]: f_dns    ::: dns: dnsmasq, dns_dir: /tmp/dnsmasq.d, dns_file: adb_list.overall, dns_user: dnsmasq, dns_instance: 0, backup_dir: /tmp/adblock-Backup, jail_dir: /tmp
Thu Feb 20 19:52:51 2025 user.debug adblock-4.2.5-r1[19279]: f_fetch  ::: fetch_util: /bin/uclient-fetch, fetch_parm:  --timeout=20 -O
Thu Feb 20 19:52:51 2025 user.info adblock-4.2.5-r1[19279]: adblock instance started ::: action: restart, priority: 0, pid: 19279
Thu Feb 20 19:52:57 2025 user.debug adblock-4.2.5-r1[19279]: f_dnsup  ::: dns: dnsmasq, cache_cmd: -, lookup_cmd: /usr/bin/nslookup, lookup_domain: google.com, restart_rc: 0, dns_flush: 0, dns_timeout: 20, dns_pid: 19598, dns_cnt: 0, in_rc: 0, out_rc: 0
Thu Feb 20 19:52:57 2025 user.debug adblock-4.2.5-r1[19279]: f_list   ::: name: blacklist, mode: blacklist, cnt: 0, in_rc: 0, out_rc: 0
Thu Feb 20 19:52:58 2025 user.debug adblock-4.2.5-r1[19279]: f_list   ::: name: whitelist, mode: whitelist, cnt: 1, in_rc: 0, out_rc: 0
Thu Feb 20 19:52:58 2025 user.debug adblock-4.2.5-r1[19279]: f_list   ::: name: iplist, mode: iplist, cnt: 0, in_rc: 0, out_rc: 0
Thu Feb 20 19:52:58 2025 user.debug adblock-4.2.5-r1[19279]: f_list   ::: name: adaway, mode: restore, cnt: 6540, in_rc: 4, out_rc: 0
Thu Feb 20 19:52:58 2025 user.debug adblock-4.2.5-r1[19279]: f_list   ::: name: -, mode: merge, cnt: 6540, in_rc: 4, out_rc: 0
Thu Feb 20 19:52:58 2025 user.debug adblock-4.2.5-r1[19279]: f_tld    ::: name: -, cnt: 6540, cnt_tld: 4289, cnt_rem: 4289
Thu Feb 20 19:52:58 2025 user.debug adblock-4.2.5-r1[19279]: f_list   ::: name: -, mode: final, cnt: 4289, in_rc: 4, out_rc: 0
Thu Feb 20 19:53:03 2025 user.debug adblock-4.2.5-r1[19279]: f_dnsup  ::: dns: dnsmasq, cache_cmd: -, lookup_cmd: /usr/bin/nslookup, lookup_domain: google.com, restart_rc: 0, dns_flush: 0, dns_timeout: 20, dns_pid: 19848, dns_cnt: 0, in_rc: 0, out_rc: 0
Thu Feb 20 19:53:04 2025 user.info adblock-4.2.5-r1[19279]: blocklist with overall 4.289 blocked domains loaded successfully (Netgear R6220, ramips/mt7621, OpenWrt 24.10-SNAPSHOT r28432-7609571dcd )
```
Thank you very much for your interest and help

closed 2025-02-20 by dibdot